probe-rs/probe-rs #2720: rtthost can only be run once before having to power cycle target

arbrauns, 2024-07-30:
**Describe the bug**
When rtthost is run, it cannot be run again until the target is power cycled.

**To Reproduce**
```
$ git describe
v0.1.0-153-gf8bff658
$ RUST_LOG=probe_rs=trace cargo run --quiet --bin rtthost -- --chip STM32F767VITx --scan-region 0x20007120 --list
Attaching to RTT...
Found control block at 0x20007120
Up channels:
  0: Terminal (buffer size 16384)
  1: btmonitor (buffer size 1024)
Down channels:
  0: Terminal (buffer size 32)
$ RUST_LOG=probe_rs=trace cargo run --quiet --bin rtthost -- --chip STM32F767VITx --scan-region 0x20007120 --list
Error: An ARM specific error occurred.
```

Also, as can be seen, tracing doesn't work.

**Expected behavior**
rtthost doesn't soft-brick the target.

**Desktop:** Ubuntu 22.04 LTS

bugadani, 2024-07-30:
Yeah rtthost is a bit left behind... Does probe-rs attach work for you, or you're using rtthost because you don't have the .elf at hand?

arbrauns, 2024-07-30:
No, `probe-rs attach` also fails the second time around:

```
$ cargo run --quiet --bin probe-rs -- attach redacted.elf --chip STM32F767VITx
 WARN probe_rs::session: Could not clear all hardware breakpoints: Arm(Timeout)
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: A timeout occurred during an operation.
```

Either way, it doesn't allow selecting RTT channels, so it's not really an option for me.

arbrauns, 2024-07-30:
Stopping a working `probe-rs attach` with `^C` gives the following warnings:

```
 WARN probe_rs::probe::arm_debug_interface: DAP FAULT
 WARN probe_rs::probe::arm_debug_interface: Writing DAP register failed. Ctrl/Stat register value is: Ctrl {
    .0: 4160753539,
    csyspwrupack: true,
    csyspwrupreq: true,
    cdbgpwrupack: true,
    cdbgpwrupreq: true,
    cdbgrstack: true,
    c_dbg_rst_req: false,
    errmode: false,
    trn_cnt: 0,
    mask_lane: 15,
    w_data_err: true,
    read_ok: false,
    sticky_err: false,
    stick_cmp: false,
    trn_mode: 0,
    sticky_orun: true,
    orun_detect: true,
}
 WARN probe_rs::session: Failed to deconfigure device during shutdown: Arm(AccessPort { address: FullyQualifiedApAddress { dp: Default, ap: V1(0) }, source: RegisterWrite { address: 12, name: "DRW", source: Dap(FaultResponse) } })
 WARN probe_rs::probe::arm_debug_interface: DAP FAULT
 WARN probe_rs::probe::arm_debug_interface: Writing DAP register failed. Ctrl/Stat register value is: Ctrl {
    .0: 4160753539,
    csyspwrupack: true,
    csyspwrupreq: true,
    cdbgpwrupack: true,
    cdbgpwrupreq: true,
    cdbgrstack: true,
    c_dbg_rst_req: false,
    errmode: false,
    trn_cnt: 0,
    mask_lane: 15,
    w_data_err: true,
    read_ok: false,
    sticky_err: false,
    stick_cmp: false,
    trn_mode: 0,
    sticky_orun: true,
    orun_detect: true,
}
```