probe-rs/probe-rs #2761: Version 0.24.0 of probe-rs-tools on crates.io fails to install

BartMassey, 2024-08-13:
Right now, `cargo install probe-rs-tools` fails with

```
rror[E0252]: the name `FromBytes` is defined multiple times
  --> /home/bart/.cargo/registry/src/index.crates.io-6f17d22bba15001f/probe-rs-0.24.0/src/rtt.rs:59:23
   |
58 | use zerocopy::{AsBytes, FromBytes};
   |                         --------- previous import of the macro `FromBytes` here
59 | use zerocopy_derive::{FromBytes, FromZeroes};
   |                       ^^^^^^^^^--
   |                       |
   |                       `FromBytes` reimported here
   |                       help: remove unnecessary import
   |
   = note: `FromBytes` must be defined only once in the macro namespace of this module
```

It looks like commit 6e975baf fixes this; current `master` compiles and installs fine.

Any chance we could get a 0.24.1 on `crates.io` that resolves the issue?

Discussion at rust-embedded/discovery/#579

bugadani, 2024-08-13:
Thank you. This is a duplicate of https://github.com/probe-rs/webpage/issues/149. You can install 0.24 from any binary distribution source, or `cargo install probe-rs-tools --locked`.

closed 2024-08-13 by bugadani

BartMassey, 2024-08-13:
Thanks! Sorry for the duplicate.

Would still suggest an 0.24.1 release if possible, to avoid more of this.

Yatekii, 2024-08-13:
We do not promote installation via cargo. The release of the crate was actually an oversight.

BartMassey, 2024-08-13:
Good to know. In this case, it would be nice to have an explanation and rationale for this unusual policy somewhere prominent on the site and on the release page: I looked briefly for one and didn't find it.

brainstorm, 2024-12-02:
Agreed, I found myself on the same situation as @BartMassey right now.

Tiwalun, 2024-12-02:
This will always be a problem with installing from crates.io. We can of course fix this temporarily, but it can always happen that there is an issue with a dependency. 

So running `cargo install --locked` is the only solution which should always work.

brainstorm, 2024-12-02:
> This will always be a problem with installing from crates.io. We can of course fix this temporarily, but it can always happen that there is an issue with a dependency.

Can't you just fork that dependency?

bugadani, 2024-12-02:
No

BartMassey, 2024-12-02:
As long as `cargo install --locked` continues to work in future releases I'm good. I do think it's ideal that folks be able to install from `crates.io` if possible for trust and convenience reasons. If this is not possible going forward, I'd really like to know why, so that I can explain it to others.

Thanks much for the work on `probe-rs-tools`. It is really appreciated.

BartMassey, 2024-12-02:
I would encourage you to continue to publish `probe-rs-tools` to `crates.io` and support `cargo install --locked` from there. Folks will need to have appropriate dependencies, and that does increase your support burden a bit —  you need to point people to the page that describes the dependencies. However, there are quite a few people, especially in corporate environments, who are not going to be willing or permitted to use tools installed by a shell script from an unfamiliar site on the Internet; this approach provides an alternative for them. Installing from the `master` branch of the git repo seems problematic for new users, as it is presumably more fragile and less documented than a release version.

In any case, it looks like the current [installation instructions](https://probe.rs/docs/getting-started/installation/#installing-from-source-%28cargo-install%29) say that `cargo install --locked` is a supported alternative. I will point my users to this installation documentation and we'll take it from there.

Thanks again!

Yatekii, 2024-12-03:
> However, there are quite a few people, especially in corporate environments

Ultimately every installer runs arbitrary code and could mess with your system. While I am not necessarily against publishing to crates.io, I am very much against the sentiment that installing via shellscript is somehow the devil when we execute arbitrary code in proc_macros or brew formulas all the time. Even if any of those get vetted, it's easy to just override a brew formula or a proc_macro inbetween the vetting and the installing by your users.

BartMassey, 2024-12-03:
@Yatekii The problem isn't so much the credibility of the installer, it's the credibility of the organization.

Most Rust folks know, for example, that `crates.io`'s web infrastructure is actively and competently managed: it is unlikely that someone could break into the website and hack things up to cause a security issue. We know that accessing `crates.io` through `cargo` goes through a well-vetted security path, with no real chance of domain spoofing or typosquatting or whatever.

I believe that these things are true for `probe.rs` as well. However, you can see why corporate users would prefer to validate a few highly-visible upstream sources and be done with it, rather than a whole bunch of initially-mysterious websites asking for their shell scripts to be run.

Anyhow, I think I'm good. I'm going to gratefully leave this closed and go poke at what I need to. Thanks again to all!