probe-rs/probe-rs #2772: Issue with HPM53XX Flash Algorithm

rayc345, 2024-08-19:
**Describe the bug**
Tried to erase and program the flash of HPM5301 chip using probe-rs, but failed:

**To Reproduce**
Steps to reproduce the behavior:

Connect HOM5301 to PC using J-link
Call command:
```probe-rs erase --chip=HPM5301 --protocol=jtag```
probe-rs is build upon latest github mater branch using 'cargo build' command
Output:
Erasing ✔ [00:00:00] [--------------------------------------------------------]      0 B/1.00 MiB @      0 B/s (eta 0s )Error: The initialization of the flash algorithm failed.

Tried another command:
```probe-rs download --chip=HPM5301 --protocol=jtag C:\Users\Admin\Documents\sdk_env\hpm_sdk\samples\hello_world\hpm5301evklite_flash_xip_debug\segger_embedded_studio\Output\Debug\Exe\demo.elf```
Failed too, output:
      Erasing ✔ [00:00:00] [-------------------------------------------------]      0 B/32.00 KiB @      0 B/s (eta 0s )
  Programming ✔ [00:00:00] [--------------------------------------------------]      0 B/     0 B @      0 B/s (eta 0s )Error: The flashing procedure failed for 'C:\Users\Admin\Documents\sdk_env\hpm_sdk\samples\hello_world\hpm5301evklite_flash_xip_debug\segger_embedded_studio\Output\Debug\Exe\demo.elf'.

Caused by:
    0: The initialization of the flash algorithm failed.
    1: Stack overflow detected during Erase.

Caused by:
    Stack overflow detected during Erase.

**Desktop (please complete the following information):**

- Windows 10 

**Additional context**
Add any other context about the problem here.

The log file of the erase operation is attached. I'm new to rust and cannot fix it alone.
[x.zip](https://github.com/user-attachments/files/16656876/x.zip)

bugadani, 2024-08-19:
Please help us find a working stack size.

To do this:

- Copy [this](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/targets/HPM5300_Series.yaml) file to somewhere on your PC.
- Add a `stack_size` field to the flash algorithm like [this](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/targets/MIMXRT500.yaml#L319)
- Try your modification by running `probe-rs erase  --chip HPM5301 --protocol jtag --chip-description-path <path to your modified yaml>`

rayc345, 2024-08-19:
Hello Dániel:
I tried your method, it worked!
I added ```stack_size: 1024``` and it works just fine, 768 is still not enough.

closed 2024-08-19 by rayc345

bugadani, 2024-08-19:
You don't have to manually close issues, merging the PR automatically does that for us at the appropriate time :) Thanks for the help!

PR #2773 (2024-08-29): Fix stack size of HPM5300 flash algo
Closes #2772