probe-rs/probe-rs #2774: Compatibility with Jlink

rayc345, 2024-08-19:
**Describe the bug**
When Jlink is configured to be in CMSIS-DAP mode, probe-rs could not use it in both Jlink and CMSIS-DAP mode.

**To Reproduce**
1. Set a Jlink probe to be in CMSIS-DAP v2 mode using Jlink configurator.
2. Run: ```probe-rs gdb --chip=HPM5361 --protocol=jtag --gdb-connection-string=127.0.0.1:3333```
   Software output:

```
Available Probes:
0: J-Link CMSIS-DAP -- 1366:1080:000821009099 (CMSIS-DAP)
1: J-Link CMSIS-DAP -- 1366:1080:000821009099 (J-Link)
Selection: 1
Error: Failed to open the debug probe.

Caused by:
    0: An error which is specific to the debug probe in use occurred.
    1: device is not a J-Link device
```

Whether select 0 or 1 would give the same error result.

**Desktop (please complete the following information):**

- Windows 10

**Additional context**
Add any other context about the problem here.

bugadani, 2024-08-19:
Could you please run the same command with `RUST_LOG="probe_rs::probe::cmsisdap=trace"`, and selecting the CMSIS-DAP option from the list? It might give us some hints as to what's happening.

rayc345, 2024-08-19:
Sorry, I tried to set the environment, but the console output did not change. I tried run the command with log enable, I got a little log data:
```
{"timestamp":"2024-08-19T12:40:34.710808Z","level":"INFO","fields":{"message":"Writing log to \"x.log\""},"target":"probe_rs::util::logging","filename":"probe-rs-tools\\src\\bin\\probe-rs\\util\\logging.rs","line_number":22}
{"timestamp":"2024-08-19T12:40:34.740918Z","level":"DEBUG","fields":{"message":"Searching registry for chip with name HPM5301"},"target":"probe_rs::config::registry","filename":"probe-rs\\src\\config\\registry.rs","line_number":162}
{"timestamp":"2024-08-19T12:40:34.741041Z","level":"DEBUG","fields":{"message":"Exact match for chip name: HPM5301"},"target":"probe_rs::config::registry","filename":"probe-rs\\src\\config\\registry.rs","line_number":172}
{"timestamp":"2024-08-19T12:40:34.741321Z","level":"INFO","fields":{"message":"Using sequence Riscv(DefaultRiscvSequence(()))"},"target":"probe_rs::config::target","filename":"probe-rs\\src\\config\\target.rs","line_number":121}
{"timestamp":"2024-08-19T12:40:34.741432Z","level":"INFO","fields":{"message":"new"},"target":"probe_rs::probe::cmsisdap::tools","filename":"probe-rs\\src\\probe\\cmsisdap\\tools.rs","line_number":18,"span":{"name":"list_cmsisdap_devices"},"spans":[]}
{"timestamp":"2024-08-19T12:40:34.741470Z","level":"INFO","fields":{"message":"enter"},"target":"probe_rs::probe::cmsisdap::tools","filename":"probe-rs\\src\\probe\\cmsisdap\\tools.rs","line_number":18,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2024-08-19T12:40:34.741502Z","level":"DEBUG","fields":{"message":"Searching for CMSIS-DAP probes using nusb"},"target":"probe_rs::probe::cmsisdap::tools","filename":"probe-rs\\src\\probe\\cmsisdap\\tools.rs","line_number":20,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2024-08-19T12:40:34.741675Z","level":"DEBUG","fields":{"message":"Probing device \"USB\\\\VID_1366&PID_1080\\\\000821009099\"","log.target":"nusb::platform::windows_winusb::enumeration","log.module_path":"nusb::platform::windows_winusb::enumeration","log.file":"C:\\Users\\Admin\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\nusb-0.1.10\\src\\platform\\windows_winusb\\enumeration.rs","log.line":36},"target":"nusb::platform::windows_winusb::enumeration","filename":"C:\\Users\\Admin\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\nusb-0.1.10\\src\\platform\\windows_winusb\\enumeration.rs","line_number":36,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2024-08-19T12:40:34.741879Z","level":"DEBUG","fields":{"message":"Probing device \"USB\\\\VID_3277&PID_0055\\\\200901010001\"","log.target":"nusb::platform::windows_winusb::enumeration","log.module_path":"nusb::platform::windows_winusb::enumeration","log.file":"C:\\Users\\Admin\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\nusb-0.1.10\\src\\platform\\windows_winusb\\enumeration.rs","log.line":36},"target":"nusb::platform::windows_winusb::enumeration","filename":"C:\\Users\\Admin\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\nusb-0.1.10\\src\\platform\\windows_winusb\\enumeration.rs","line_number":36,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2024-08-19T12:40:34.742553Z","level":"DEBUG","fields":{"message":"Probing device \"USB\\\\VID_8087&PID_0033\\\\5&1423ACB5&0&10\"","log.target":"nusb::platform::windows_winusb::enumeration","log.module_path":"nusb::platform::windows_winusb::enumeration","log.file":"C:\\Users\\Admin\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\nusb-0.1.10\\src\\platform\\windows_winusb\\enumeration.rs","log.line":36},"target":"nusb::platform::windows_winusb::enumeration","filename":"C:\\Users\\Admin\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\nusb-0.1.10\\src\\platform\\windows_winusb\\enumeration.rs","line_number":36,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2024-08-19T12:40:34.742691Z","level":"TRACE","fields":{"message":"J-Link CMSIS-DAP: CMSIS-DAP device with 0 interfaces"},"target":"probe_rs::probe::cmsisdap::tools","filename":"probe-rs\\src\\probe\\cmsisdap\\tools.rs","line_number":96,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2024-08-19T12:40:34.742715Z","level":"TRACE","fields":{"message":"No HID interface for CMSIS-DAP found."},"target":"probe_rs::probe::cmsisdap::tools","filename":"probe-rs\\src\\probe\\cmsisdap\\tools.rs","line_number":105,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2024-08-19T12:40:34.742759Z","level":"DEBUG","fields":{"message":"Found 1 CMSIS-DAP probes using nusb, searching HID"},"target":"probe_rs::probe::cmsisdap::tools","filename":"probe-rs\\src\\probe\\cmsisdap\\tools.rs","line_number":32,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2024-08-19T12:40:34.746243Z","level":"DEBUG","fields":{"message":"Found 1 CMSIS-DAP probes total"},"target":"probe_rs::probe::cmsisdap::tools","filename":"probe-rs\\src\\probe\\cmsisdap\\tools.rs","line_number":54,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_cmsisdap_devices"}]}
[... cut ...]  In,\n            transfer_type: Bulk,\n            max_packet_size: 0x200,\n            packets_per_microframe: 0x1,\n            interval: 0x1,\n        },\n    ],\n}"},"target":"probe_rs::probe::jlink","filename":"probe-rs\\src\\probe\\jlink\\mod.rs","line_number":114}
{"timestamp":"2024-08-19T12:40:38.127704Z","level":"INFO","fields":{"message":"Wrote log to \"x.log\""},"target":"probe_rs::util::logging","filename":"probe-rs-tools\\src\\bin\\probe-rs\\util\\logging.rs","line_number":33}

```

rayc345, 2024-08-19:
Using msys2 I got what we need:
```
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: J-Link CMSIS-DAP: CMSIS-DAP device with 0 interfaces
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using nusb, searching HID
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
Available Probes:
0: J-Link CMSIS-DAP -- 1366:1080:000821009099 (CMSIS-DAP)
1: J-Link CMSIS-DAP -- 1366:1080:000821009099 (J-Link)
Selection: 0
TRACE probe_rs::probe::cmsisdap::tools: Attempting to open device matching 1366:1080:000821009099
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 0, device_address: 5, vendor_id: 0x1366, product_id: 0x1080, device_version: 0x0100, class: 0x00, subclass: 0x00, protocol: 0x00, speed: Some(High), manufacturer_string: None, product_string: Some("J-Link CMSIS-DAP"), serial_number: Some("000821009099"), instance_id: "USB\\VID_1366&PID_1080\\000821009099", parent_instance_id: "USB\\ROOT_HUB30\\4&37c5bd69&0&0", port_number: 4, driver: Some("WINUSB"), interfaces: [] }
TRACE probe_rs::probe::cmsisdap::tools: J-Link CMSIS-DAP: CMSIS-DAP device with 0 interfaces
TRACE probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG probe_rs::probe::cmsisdap::tools: Could not open 1366:1080 in CMSIS-DAP v2 mode
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 0, device_address: 2, vendor_id: 0x3277, product_id: 0x0055, device_version: 0x0103, class: 0xEF, subclass: 0x02, protocol: 0x01, speed: Some(High), manufacturer_string: None, product_string: Some("USB2.0 FHD UVC WebCam"), serial_number: Some("200901010001"), instance_id: "USB\\VID_3277&PID_0055\\200901010001", parent_instance_id: "USB\\ROOT_HUB30\\4&37c5bd69&0&0", port_number: 9, driver: Some("usbccgp"), interfaces: [InterfaceInfo { interface_number: 0, class: 0x0E, subclass: 0x03, protocol: 0x00, interface_string: Some("USB2.0 FHD UVC WebCam") }, InterfaceInfo { interface_number: 2, class: 0x0E, subclass: 0x03, protocol: 0x00, interface_string: Some("USB2.0 IR UVC WebCam") }, InterfaceInfo { interface_number: 4, class: 0xFE, subclass: 0x01, protocol: 0x00, interface_string: Some("Camera DFU Device") }] }
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 0, device_address: 3, vendor_id: 0x8087, product_id: 0x0033, device_version: 0x0000, class: 0xE0, subclass: 0x01, protocol: 0x01, speed: Some(Full), manufacturer_string: None, product_string: None, serial_number: None, instance_id: "USB\\VID_8087&PID_0033\\5&1423ACB5&0&10", parent_instance_id: "USB\\ROOT_HUB30\\4&37c5bd69&0&0", port_number: 10, driver: Some("BTHUSB"), interfaces: [] }
DEBUG probe_rs::probe::cmsisdap::tools: Attempting to open 1366:1080 in CMSIS-DAP v1 mode
Error: Failed to open the debug probe.

Caused by:
    0: An error which is specific to the debug probe in use occurred.
    1: device is not a J-Link device

Admin@DESKTOP-8E7HGPQ MINGW64 /c/Users/Admin/Documents/probe-rs/target/debug
```

rayc345, 2024-08-19:
Another output with all data from probe besides CMSIS-DAP
```
Admin@DESKTOP-8E7HGPQ MINGW64 /c/Users/Admin/Documents/probe-rs/target/debug
$ RUST_LOG="probe_rs::probe=trace" ./probe-rs debug --chip=HPM5301 --protocol=jtag
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: J-Link CMSIS-DAP: CMSIS-DAP device with 0 interfaces
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using nusb, searching HID
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
DEBUG list_wlink_devices: probe_rs::probe::wlink: Searching for WCH-Link(RV) probes
DEBUG list_wlink_devices: probe_rs::probe::wlink: Found 0 WCH-Link probes total
Available Probes:
0: J-Link CMSIS-DAP -- 1366:1080:000821009099 (CMSIS-DAP)
1: J-Link CMSIS-DAP -- 1366:1080:000821009099 (J-Link)
Selection: 0
TRACE probe_rs::probe::cmsisdap::tools: Attempting to open device matching 1366:1080:000821009099
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 0, device_address: 5, vendor_id: 0x1366, product_id: 0x1080, device_version: 0x0100, class: 0x00, subclass: 0x00, protocol: 0x00, speed: Some(High), manufacturer_string: None, product_string: Some("J-Link CMSIS-DAP"), serial_number: Some("000821009099"), instance_id: "USB\\VID_1366&PID_1080\\000821009099", parent_instance_id: "USB\\ROOT_HUB30\\4&37c5bd69&0&0", port_number: 4, driver: Some("WINUSB"), interfaces: [] }
TRACE probe_rs::probe::cmsisdap::tools: J-Link CMSIS-DAP: CMSIS-DAP device with 0 interfaces
TRACE probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG probe_rs::probe::cmsisdap::tools: Could not open 1366:1080 in CMSIS-DAP v2 mode
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 0, device_address: 2, vendor_id: 0x3277, product_id: 0x0055, device_version: 0x0103, class: 0xEF, subclass: 0x02, protocol: 0x01, speed: Some(High), manufacturer_string: None, product_string: Some("USB2.0 FHD UVC WebCam"), serial_number: Some("200901010001"), instance_id: "USB\\VID_3277&PID_0055\\200901010001", parent_instance_id: "USB\\ROOT_HUB30\\4&37c5bd69&0&0", port_number: 9, driver: Some("usbccgp"), interfaces: [InterfaceInfo { interface_number: 0, class: 0x0E, subclass: 0x03, protocol: 0x00, interface_string: Some("USB2.0 FHD UVC WebCam") }, InterfaceInfo { interface_number: 2, class: 0x0E, subclass: 0x03, protocol: 0x00, interface_string: Some("USB2.0 IR UVC WebCam") }, InterfaceInfo { interface_number: 4, class: 0xFE, subclass: 0x01, protocol: 0x00, interface_string: Some("Camera DFU Device") }] }
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 0, device_address: 3, vendor_id: 0x8087, product_id: 0x0033, device_version: 0x0000, class: 0xE0, subclass: 0x01, protocol: 0x01, speed: Some(Full), manufacturer_string: None, product_string: None, serial_number: None, instance_id: "USB\\VID_8087&PID_0033\\5&1423ACB5&0&10", parent_instance_id: "USB\\ROOT_HUB30\\4&37c5bd69&0&0", port_number: 10, driver: Some("BTHUSB"), interfaces: [] }
DEBUG probe_rs::probe::cmsisdap::tools: Attempting to open 1366:1080 in CMSIS-DAP v1 mode
DEBUG probe_rs::probe::jlink: scanning 1 interfaces
TRACE probe_rs::probe::jlink: active configuration descriptor: Configuration {
    configuration_value: 0x1,
    num_interfaces: 0x1,
    attributes: 0x80,
    max_power: 0x32,
    string_index: Some(
        0x4,
    ),
    interface_alt_settings: [
        InterfaceAltSetting {
            interface_number: 0x0,
            alternate_setting: 0x0,
            num_endpoints: 0x2,
            class: 0xff,
            subclass: 0x0,
            protocol: 0x0,
            string_index: Some(
                0x5,
            ),
            endpoints: [
                Endpoint {
                    address: 0x01,
                    direction: Out,
                    transfer_type: Bulk,
                    max_packet_size: 0x200,
                    packets_per_microframe: 0x1,
                    interval: 0x1,
                },
                Endpoint {
                    address: 0x81,
                    direction: In,
                    transfer_type: Bulk,
                    max_packet_size: 0x200,
                    packets_per_microframe: 0x1,
                    interval: 0x1,
                },
            ],
        },
    ],
}
TRACE probe_rs::probe::jlink: interface #0 descriptors:
TRACE probe_rs::probe::jlink: InterfaceAltSetting {
    interface_number: 0x0,
    alternate_setting: 0x0,
    num_endpoints: 0x2,
    class: 0xff,
    subclass: 0x0,
    protocol: 0x0,
    string_index: Some(
        0x5,
    ),
    endpoints: [
        Endpoint {
            address: 0x01,
            direction: Out,
            transfer_type: Bulk,
            max_packet_size: 0x200,
            packets_per_microframe: 0x1,
            interval: 0x1,
        },
        Endpoint {
            address: 0x81,
            direction: In,
            transfer_type: Bulk,
            max_packet_size: 0x200,
            packets_per_microframe: 0x1,
            interval: 0x1,
        },
    ],
}
Error: Failed to open the debug probe.

Caused by:
    0: An error which is specific to the debug probe in use occurred.
    1: device is not a J-Link device

Admin@DESKTOP-8E7HGPQ MINGW64 /c/Users/Admin/Documents/probe-rs/target/debug
$

```

bugadani, 2024-08-19:
Can you please re-do this test with https://github.com/probe-rs/probe-rs/pull/2775 ?

rayc345, 2024-08-19:
I tried but the output didn't change at all.
After set breakpoints I found all the trace code added was not hit. I am digging myself currently, to see the flow of the code.

rayc345, 2024-08-19:
I commented:
```
            let Some(interface_str) = device_info
                .interfaces()
                .find(|i| i.interface_number() == interface.interface_number())
                .and_then(|i| i.interface_string())
            else {
                continue;
            };
            if !is_cmsis_dap(interface_str) {
                continue;
            }
```
And now output:
```
Admin@DESKTOP-8E7HGPQ MINGW64 /c/Users/Admin/Documents/probe-rs/target/debug
$ RUST_LOG="probe_rs::probe::cmsisdap=trace" ./probe-rs debug --chip=HPM5301 --protocol=jtag
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: J-Link CMSIS-DAP: CMSIS-DAP device with 0 interfaces
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using nusb, searching HID
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
Available Probes:
0: J-Link CMSIS-DAP -- 1366:1080:000821009099 (CMSIS-DAP)
1: J-Link CMSIS-DAP -- 1366:1080:000821009099 (J-Link)
Selection: 0
TRACE probe_rs::probe::cmsisdap::tools: Attempting to open device matching 1366:1080:000821009099
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 0, device_address: 5, vendor_id: 0x1366, product_id: 0x1080, device_version: 0x0100, class: 0x00, subclass: 0x00, protocol: 0x00, speed: Some(High), manufacturer_string: None, product_string: Some("J-Link CMSIS-DAP"), serial_number: Some("000821009099"), instance_id: "USB\\VID_1366&PID_1080\\000821009099", parent_instance_id: "USB\\ROOT_HUB30\\4&37c5bd69&0&0", port_number: 4, driver: Some("WINUSB"), interfaces: [] }
TRACE probe_rs::probe::cmsisdap::tools: J-Link CMSIS-DAP: CMSIS-DAP device with 0 interfaces
TRACE probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
Interface num: 0
DEBUG probe_rs::probe::cmsisdap::tools: Opening 1366:1080 in CMSIS-DAPv2 mode
DEBUG probe_rs::probe::cmsisdap::commands: Draining probe of any pending data.
DEBUG probe_rs::probe::cmsisdap::commands: Attempt 1 to find packet size
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [00, 02, 00, 02]...
DEBUG probe_rs::probe::cmsisdap::commands: Success: packet size is 512
DEBUG probe_rs::probe::cmsisdap::commands: Configuring probe to use packet size 512
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FE]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [00, 01, 08]...
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, F0]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [00, 01, 07]...
DEBUG probe_rs::probe::cmsisdap: Detected probe capabilities: Capabilities { _swd_implemented: true, _jtag_implemented: true, swo_uart_implemented: true, swo_manchester_implemented: false, _atomic_commands_implemented: false, _test_domain_timer_implemented: false, swo_streaming_trace_implemented: false, _uart_communication_port_implemented: false, uart_com_port_implemented: false }
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FD]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [00, 04, 00]...
DEBUG probe_rs::probe::cmsisdap: Probe SWO buffer size: 0
DEBUG attach: probe_rs::probe::cmsisdap: Attaching to target system (clock = 1000kHz)
TRACE attach: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 02, 02]...
TRACE attach: probe_rs::probe::cmsisdap::commands: Receive buffer: [02, 02]...
 INFO attach: probe_rs::probe::cmsisdap: Using protocol JTAG
TRACE attach: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 11, 40, 42, 0F, 00]...
TRACE attach: probe_rs::probe::cmsisdap::commands: Receive buffer: [11, 00]...
TRACE attach: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 04, 00, FF, FF, 00]...
TRACE attach: probe_rs::probe::cmsisdap::commands: Receive buffer: [04, 00]...
TRACE attach: probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 01, 00, 01]...
TRACE attach: probe_rs::probe::cmsisdap::commands: Receive buffer: [01, 00]...
DEBUG probe_rs::probe::cmsisdap: Detaching from CMSIS-DAP probe
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 03]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [03, 00]...
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 01, 00]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [01, 00]...
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: The selected probe does not support the 'RISC-V' interface.

Admin@DESKTOP-8E7HGPQ MINGW64 /c/Users/Admin/Documents/probe-rs/target/debug
$
```

bugadani, 2024-08-19:
To work around that issue, there was https://github.com/probe-rs/probe-rs/pull/2578

rayc345, 2024-08-19:
The issue was resolved by using that PR for a RISC-V chip using JTAG.
The Arm SWD issue still exists, I will check this issue with another self-made CMSIS-DAP probe instead of a Jlink CMSIS-DAP configured.

bugadani, 2024-08-19:
I have a J-Link EDU (hardware version 11) that seems to just work, on windows 11, as a CMSIS-DAP probe. probe-rs is able to work with it as a CMSIS-DAP v1 device:

```
TRACE probe_rs::probe::cmsisdap::tools: J-Link CMSIS-DAP: CMSIS-DAP device with 0 interfaces
TRACE probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG probe_rs::probe::cmsisdap::tools: Could not open 1366:1008 in CMSIS-DAP v2 mode
DEBUG probe_rs::probe::cmsisdap::tools: Attempting to open 1366:1008 in CMSIS-DAP v1 mode
DEBUG probe_rs::probe::cmsisdap::commands: Draining probe of any pending data.
DEBUG probe_rs::probe::cmsisdap::commands: Attempt 1 to find packet size
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FF, 00]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [00, 02, 40, 00]...
DEBUG probe_rs::probe::cmsisdap::commands: Success: packet size is 64
DEBUG probe_rs::probe::cmsisdap::commands: Configuring probe to use packet size 64
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, FE, 00]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [00, 01, 40, 00]...
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 00, F0, 00]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [00, 01, 07, 00]...
DEBUG probe_rs::probe::cmsisdap: Detected probe capabilities: Capabilities { _swd_implemented: true, _jtag_implemented: true, swo_uart_implemented: true, swo_manchester_implemented: false, _atomic_commands_implemented: false, _test_domain_timer_implemented: false, swo_streaming_trace_implemented: false, _uart_communication_port_implemented: false, uart_com_port_implemented: false }
```

Could you try updating your probe's firmware?

rayc345, 2024-08-20:
Hello, I tried CMSIS-DAP v1 mode yesterday, and I worked as expected. However, in CMSIS-DAP v2 mode (Currently unavailble feeature for V11 hardware), it does work for probe-rs. Segger only releases firmware that supports CMSIS-DAP v1 for V11 hardware.  For reference: https://www.segger.com/downloads/jlink/ReleaseNotes_JLink.html
```
Version V7.98a (2024-07-17):
Firmware
J-Link EDU/BASE/PLUS V12: Added support for CMSIS-DAPv2 boot mode.
J-Link ULTRA+ V6: Added support for CMSIS-DAPv2 boot mode.
J-Link PRO V6: Added support for CMSIS-DAPv2 boot mode.
J-Link PRO PoE V6: Added support for CMSIS-DAPv2 boot mode.
J-Link WiFi V2: Added support for CMSIS-DAPv2 boot mode.
```

Tiwalun, 2024-12-13:
The detection part should be fixed in #2954. I don't have a newer hardware model available to check what happens with a CMSIS-DAPv2 configured one.