probe-rs/probe-rs #2778: Cannot connect to RISC-V chip via CMSIS-DAP JTAG Protocol

rayc345, 2024-08-22:
**Describe the bug**
Using CMSIS-DAP on a self-made probe, issues were found on connection to RISC-V chips. But it works fine with OpenOCD.

**To Reproduce**
Steps to reproduce the behavior:

1. Using CMSIS-DAP Probe
2. Connect to RISC-V chip via JTAG
3. Then:
```
C:\Users\Admin\Documents\probe-rs\target\debug>probe-rs debug --chip=HPM5361 --protocol=jtag
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: The selected probe does not support the 'RISC-V' interface.
```

**Expected behavior**
Connect successfully.

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**

- Windows 10

**Additional context**
Log file:
[x.log](https://github.com/user-attachments/files/16713178/x.log)

JTAG communication waveform captured by logic analyzer shows when the target is RISC-V and using JTAG, there is no waveform on the debug wire. But if we select an Arm chip and try to connect (Still use a hardware RISC-V chip) there is waveform and appropriate data in TDI/TDO though connect failed since the target specified is not in accordance with the hardware.
So, it means all RISC-V chips do not work via JTAG connection.
Communication with Arm chips via JTAG works fine.

rayc345, 2024-08-22:
In #2578 contributions were declined as we could not verify the effectiveness and universality. But I think the problem may lies otherwhere since no communication in JTAG wire means no chip could be connected so universality for other chips besides CPU IP from Andes does not hold on.
And I hope to dig and fix this issue with other contributors together that's way a new issue was created.

bugadani, 2024-08-22:
Thank you, this is a duplicate of https://github.com/probe-rs/probe-rs/issues/2565 . And just for the record, the contribution wasn't declined, it was retracted.

closed 2024-08-22 by bugadani

rayc345, 2024-08-22:
Sorry, I didn't see #2565. So, this is not a bug, but an unimplemented feature, right?
Currently, RISC-V support is only availble via Jlink, FT2232, WCH-Link right?

bugadani, 2024-08-22:
Let's call this a limitation.