probe-rs/probe-rs #2781: Reading returns the first written value over and over

simonalmbro, 2024-08-24:
**Describe the bug**
Reading multiple values just returns the same value over and over and appears to match what was previously written

**To Reproduce**
Steps to reproduce the behavior:

1. `probe-rs write --chip <chip> b32 0x20000000 0x11223344 0x22222222 0x3333333 0x44444444`
2. `probe-rs read --chip <chip> b32 0x20000000 4`

**Expected behavior**
The read values should be 0x11223344 0x22222222 0x3333333 0x44444444.

**Stacktrace**
N/A

**Desktop (please complete the following information):**

- Linux Ubuntu

**Additional context**
I have a custom board with an FTDI FT2232H chip to flash an STM32L431RCT6 that I normally use to flash C++ code with openocd and that has been stable for about 6 months and still works fine. I'm now trying to flash some Rust but have encountered problems. With some help from the Rust community the issue appears be related to probe-rs.

My exact output:
> $ probe-rs write --chip STM32L431RCTx b32 0x20000000 0x11223344 0x22222222 0x3333333 0x44444444
> $ probe-rs read --chip STM32L431RCTx b32 0x20000000 4
> 11223344 11223344 11223344 11223344 

I also see the same when trying to flash using 'cargo embed". This is a part of the output when trying to flash:

> ERROR probe_rs::flashing::flasher: Failed to verify flash algorithm. Data mismatch at address 0x20000004
> ERROR probe_rs::flashing::flasher: Original instruction: 0x062d780d
> ERROR probe_rs::flashing::flasher: Readback instruction: 0xe00abe00
> ERROR probe_rs::flashing::flasher: Original: [e00abe00, 62d780d, 24084068, d3000040, 1e644058, 1c49d1fa, 2a001e52, 4770d1f, 8f4ff3bf, 48584770, 49586800, d000500, d0001840, 47702001, 6a004855, fc00280, b5004770, f7ff4602, 2801ffee, f7ffd108, 2801fff3, 484fd104, d3014282, bd002001, bd002000, 4602b500, ffddf7ff, d0022801, d8002d0, 4948bd00, 40080ad0, d5f90391, 300130ff, 4842bd00, 60814944, 60814944, 60012100, 61014943, 3c06a00, 4843d406, 60014941, 60412106, 60814941, 47702000, 49372001, 614807c0, 47702000, 47702001, 49384833, 13c16101, 69416141, 4122201, 61414311, 4a354937, 6011e000, 3db6903, 2100d4fb, 46086141, b5104770, f7ff4604, 4603ffa8, f7ff4620, 4925ffb5, 610c4c29, 2d800c2, 43021c92, 6948614a, 4122201, 61484310, 8f4ff3bf, 4a244826, 6010e000, 3db690b, 2000d4fb, 69086148, d0014020, 2001610c, b5f0bd10, 4d151dc9, 4f1908c9, 612f00c9, 616b2300, e0184c1a, 616b2301, 60036813, 60436853, 8f4ff3bf, e0004b13, 692e601c, d4fb03f6, 616b2300, 423b692b, 612fd002, bdf02001, 39083008, 29003208, 2000d1e4, bdf0, e0042000, fffffbcb, 40022000, 8020000, 3bf, 45670123, cdef89ab, c3fa, 5555, 40003000, fff, aaaa, 0]
> ERROR probe_rs::flashing::flasher: Readback: [e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00, e00abe00]

bugadani, 2024-08-24:
I'm pretty sure this was fixed by https://github.com/probe-rs/probe-rs/pull/2674

simonalmbro, 2024-08-24:
> I'm pretty sure this was fixed by #2674

Is there a way for me to try to fix or do you know when it will be released. As far as I could tell i was using the latest version already.
probe-rs 0.24.0 (git commit: 6fc653a)

simonalmbro, 2024-09-05:
> I'm pretty sure this was fixed by #2674

Yeah, I've installed the latest master and I can now flash my board which should mean this is also fixed. Thanks!