probe-rs/probe-rs #2787: Support for RP235XY

ChristianJacobsen, 2024-08-31:
**Is your feature request related to a problem? Please describe.**
I can't program the Pico 2 with `probe-rs`.

**Describe the solution you'd like**
I want `probe-rs run --chip RP2350A` to succeed.

**Describe alternatives you've considered**
N/A

**Additional context**
I tried having a look at the `RP2040.yaml` target file to see if I could alter it to fit the RP2350A (which is the chip version for the Pico 2), but the memory section of the data sheet didn't make me that much wiser, and I'm unsure how to handle the dual architecture of the chip with its two Arm cores and two RISC-V cores.

I will give it a few more go's, but if anyone has some pointers for me it would be greatly appreciated. I think it would be nice to support the RP235XY chips - especially with their somewhat unique dual architecture.

sammko, 2024-09-01:
AFAIK the trouble is that the RP2350 uses a newer version of the debug interface protocol (ADIv6) that is not yet implemented in probe-rs. Most recently @ithinuel was working on this.

ithinuel, 2024-09-01:
Indeed still on it 👍 slow but steady progress :)

BollaBerg, 2024-09-18:
I'd really love this for a new project we're starting, is there anything I could do to help, @ithinuel ?

Dygear, 2024-10-08:
I don't have much if anything useful to add to this discussion, other than I wanted to thank ithinuel (Not tagged, to hopefully not bother them) for the hard work. Looking at the documentation for ADIv6 is 400 pages long. That's a super hero level of effort to get that up and running.🦸 Thank you for working on it.

mfiumara, 2024-11-15:
Thanks for the efforts, looking forward to use probe-rs on the RP2350. Curious about the progress :)

U007D, 2024-11-23:
For those interested in progress on probe-rs RP2350 compatibility, I also found this effort [here](https://github.com/konkers/probe-rs/commits/wip/2350/).

mschnell1, 2024-12-09:
I just tried ` runner = "probe-rs run --chip RP2350" `
I get an appropriate error message `not found in the list of known targets`.

(Unfortunately I also get an error message with `runner = "picotool load -u -v -x -t elf"` :) ) 

Of course when trying a lauch of the debugger with a seemingly correct configuration, I get the same error message.

Seemingly the 2350 chip still is not supported.

What can we do ?

chmanie, 2024-12-09:
> I get an appropriate error message `not found in the list of known targets`.

If you're looking to try the effort konkers has been put in as mentioned [above](https://github.com/probe-rs/probe-rs/issues/2787#issuecomment-2495521115), you can try

```bash
cargo install --git https://github.com/konkers/probe-rs --branch wip/2350 probe-rs-tools --locked
```

chmanie, 2024-12-09:
> Can I still use a Pico(1) as a debug-probe ? Do I need to load new firmware on same ?

I don't think that has changed, even though I'm not 100% sure. I'd say just wait and try it out :)

mschnell1, 2024-12-09:
At least it still can access the target 2040 :)

Mossop, 2024-12-12:
> WORKS !!! [@chmanie](https://github.com/chmanie) MANY THANKS ! as well `cargo r` as debgging, including `info!()`
> 
> Do you know why loading uf2 to the Pico 2 does not work ? neither by `runner = "elf2uf2-rs -d"` nor by creating an uf2 file by `elf2uf2-rs` and loading it to the drive the PICO2 shows when booting appropriately ? The Pico 2 stays in "drive" mode and the file is copied and visible in in's RAM, but does not get loaded onto Flash

Make sure you're using the updated memory.x file and including the special image boot block in your binary (see https://github.com/embassy-rs/embassy/blob/main/examples/rp23/src/bin/uart.rs#L17 for example)

mschnell1, 2024-12-12:
:( 

line 17 is the same as I have (copied from the `blinky `example)

and memory.x also is the same as I use.

chmanie, 2024-12-12:
I also had the same issue and ended up using `picotool` for flashing instead.

elpiel, 2024-12-12:
I got my Pico 2 flashing work and with embassy blinky  example it works.
For the firmware I'd like to run is another story... It flashes but no there are no defmt logs

mschnell1, 2024-12-13:
> I also had the same issue and ended up using `picotool` for flashing and converting instead.

doing so by `runner = "picotool load -u -v -x -t elf"`

I get: 
``` 
     Running `picotool load -u -v -x -t elf 'target\thumbv8m.main-none-eabihf\debug\blinky_test'`
ERROR: Unknown command: load
```

How to install an appropriate version of picotool (I got picotool by the C++ SDK)

What exact is picotool, anyway ? 

Thanks !

mschnell1, 2024-12-13:
@Mossop 
> Make sure you're using the updated memory.x file and including the special image boot block in your binary 

That is not the cause of the problem. The embassy feature of creating UF2 information seems to be buggy still. 

I can do:
```
cp ./target/thumbv8m.main-none-eabihf/debug/blinky_test ./target/thumbv8m.main-none-eabihf/debug/blinky_test.elf
picotool uf2 convert  --verbose  'target\thumbv8m.main-none-eabihf\debug\blinky_test.elf' -t elf 'target\thumbv8m.main-none-eabihf\debug\blinky_test.UF2'
```

With this UF2 the pico2 bootloader does work !

chmanie, 2024-12-13:
I'm happy to help but I feel like this has gone a bit off topic. Let's move this to the embassy matrix channel (https://matrix.to/#/%23embassy-rs:matrix.org), or just send me an email (in bio). Talk soon :)

mschnell1, 2024-12-13:
Can't follow there. 
But I am happily working right now. 
And hoping that I will find that a released embassy version for 235X  (not necessary to d/l from, Github) will be available.

tommy-gilligan, 2024-12-15:
```
cargo install --git https://github.com/konkers/probe-rs --branch wip/2350 probe-rs-tools --locked
```
works like a charm. ty ithinuel, konkers and others!

elpiel, 2024-12-17:
I can confirm that after adding some of the linker sections from embassy's blinky example my firmware now flashes and shows defmt logs!

ryan-summers, 2025-01-03:
I've assembled the work by @ithinuel and @konkers into a PR for mainline probe-rs in #3005. Please reach out if you have any issues! I'll be working further to ensure other functions, like GDB, are also supported.

mschnell1, 2025-01-07:
>  And hoping that I will find that a released embassy version for 235X (not necessary to d/l from, Github) will be available.

I upgraded the embassy-rp installation from github I used to reference with my PICO-2 testing project and found that it did not compile any more with the 2025 distribution

So I recreated my cargo.toml file to use the official version of the appropriate crates from crates.io. 

All works fine now !
Thanks a lot !!!!

andrewdavidmackenzie, 2025-01-26:
> ```
> cargo install --git https://github.com/konkers/probe-rs --branch wip/2350 probe-rs-tools --locked
> ```
> 
> works like a charm. ty ithinuel, konkers and others!

Is that work, or others mentioned, planned to be merged into probe-rs, so that it supports Pi Pico 2, any time soon?

konkers, 2025-01-26:
Work on upstreaming is happening in #3005

andrewdavidmackenzie, 2025-01-28:
Thanks for responding. Wow! That's a lot of work to go from Pico to Pico 2 eh, who would have thought it.
Thanks and appreciation to the folks doing all that work!

PR #3005 (2025-02-01): Adding RP235x flashing support
This fixes #2787 

This PR adds support for flashing RP235x devices using `probe-rs`.

This is primarily based on the fantastic work of @konkers and @ithinuel. I've done a manual merge to `master` from their respective branches. I then reviewed all of the changes and pared down the file changes to the bare minimum to support RP235x devices.

This PR successfully:

* Queries the RP2350 CoreSight ROM tables via `probe-rs info --chip RP2350`
<details>

```sh

Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv3, MINDP, DP Designer: ARM Ltd
├──  [... cut]
closed 2025-02-01 by Yatekii