probe-rs/probe-rs #2788: Arm FaultResponse

jaredwolff, 2024-09-01:
**Describe the bug**
Since 9eb1c2f, I'm unable to enumerate my CMSIS-DAP based probe with the nRF9160

The error:

```
thread 'main' panicked at modem_updater/src/lib.rs:373:44:
called `Result::unwrap()` on an `Err` value: Arm(AccessPort { address: FullyQualifiedApAddress { dp: Default, ap: V1(0) }, source: RegisterRead { address: 12, name: "DRW", source: Dap(FaultResponse) } })
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

**To Reproduce**

This snippet is the failure point:

```
    fn setup_device(&mut self) {
        let mut target = self.session.core(0).unwrap();

        // Init UICR.HFXOSR if necessary
        if target.read_word_32(0x00FF801C).unwrap() == 0xFFFFFFFF {
            log::info!("UICR.HFXOSR is not set, setting it to 0x0E");
            target.write_32(0x00FF801C, &[0x0000000E]).unwrap();
        }
```

The read in particular causes the exception.

I can also simply reproduce it by running `probe-rs download --chip nRF9160_xxAA --binary-format hex ../_bin/v5-2-ga33aa1f-fast-blink.hex --allow-erase-all` See full output below.

**Expected behavior**

To read and write to registers without access issues.

**Stacktrace**

```
feather-tester-software/modem_updater on  main [$!?⇡] is 📦 v0.1.0 via 🦀 v1.80.0
❯ probe-rs download --chip nRF9160_xxAA --binary-format hex ../_bin/v5-2-ga33aa1f-fast-blink.hex --allow-erase-all
 INFO probe_rs::config::target: Using sequence Arm(Nrf9160(()))
 INFO attach: probe_rs::probe::cmsisdap: Using protocol SWD
 INFO probe_rs::architecture::arm::sequences: Debug port Default is powered down, powering up
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Checking if core 0 is unlocked
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Core 0 is already unlocked
 INFO probe_rs::session: Core status: Running
 INFO probe_rs::session: Halting core...
 INFO probe_rs::session: Resuming core...
 INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }                                                                                                                                                          INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }                                                                                                                                                              INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack                                                                                                                                                                                                          INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002778                                                                                                                                                                                                                                                   INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }
 INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }
 INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack
 INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002778
 INFO session_drop: probe_rs::session: Core status: Halted(Exception)                                                                                                                                                                                                                                              INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                                                                                                                                                                                                                     INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                                                                                                                                                                                                                    Error: The flashing procedure failed for '../_bin/v5-2-ga33aa1f-fast-blink.hex'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.
    3: Failed to write register DRW at address 0x0c
    4: The debug probe encountered an error.
    5: An error which is specific to the debug probe in use occurred.
    6: CMSIS-DAP responded with an error
```

**Desktop (please complete the following information):**

PopOS 22.04

Yatekii, 2024-09-01:
cc @ithinuel :)

ithinuel, 2024-09-01:
It is odd that this change broke anything 🤔 especially when the only change for the nrf91 sequences is renaming of the `AccessPortType` trait.
Would you mind posting the result of running the same command with `RUST_LOG=trace` please ? (I don’t have an nrf91 board).

jaredwolff, 2024-09-01:
Not many insights from what I could tell. See the attached.

[trace.log](https://github.com/user-attachments/files/16830713/trace.log)

felipebalbi, 2024-09-05:
I have the same issue with MIMXRT685S. Trace attached. One extra piece of information is that if I reset to the commit right before  9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad to get a working run, then reset back to 9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad and run again without power cycling the board, then the "offending" commit works fine. As soon as I power cycle the board, it starts to fail again.

[trace.log](https://github.com/user-attachments/files/16897250/trace.log)

felipebalbi, 2024-09-05:
Here's the bisection log, in case it helps in any way:

```console
git bisect start
# status: waiting for both good and bad commits
# good: [d0a7fccc75a53f9c52d32a907aeec957018a96d9] Add RT685S Series support (#2560)
git bisect good d0a7fccc75a53f9c52d32a907aeec957018a96d9
# status: waiting for bad commit, 1 good commit known
# bad: [a06663a48783470c7a344c9daa76b313fcfc6fa4] Bump quinn-proto from 0.11.3 to 0.11.8 (#2797)
git bisect bad a06663a48783470c7a344c9daa76b313fcfc6fa4
# good: [21a3d98dfd50945d61fa67e454f78aff26e75fa4] Update all non-major dependencies (#2691)
git bisect good 21a3d98dfd50945d61fa67e454f78aff26e75fa4
# bad: [365f42ad65c25dbd81c10e28cd24e472a61bcba4] Remove direct crossterm dependency (#2747)
git bisect bad 365f42ad65c25dbd81c10e28cd24e472a61bcba4
# good: [bf29eddb88655a1a7794f4320ffe0d262fe1f620] Cancel in-progress CI jobs (#2707)
git bisect good bf29eddb88655a1a7794f4320ffe0d262fe1f620
# good: [1739f6b7e6d6ca15b8805a85476c5e50ccd4b275] Added aducm302x target support (#2721)
git bisect good 1739f6b7e6d6ca15b8805a85476c5e50ccd4b275
# bad: [9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad] arm: Use AP specific types & enumerate Addresses rather than APs (#2683)
git bisect bad 9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad
# good: [d6c2f2cfb9a12bb0acac7b904cc1055a45515d2d] Erase progress bars (#2724)
git bisect good d6c2f2cfb9a12bb0acac7b904cc1055a45515d2d
# good: [879c53efb2fecd22aa22d7129ab0262ae867e2fa] Lazy-load flash algo (#2725)
git bisect good 879c53efb2fecd22aa22d7129ab0262ae867e2fa
# first bad commit: [9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad] arm: Use AP specific types & enumerate Addresses rather than APs (#2683)
```

ithinuel, 2024-09-06:
`csyspwrupack: false, csyspwrupreq: false, cdbgpwrupack: false, cdbgpwrupreq: false,`.
The thing isn’t turned on. Let me check if I lost one in the process.

ithinuel, 2024-09-07:
Would you be able to get a before/after trace set? I think that’d help pinpoint which operation is missing.

felipebalbi, 2024-09-07:
> Would you be able to get a before/after trace set? I think that’d help pinpoint which operation is missing.

You mean a working trace and a failing trace? Sure. I'll try to get that tomorrow.

felipebalbi, 2024-09-08:
Alright, here they are. Trace contains the commit hash on top of which they ran. I ran a working case on 879c53efb2fecd22aa22d7129ab0262ae867e2fa, a working case on 9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad (run without power cycling) and then power cycled the board to get a failing case on 9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad. Sadly, there are too many differences for a simple `diff` to help out.

[trace-failing-9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad.txt](https://github.com/user-attachments/files/16920196/trace-failing-9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad.txt)
[trace-working-9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad.txt](https://github.com/user-attachments/files/16920197/trace-working-9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad.txt)
[trace-working-879c53efb2fecd22aa22d7129ab0262ae867e2fa.txt](https://github.com/user-attachments/files/16920198/trace-working-879c53efb2fecd22aa22d7129ab0262ae867e2fa.txt)

felipebalbi, 2024-09-08:
And, sorry, here are the same files using `--log-file` instead of terminal redirection.

[trace-failing-9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad.txt](https://github.com/user-attachments/files/16920213/trace-failing-9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad.txt)
[trace-working-9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad.txt](https://github.com/user-attachments/files/16920214/trace-working-9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad.txt)
[trace-working-879c53efb2fecd22aa22d7129ab0262ae867e2fa.txt](https://github.com/user-attachments/files/16920216/trace-working-879c53efb2fecd22aa22d7129ab0262ae867e2fa.txt)

ithinuel, 2024-09-08:
Thank you! So, running 

```shell
# ignoring nusb, async and cmsisdap related logs
alias filter="rg -v --no-line-number '(nusb|async_io|cmsisdap)'"
meld <(filter trace-working-879c53efb2fecd22aa22d7129ab0262ae867e2fa.txt) <(filter trace-failing-9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad.txt) <(filter trace-working-9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad.txt)
```

shows the relevant diff is the missing:
```diff
 DEBUG AP discovery:valid_access_ports: probe_rs::architecture::arm::ap: AP V1(1) is not valid, IDR = 0 dp=Default
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::communication_interface: Changing AP to 0, AP_BANK_SEL to 15 ap=0 register="IDR"
-DEBUG AP discovery:read_ap_register:write_dp_register: probe_rs::architecture::arm::dp: Writing DP register SELECT, value=0x000000f0 ap=0 register="IDR" dp=Default register=Select { .0: 240, ap_sel: 0, ap_bank_sel: 15, dp_bank_sel: 0 }
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::ap: Register read succesful ap=0 register="IDR" value=2222391297
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::ap: Register read succesful ap=0 register="BASE" value=3759136771
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::ap: Register read succesful ap=0 register="BASE2" value=0
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::communication_interface: Changing AP to 0, AP_BANK_SEL to 0 ap=0 register="CSW"
-DEBUG AP discovery:read_ap_register:write_dp_register: probe_rs::architecture::arm::dp: Writing DP register SELECT, value=0x00000000 ap=0 register="CSW" dp=Default register=Select { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::ap: Register read succesful ap=0 register="CSW" value=192938066
-DEBUG AP discovery: probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: 1, HNONSEC: 1, PROT: 6, CACHE: b, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U8 }
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::ap: Register read succesful ap=0 register="CSW" value=1266679888
-DEBUG AP discovery: probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: 0, HNONSEC: 0, PROT: 0, CACHE: b, SPIDEN: 1, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 1, AddrInc: Single, _RES1: 0, SIZE: U32 }
-DEBUG AP discovery: probe_rs::architecture::arm::communication_interface: HNONSEC supported: true
-DEBUG AP discovery: probe_rs::architecture::arm::communication_interface: Device enabled: true
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::communication_interface: Changing AP to 0, AP_BANK_SEL to 15 ap=0 register="CFG"
-DEBUG AP discovery:read_ap_register:write_dp_register: probe_rs::architecture::arm::dp: Writing DP register SELECT, value=0x000000f0 ap=0 register="CFG" dp=Default register=Select { .0: 240, ap_sel: 0, ap_bank_sel: 15, dp_bank_sel: 0 }
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::ap: Register read succesful ap=0 register="CFG" value=0
-DEBUG AP discovery: probe_rs::architecture::arm::communication_interface: AP GenericAp { address: FullyQualifiedApAddress { dp: Default, ap: V1(0) } }: MemoryAp(MemoryApInformation { address: FullyQualifiedApAddress { dp: Default, ap: V1(0) }, supports_only_32bit_data_size: false, debug_base_address: 3759136768, supports_hnonsec: true, has_large_address_extension: false, has_large_data_extension: false, device_enabled: true })
 DEBUG debug_device_unlock: probe_rs::architecture::arm::sequences: debug_device_unlock - empty by default
```

and more specifically

```diff
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::ap: Register read succesful ap=0 register="CSW" value=192938066
-DEBUG AP discovery: probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: 1, HNONSEC: 1, PROT: 6, CACHE: b, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U8 }
-DEBUG AP discovery:read_ap_register: probe_rs::architecture::arm::ap: Register read succesful ap=0 register="CSW" value=1266679888
-DEBUG AP discovery: probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: 0, HNONSEC: 0, PROT: 0, CACHE: b, SPIDEN: 1, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 1, AddrInc: Single, _RES1: 0, SIZE: U32 }
-DEBUG AP discovery: probe_rs::architecture::arm::communication_interface: HNONSEC supported: true
-DEBUG AP discovery: probe_rs::architecture::arm::communication_interface: Device enabled: true
```

This comes from [this](https://github.com/probe-rs/probe-rs/commit/9eb1c2fe3ae5c836136ea3fc5ff39f5651b976ad#diff-a489871a99b42c40097b5750fae3794f6be7d7de8e740254cfb1c969ae997412L254-L302) piece of code that was removed.

It writes `CSW` and reads it back to check if `U8` data size is supported. This is no longer done because the MemAp types inherently know what they are mandated to support or not. So there’s no need for runtime discovery of that feature.

An unexpected side effect (unexpected to me) is that [`CSW::new`](https://github.com/probe-rs/probe-rs/blob/879c53efb2fecd22aa22d7129ab0262ae867e2fa/probe-rs/src/architecture/arm/ap/memory_ap/mod.rs#L390-L391) sets `DbgSwEnable` and `HNONSEC` to 1 which is different from your device’s default state and causes the behaviour we see here.

ithinuel, 2024-09-08:
@jaredwolff & @felipebalbi would you mind testing that PR and let me know if that solves your issue ?

jaredwolff, 2024-09-08:
Yup. I can try later today.

jaredwolff, 2024-09-08:
As far as I can tell same error:

My cargo.toml

```toml
probe-rs = { git = "https://github.com/ithinuel/probe-rs.git", rev = "7ad674099979225bcf95692b245a2dd0fe619c38" }
```

Trace attached.

[trace.log](https://github.com/user-attachments/files/16922051/trace.log)

I'll be building more boards in the near future. I can likely send you one @ithinuel if that helps.

ithinuel, 2024-09-08:
Thank you @jaredwolff, could you check what your `RUST_LOG` variable contains? It seems to only trace the `nusb` module where I’m more interested in `probe_rs`’s output.

Having a board able to cause the issue would definitely help reproduce and resolve this issue ❤️.

jaredwolff, 2024-09-08:
```
❯ env | rg RUST_LOG
RUST_LOG=trace
```

Do I need to enable a feature or do something else to get the probe-rs library logging going?

Here's the log for running this:

```
probe-rs download --chip nRF9160_xxAA --binary-format hex ../_bin/merged_nrf91x1_g7c2eceb.hex --allow-erase-all --lo
g-file trace-info.log
```

[trace.log](https://github.com/user-attachments/files/16922982/trace.log)

And the output on the console:

```
❯ probe-rs download --chip nRF9160_xxAA --binary-format hex ../_bin/merged_nrf91x1_g7c2eceb.hex --allow-erase-all --lo
g-file trace-info.log
 INFO probe_rs::util::logging: Writing log to "trace-info.log"
 INFO probe_rs::config::target: Using sequence Arm(Nrf9160(()))
 INFO attach: probe_rs::probe::cmsisdap: Using protocol SWD
 INFO probe_rs::architecture::arm::sequences: Debug port Default is powered down, powering up
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Checking if core 0 is unlocked
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Core 0 is already unlocked
 INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None } INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None } INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack              INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002778                                                      ░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }
 INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }
 INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack
 INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002778
 INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                         INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                         INFO probe_rs::util::logging: Wrote log to "trace-info.log"                                                          Error: The flashing procedure failed for '../_bin/merged_nrf91x1_g7c2eceb.hex'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.
    3: Failed to write register DRW at address 0x0c
    4: The debug probe encountered an error.
    5: An error which is specific to the debug probe in use occurred.
    6: CMSIS-DAP responded with an error.
```

jaredwolff, 2024-09-08:
> Having a board able to cause the issue would definitely help reproduce and resolve this issue ❤️.

Email me your address jared@circuitdojo.com :)

ithinuel, 2024-09-09:
Update: I could reproduce the issue using an hs-probe on an LPC55S69 (I can’t get the onboard probe to work).
So I’ll be debugging this ASAP.

ithinuel, 2024-09-09:
There, I had errouneously set `HNONSEC` on the CSW.
Let me know if this new version works :)

jaredwolff, 2024-09-09:
Sorry @ithinuel no dice.

```
❯ probe-rs download --chip nRF9160_xxAA --binary-format hex _bin/merged_nrf91x1_g7c2eceb.hex --allow-erase-all
 INFO probe_rs::config::target: Using sequence Arm(Nrf9160(()))
 INFO attach: probe_rs::probe::cmsisdap: Using protocol SWD
 INFO probe_rs::architecture::arm::sequences: Debug port Default is powered down, powering up
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Checking if core 0 is unlocked
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Core 0 is already unlocked
 INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }       INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }           INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack                                                       INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002778                                                                                                INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }
 INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }
 INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack
 INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002778
 INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                                                                  INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                                                                 Error: The flashing procedure failed for '_bin/merged_nrf91x1_g7c2eceb.hex'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.
    3: Failed to write register DRW at address 0x0c
    4: The debug probe encountered an error.
    5: An error which is specific to the debug probe in use occurred.
    6: CMSIS-DAP responded with an error.
```

[trace.log](https://github.com/user-attachments/files/16935343/trace.log)

felipebalbi, 2024-09-10:
I can still get this to fail with Rust Probe and LPC CMSIS DAP against an RT685S EVK.

Dirbaio, 2024-09-10:
with latest git master including #2802?

felipebalbi, 2024-09-10:
Logs
[trace-failing-6fee4b6c51305efac4f00739c4911be16a321448-jlink-compact-plus.txt](https://github.com/user-attachments/files/16946731/trace-failing-6fee4b6c51305efac4f00739c4911be16a321448-jlink-compact-plus.txt)
[trace-failing-6fee4b6c51305efac4f00739c4911be16a321448-lpc-link2.txt](https://github.com/user-attachments/files/16946732/trace-failing-6fee4b6c51305efac4f00739c4911be16a321448-lpc-link2.txt)
[trace-failing-6fee4b6c51305efac4f00739c4911be16a321448-rusty-probe.txt](https://github.com/user-attachments/files/16946733/trace-failing-6fee4b6c51305efac4f00739c4911be16a321448-rusty-probe.txt)

felipebalbi, 2024-09-10:
> with latest git master including #2802?

Yup:

```console
❯ git show --stat
commit 6fee4b6c51305efac4f00739c4911be16a321448 (HEAD -> master, origin/master, origin/HEAD)
Author: Wilfried Chauveau <wilfried.chauveau@ithinuel.me>
Date:   Tue Sep 10 16:35:33 2024 +0200

    arm: Set DbgSwEnable and some other flags on AP init. (#2802)

 probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb3.rs       | 19 ++++++++++++-------
 probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb5.rs       | 33 ++++++++++++++++++++++++---------
 probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb5_hprot.rs | 47 +++++++++++++++++++++++++++++++++++------------
 probe-rs/src/architecture/arm/ap/memory_ap/amba_apb2_apb3.rs  |  5 +++--
 probe-rs/src/architecture/arm/ap/memory_ap/amba_apb4_apb5.rs  |  9 +++++----
 probe-rs/src/architecture/arm/ap/memory_ap/amba_axi3_axi4.rs  | 12 +++++++-----
 probe-rs/src/architecture/arm/ap/memory_ap/amba_axi5.rs       | 16 +++++++++-------
 7 files changed, 95 insertions(+), 46 deletions(-)
```

jaredwolff, 2024-09-11:
Same here.

```
❯ probe-rs --version
probe-rs 0.24.0 (git commit: v0.1.0-203-g6fee4b6c)
```

```
❯ git show --stat
commit 6fee4b6c51305efac4f00739c4911be16a321448 (HEAD, origin/master, origin/HEAD)
```

```
❯ probe-rs download --chip nRF9160_xxAA --binary-format hex _bin/merged_nrf91x1_g7c2eceb.hex --allow-erase-all --log-file trace.log
 INFO probe_rs::util::logging: Writing log to "trace.log"
 INFO probe_rs::config::target: Using sequence Arm(Nrf9160(()))
 INFO attach: probe_rs::probe::cmsisdap: Using protocol SWD
 INFO probe_rs::architecture::arm::sequences: Debug port Default is powered down, powering up
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Checking if core 0 is unlocked
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Core 0 is already unlocked
 INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }       INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }           INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack                                                       INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002778                                                                                                INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }
 INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20040000, cores: ["main"], access: None }
 INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack
 INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002778
 INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                                                                  INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                                                                  INFO probe_rs::util::logging: Wrote log to "trace.log"                                                                                                        Error: The flashing procedure failed for '_bin/merged_nrf91x1_g7c2eceb.hex'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.
    3: Failed to write register DRW at address 0x0c
    4: The debug probe encountered an error.
    5: An error which is specific to the debug probe in use occurred.
    6: CMSIS-DAP responded with an error.
```

[trace.log](https://github.com/user-attachments/files/16966695/trace.log)

felipebalbi, 2024-10-25:
Coming back to this bug after quite a while. Turns out that if I mark the transaction as secure (by setting `HNONSEC` to false) then everything works fine:

```diff
diff --git a/probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb3.rs b/probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb3.rs
index a84ca270..994a5555 100644
--- a/probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb3.rs
+++ b/probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb3.rs
@@ -31,6 +31,7 @@ impl AmbaAhb3 {
         let me = Self { address, csw, cfg };
         let csw = CSW {
             DbgSwEnable: true,
+            HNONSEC: false,
             MasterType: true,
             Cacheable: true,
             Privileged: true,
```

IMO, we should not be setting transactions as Secure, but it is, indeed, what probe-rs prior to the commit in question did.

ithinuel, 2024-10-25:
hm I though that I had fixed this. 🤔

felipebalbi, 2024-10-25:
> hm I though that I had fixed this. 🤔

The original `supports_hnonsec` check ended up being removed with 9eb1c2f.

felipebalbi, 2024-10-25:
Now I'm confused. According to `probe-rs` traces, this device sports an AHB3:

```console
DEBUG probe_rs::architecture::arm::ap::memory_ap: reading IDR: IDR { REVISION: 8, DESIGNER: JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")), CLASS: MemAp, _RES0: 0, VARIANT: 0, TYPE: AmbaAhb3 }
```

According to section E1.4 of ADIv5, AHB3 does *not* have `HNONSEC` bit, only AHB5 can have it. However, existing code and code prior to 9eb1c2f reference the bit's existence in AHB3. What should I make of all this? Do we have a bug in silicon, in probe-rs or both?

felipebalbi, 2024-10-25:
Sorry, more information here. It turns out that bits 30:24 can also be implementation-defined, according to section 8.2.9 of [ADIv5](https://developer.arm.com/documentation/ihi0031/a/The-Memory-Access-Port--MEM-AP-/MEM-AP-functions/Additional-implementation-defined-features-of-a-MEM-AP?lang=en). Reset value is supposed to select privileged access, which would match `HNONSEC` set to `false`. This would mean that if `probe-rs` were to leave CSW `30:24` unchanged, it should always work. It's getting late, but I'll try to spend some on this during the weekend and next week, unless someone beats me to it.

felipebalbi, 2024-10-26:
Few more details here:

1. ADIv5 claims that bits `30:24` are implementation-defined (see previous comment)
2. ADIv5 also claims that if these bits are used for secure debugging, privileged/secure mode should be the reset value
3. At least MIMXRT685S parts, bit 30 is used as HNONSEC (this is lightly mentioned in their manual), however the bit appears to be `true` (non-privileged) out of reset (see trace below)
4. Setting bit 30 to !SPIDEN results in a stable connection with today's probe-rs HEAD (feb615725585d35f7a0a224dbbcab8433c5ad1e0).

Tracing debug patch:

```diff
diff --git a/probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb3.rs b/probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb3.rs
index 994a55551cdd..5ae8e6a0b096 100644
--- a/probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb3.rs
+++ b/probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb3.rs
@@ -28,10 +28,12 @@ impl AmbaAhb3 {
         let cfg = probe.read_raw_ap_register(&address, super::registers::CFG::ADDRESS)?;
         let (csw, cfg) = (csw.try_into()?, cfg.try_into()?);
 
+        tracing::trace!("===== CSW {:?}", csw);
+
         let me = Self { address, csw, cfg };
         let csw = CSW {
             DbgSwEnable: true,
-            HNONSEC: false,
+            HNONSEC: !csw.SPIDEN,
             MasterType: true,
             Cacheable: true,
             Privileged: true,
@@ -39,6 +41,9 @@ impl AmbaAhb3 {
             AddrInc: AddressIncrement::Single,
             ..me.csw
         };
+
+        tracing::trace!("==== after CSW {:?}", csw);
+
         probe.write_ap_register(&me, csw)?;
         Ok(Self { csw, ..me })
     }
```

Output:

```console
TRACE probe_rs::architecture::arm::ap::memory_ap::amba_ahb3: ===== CSW CSW { DbgSwEnable: false, HNONSEC: false, MasterType: false, Allocate: false, Cacheable: true, Bufferable: false, Privileged: true, Data: true, SPIDEN: true, TrInProg: false, DeviceEn: true, AddrInc: Single, Size: U32, _reserved_bits: 0 }
TRACE probe_rs::architecture::arm::ap::memory_ap::amba_ahb3: ==== after CSW CSW { DbgSwEnable: true, HNONSEC: false, MasterType: true, Allocate: false, Cacheable: true, Bufferable: false, Privileged: true, Data: true, SPIDEN: true, TrInProg: false, DeviceEn: true, AddrInc: Single, Size: U32, _reserved_bits: 0 }
```

I'll send a PR setting `HNONSEC` to `!SPIDEN` for AHB3 and AHB5 as I think that should be a good default.

PR #2875 (2024-10-28): probe-rs: arm: memory_ap: Default HNONSEC to !SPIDEN
Some MCUs appear to have HNONSEC set to non-privileged mode by default. Because ADIv5 calls for HNONSEC to be privileged (that is, false) by default when the bit is implemented, let's default to that for AHB3 and AHB5 interconnects.

Fixes #2788
closed 2024-10-28 by Tiwalun