probe-rs/probe-rs #2789: `--preverify` does not seem to work with RP2040

bugadani, 2024-09-01:
Calling `probe-rs run --preverify --chip rp2040 <path-to-elf>` multiple times with the same .elf reflashes the firmware every time.

d3zd3z, 2025-01-31:
I have just confirmed that this still is the case.

elfmimi, 2025-02-02:
The important question to ask here is, does --verify works correctly or not for you?

elfmimi, 2025-02-03:
verify command, download command with --verify option and download command with --preverify option all fail consistently for rp2040.

this is because activating the flash algo means disconnecting the SPI flash memory from the XIP for rp2040.
so, while it is active the memory region at 0x10000000 will read all zero. hence it is always a verify error.

quick dirty fix will be like this.

```diff
diff --git a/probe-rs/src/flashing/flasher.rs b/probe-rs/src/flashing/flasher.rs
index a50c7770..1735bc4d 100644
--- a/probe-rs/src/flashing/flasher.rs
+++ b/probe-rs/src/flashing/flasher.rs
@@ -244,9 +244,15 @@ impl Flasher {
         F: FnOnce(&mut ActiveFlasher<'_, Verify>) -> Result<T, FlashError> + Sized,
     {
         let mut active = self.init(session, progress, None)?;
-        let r = f(&mut active)?;
-        active.uninit()?;
-        Ok(r)
+        let r;
+        if active.flash_algorithm.pc_verify.is_none() {
+            active.uninit()?;
+            r = f(&mut active);
+        } else {
+            r = f(&mut active);
+            active.uninit()?;
+        }
+        Ok(r?)
     }
 
     pub(super) fn is_chip_erase_supported(&self, session: &Session) -> bool {
```

bugadani, 2025-02-03:
Thanks for the investigation, I'll try and fix this soon-ish.

Yatekii, 2025-02-10:
The real issue is hidden here: https://github.com/rp-rs/flash-algo/blob/main/src/main.rs#L164 essentially we disable XIP (execute in place) to run the firmware from RAM instead of Flash. Unfortunately that makes the Flash invisible to us or at least mapped to a different address. Details can be found in https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf.

I am not sure how to fix this best. I think ideally we implement the flash verify and/or the flash read functions in the flash algo.

devtanc, 2025-05-21:
At a minimum it would be nice to have some sort of output that states that this does not currently work with RP2040 chips, just to help out those in this situation. Took me a few hours to end up here.

bugadani, 2025-05-21:
probe-rs doesn't know this.

devtanc, 2025-05-21:
I'm not really sure what you even mean by that. Yes, it doesn't "know" this, and that is the issue. We DO know this. That's why there is an issue. While it doesn't "know" that the verify does or doesn't work, at least for the moment some output could be added when that chip is selected and verify is selected that notifies the user that verify is not working for this chip. It doesn't need to do anything to early exit or anything, just a quick message to let the user know so that they're not stuck debugging for hours trying to get it to verify.

bugadani, 2025-05-21:
would you go through ~3k chips and filter out which ones need this to be printed?