probe-rs/probe-rs #2800: `probe-rs` master branch doesn't build as a library

xobs, 2024-09-08:
**Describe the bug**
When building a project using `probe-rs` from the master branch, the resulting program doesn't compile.

**To Reproduce**
Steps to reproduce the behavior:

```
mkdir test-probe-rs
cd test-probe-rs
cargo init
cargo add --git https://github.com/probe-rs/probe-rs.git probe-rs
cargo build
```

**Expected behavior**
The user should be able to build from master.

**Additional context**
This may be due to relying on a particular package in `Cargo.lock` that is improperly specified, since a similar breakage happens when building `probe-rs` within the master branch after removing `Cargo.lock`.

xobs, 2024-09-08:
This is due to the `bitfield` crate not being semver-compatible. It is possible to fix `probe-rs` by pinning `bitfield = "=0.16.1"`

xobs, 2024-09-08:
I reported the issue at https://github.com/dzamlo/rust-bitfield/issues/50

xobs, 2024-09-08:
The author of `bitfield` yanked `0.16.2`, so that is now no longer an issue.

As a separate note, `insta` changed between `1.39.0` and `1.40.0` such that it is no longer compatible. The following patch is required:

```patch
diff --git a/probe-rs-tools/src/bin/probe-rs/cmd/dap_server/debug_adapter/dap/repl_commands.rs b/probe-rs-tools/src/bin/probe-rs/cmd/dap_server/debug_adapter/dap/repl_commands.rs
index f7a0a775..41862c7d 100644
--- a/probe-rs-tools/src/bin/probe-rs/cmd/dap_server/debug_adapter/dap/repl_commands.rs
+++ b/probe-rs-tools/src/bin/probe-rs/cmd/dap_server/debug_adapter/dap/repl_commands.rs
@@ -215,7 +215,6 @@ pub(crate) static REPL_COMMANDS: &[ReplCommand<ReplHandler>] = &[
             let yaml_data = insta_yaml::serialize_value(
                 &target_core.core_data.stack_frames,
                 insta_yaml::SerializationFormat::Yaml,
-                insta_yaml::SnapshotLocation::File,
             );

             let response_message = if let Some(location) = write_to_file {
```

However, as this is a command line tool and thus subject to `Cargo.lock`, this is not an issue.

closed 2024-09-08 by xobs

Yatekii, 2024-09-08:
Maybe it's time to get rid of that dep that breaks us every other day.

chrysn, 2024-09-09:
Well, the big appears through insta_yaml imported from insta::_macro_support -- looks like the authors are applying Python convention on things being private, and have items that are technically pub but hidden from docs (and underscore-prefixed to scare off people). It's in the fringe zones of the stability guarantee, and chances are this won't break as often if insta_yaml is not used any more. (I have no clue *how precisely* it is being used, just grepped around, but quite possibly it is easier to get rid of the access to kind-of-private members than to get rid of the dependency as a whole).

jaredwolff, 2024-09-18:
Still fails to build on master as of today

```
error[E0061]: this function takes 2 arguments but 3 arguments were supplied
   --> probe-rs-tools/src/bin/probe-rs/cmd/dap_server/debug_adapter/dap/repl_commands.rs:215:29
    |
215 |               let yaml_data = insta_yaml::serialize_value(
    |                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
216 |                   &target_core.core_data.stack_frames,
217 |                   insta_yaml::SerializationFormat::Yaml,
    |  ______________________________________________________-
218 | |                 insta_yaml::SnapshotLocation::File,
    | |                 ----------------------------------
    | |_________________|________________________________|
    |                   |                                help: remove the extra argument
    |                   unexpected argument of type `SnapshotLocation`
    |
note: function defined here
   --> /home/jaredwolff/.cargo/registry/src/index.crates.io-6f17d22bba15001f/insta-1.40.0/src/serialization.rs:91:8
```

Tiwalun, 2024-09-18:
You need to use `--locked`, then it works. The `insta` issue should be fixed in https://github.com/probe-rs/probe-rs/pull/2814, but in general issues like this can always happen if the lock file is not used. There is no guarantee that all dependencies follow semver all the time.