probe-rs/probe-rs #2801: FR: Ability to exit back to host's command prompt (exit or similar, in the embedded)

lure23, 2024-09-08:
**Is your feature request related to a problem? Please describe.**

I write examples for a sensor code, running on an embedded board. These take a few seconds to run, and I'd like to return control to the host, for example so that a new example can be run.

I cannot see how this is possible. What I think I'd want is an `exit()` feature in `defmt` (which I use for logging), and then `probe-rs` to finish its process, in a controlled manner.

Am I missing something?

**Describe the solution you'd like**

Ability to return control to the host command line, once a batch-processing-like embedded program is done.

**Describe alternatives you've considered**

What I currently do: 

```
info!("End of demo");
loop {}
```

+ expect the user to press `Ctrl-C`.

I don't know of a way to actually work-around the issue (i.e. have embedded execution, or at least the `probe-rs` following it, stop).

**Additional context**

Source to my [example](https://github.com/lure23/VL53L5CX_rust/blob/main/vl53l5cx_uld/examples/matrix.rs#L142-L146).

bugadani, 2024-09-08:
Please take a look at https://docs.rs/semihosting/0.1.14/semihosting/process/fn.exit.html and let us know if it's what you're looking for.

lure23, 2024-09-08:
It is. This explains why I had seen a mention to `exit()` somewhere, but now need to read about semihosting, so I know whether it's something I'd be comfortable adding in the mix. Thanks! :)

lure23, 2024-09-08:
Beautiful! I think this was it.

Now, should I close this issue?  Since I was aware of semihosting (hadn't tried it), but unable to connect the dots that it would have been useful with `probe-rs`, is there anything we can do to help people understand that the two are not mutually exclusive (as I had mistaken them to be)?

Yatekii, 2024-09-09:
The webpage could use some documentation entry: https://github.com/probe-rs/webpage :)

lure23, 2024-09-09:
> The webpage could use some documentation entry: https://github.com/probe-rs/webpage :)

I think so, too.

I'll make a brief example below, but could you guys turn it into a PR on the docs. It probably could start under FAQ and be moved to Knowledge Base if there's later more details. Your call. I feel uncertain about doing a doc PR directly, since I've been using `probe-rs` only a couple of months now. I don't want to damage the nice, concise documentation you have going on. :)

I would appreciate:

- a mention to `exit` so that it gives a search hit if one looks for it, specifically (I did)
- brief mention of the relation between `probe-rs` and `semihosting`, so one understands certain features (of `probe-rs`, as a tool) are available only if one brings in another library

Furthermore, as an observation, in the `semihosting` site, `probe-rs` is [barely mentioned](https://docs.rs/semihosting/0.1.14/semihosting), only under the [`openocd-semihosting`](https://docs.rs/semihosting/0.1.14/semihosting/#optional-features-openocd-semihosting) section ("used in OpenOCD and probe-rs.")

I don't mind that; it doesn't need to help. But for a firstcomer, there's now also OpenOCD thrown into the mix. Brief clarity would be purposeful! :)

So. Brief clarity, eh=? How about:

---

`probe-rs` implements the semihosting protocol, allowing you to merge embedded tasks with your development OS, to a degree. This is handy e.g. for making short-lived examples, and/or tests.

The target-side support is in the separate [`semihosting`](https://docs.rs/semihosting/latest/semihosting/) crate. Bring it in and now you can:

- `exit()`, `abort()`
- `fs` - write files to the host filesystem

Target-specific:

- RISC V: |just works
- Xtensa: enable the [`openocd-semihosting`](https://docs.rs/semihosting/0.1.14/semihosting/#optional-features-openocd-semihosting) feature

More info: The Embedded Rust Book > [Semihosting](https://doc.rust-lang.org/stable/embedded-book/start/semihosting.html)

---

In addition, I leave it to your consideration, whether a mention that `probe-rs` implements semihosting would be valuable here: https://doc.rust-lang.org/stable/embedded-book/start/semihosting.html

lure23, 2024-09-09:
Noticed today that the (experimental) `args` feature of `semihosting` is not supported by `probe-rs`:

```
$ probe-rs run target/riscv32imc-unknown-none-elf/release/examples/set-address
[...]
 WARN probe_rs::cmd::run::normal_run_mode: Target wanted to run semihosting operation SYS_GET_CMDLINE, but probe-rs does not support this operation yet. Continuing...
 WARN probe_rs::cmd::run::normal_run_mode: Target wanted to run semihosting operation 0x13 with parameter 0x0,but probe-rs does not support this operation yet. Continuing...
```

This is of course okay, but would be nice to be mentioned alongside the above text. As a developer, I would like to see a clear listing on what I can expect to work.

lure23, 2024-12-03:
Thx. Closing because I'm pretty pleased with existing functionality.

closed 2024-12-03 by lure23