probe-rs/probe-rs #2803: Bash completions fail to install

rmsc, 2024-09-08:
**Describe the bug**

When attempting to install the bash completions, the Zsh installer is used by mistake. The fix is quite simple, PR on the way.

**To Reproduce**
Steps to reproduce the behavior:

1. $ probe-rs complete install

```
(...)
Writing the autocompletion script failed
Please write the above script to /home/<user>/.zfunc/probe-rs.bash manually
Error: Writing th script failed
(...)
```

**Expected behavior**

The installer should install to `~/.bash_completion` instead of `~/.zfunc/`.

rmsc, 2024-09-08:
It seems like the path where the completions are installed to is also wrong, at least they don't seem to be picked up automatically by `bash-completion`.

The path recommended in the rustup docs seems to work (`.local/share/bash-completion/completions/`). Will update the PR.

rmsc, 2024-12-13:
Ping?

Tiwalun, 2024-12-18:
Is this fixed with the merged PR?

rmsc, 2024-12-23:
> Is this fixed with the merged PR?

Yes, sorry!

closed 2024-12-23 by rmsc