probe-rs/probe-rs #2816: Suspecting `probe-rs` installed from GitHub `main` doesn't always output the commit checksum properly

lure23, 2024-09-16:
**Describe the bug**

>This is minor, but I wanted to mention it.

Having installed `probe-rs` with a GitHub URL, like this:

```
$ cargo install probe-rs-tools --git https://github.com/probe-rs/probe-rs --locked --force
```

.. I would always expect to see the commit id when running `--version`. This gives me confidence that the install actually happened from GitHub - and not a `crates.io` snapshot.

```
$ probe-rs --version
probe-rs 0.24.0 (git commit: crates.io)
```

Whereas on my other system (same GitHub installation command; WSL2, also Ubuntu):

```
$ probe-rs --version
probe-rs 0.24.0 (git commit: 0fb93950)
```

**To Reproduce**

Steps to reproduce the behavior:

1. Install `probe-rs` with the command above
2. `probe-rs --version | grep crates.io`

Since on my two systems, the behaviour is different, I cannot say how yours turns out.

**Expected behavior**

`probe-rs --version` (when installed from GitHub) always shows a commit id.

**Desktop (please complete the following information):**

- Ubuntu 24.04.1 LTS (under Multipass)
- Ubuntu 22.04.5 LTS (under WSL2)

Tiwalun, 2024-09-17:
I think this could just be an issue with how the git revision is shown in the version output. It seems that we fall back to showing `crates.io`if we can't get the git revision:

https://github.com/probe-rs/probe-rs/blob/33bf231572b8ccc98bf7c988b42e094412608656/probe-rs-tools/build.rs#L10

lure23, 2024-09-17:
>I think this could just be an issue with how the git revision is shown in the version output. It seems that we fall back to showing crates.ioif we can't get the git revision:

Do you have confirmation on that, or some way I can see (other than `--version`) where the actual command came from. 

I also thought that, but reality seemed to show it really was not-the-commit-I-wanted. I'll run some tests; will be right back.. :)

lure23, 2024-09-17:
>These macros do not depend on libgit, but simply uses the git binary directly. So you must have git installed somewhere in your PATH.

https://crates.io/crates/git-version

So that's it. Seems too much hassle for me to start changing. Should this be mentioned somewhere, since it matters to `probe-rs` users but is currently just a note in a build-dependency's `README`?

Tiwalun, 2024-09-17:
I think the fallback is misleading, something like `unknown` would probably be better, and would be low effort to fix.

Failing a build because `git` isn't installed would also be problematic. There might be other ways to get the git rev which work without an installed `git` command, we would certainly accept a PR to improve this.

lure23, 2024-09-17:
`unknown` would be fine, imho

bugadani, 2024-09-17:
If you `cargo install` probe-rs (you shouldn't) on a machine that doesn't have git, you'll get `crates.io` in the version string. We don't want to change that to `unknown` and we can't really tell this case apart from when you check out the sources and build it yourself on a similar setup.

bugadani, 2024-09-17:
Eh, whatever. Technically from crates.io we can figure out the git rev from the tag, but I guess if we really can't tell where probe-rs was installed from it's an error% either way.

lure23, 2024-09-18:
Since I brought this up... my only concern is that I find the current behaviour misleading. A C3 issue was more confusing to me than it needed to, since I didn't see the correct version trail.

Anyhow, from my side this is gone, since I now know to have `git` in my VM's, if I use `probe-rs` from a revision. I'll move on to check whether the actual C3 problems are fixed in `main`.

Note: Also happy to close this without changes to the version process. This issue remains as a history log for someone perhsps (or not) struggling with the same? :)

closed 2024-09-18 by lure23