probe-rs/probe-rs #2817: Support external flash loaders

florianhumblot, 2024-09-17:
**Is your feature request related to a problem? Please describe.**
Apart from flashing a device's internal flash, I often have to flash an SPI chip connected to the target MCU. The easiest way of doing this (at least for my STM32 devices) is by using an external flash loader, which gets put into RAM by the ST tools, which can then program, erase, verify, etc.. the external flash memory.

It would be nice if probe-rs supported these kinds of external flash loaders so that programmatic access to a device's auxiliary memory could be made possible for things such as on-device testing.

**Describe the solution you'd like**
Support loading external flash loaders, like the  `STM32_Programmer_CLI` or `STM32CubeProgrammer` can.

**Describe alternatives you've considered**
I can (and currently do) just use multiple tools. This is annoying because CubeProgrammer wants sole ownership over the device, so I have to keep connecting/disconnecting in the GUI and I sometimes also have to mount/unmount my ST-Link when using USBIP to share the connection between WSL and Windows.

bugadani, 2024-09-17:
> It would be nice if probe-rs supported these kinds of external flash loaders

It does, you just have to write them for your target.

florianhumblot, 2024-09-17:
> > It would be nice if probe-rs supported these kinds of external flash loaders
> 
> It does, you just have to write them for your target.

Is there a way to pass it an existing one? Or do I need to write one from scratch?

bugadani, 2024-09-17:
You need one that implements the CMSIS-PACK flash algorithm standard, and you need to package it up as part of a target description yaml. Whether you have an existing one or not, depends on your target. Writing a flash loader is a bit involved, because using external flash you most likely can't use `target-gen test` to easily try what you have.

pkoevesdi, 2024-12-01:
I think, this is currently supported, for instance You can do:
`target-gen elf -u /usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin/ExternalLoader/MT25TL01G_STM32H745I-DISCO.stldr TARGET.yaml`

But the problem is, I get 
```
Error: Failed to extract flash information from ELF file '/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin/ExternalLoader/MT25TL01G_STM32H745I-DISCO.stldr'.

Caused by:
    Failed to find 'FlashDevice' symbol in ELF file.
```
for every stm-ExternalLoader I tried.
When doing `objdump -t /usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin/ExternalLoader/MT25TL01G_STM32H745I-DISCO.stldr` it's proven right, there's no symbol `FlashDevice` in it. So, they might not be written after [CMSIS spec](https://www.keil.com/pack/doc/CMSIS_Dev/Pack/html/flashAlgorithm.html). 

Or does anyone else have another explanation or a workaround? It would save a huge amount of work for so many people, if we could use the STM library of External Loaders as flash algorithms in probe-rs.

See also my similar question [here](https://users.rust-lang.org/t/create-own-flash-algorithm-from-binary/108885?u=pkoevesdi), but it stayed unanswered concerning the missing `FlashDevice` symbol.

Tiwalun, 2024-12-01:
@pkoevesdi as explained before, we currently support flash loaders which follow the CMSIS-PACK standard. The algorithms from ST probably use their own standard. If that is documented, then it should be possible to add support for it.

The first step here would be to figure out if there is any documentation for the algorithms that ST uses.

pkoevesdi, 2024-12-01:
It is shown here: https://community.st.com/t5/stm32-mcus/how-to-implement-and-use-your-own-external-flash-loader-an/ta-p/597752 there's also some `CMSIS` include.
There are also tutorials from STM, for instance [here](https://www.st.com/content/st_com/en/support/learning/stm32-education/stm32-moocs/external_QSPI_loader.html).
And for instance [here](https://github.com/STMicroelectronics/stm32-external-loader/tree/main/STM32H7x_boards/MT25TL01G_STM32H747I-DISCO)'s the source of the above external loader.

Would be nice, if someone competent (not me) could look through this, and first confirm, that it's not CMSIS compatible, beside a include namend CMSIS, and secondly maybe propose some minor changes to be so?

Yatekii, 2024-12-03:
While this might be the worst piece of code & documentation I have seen yet, it seems to work similarly to the CMSIS-DAP spec albeit not the same.

- https://github.com/STMicroelectronics/stm32-external-loader/blob/main/STM32F4x_boards/MT25QL512A_STM32469I-EVAL/Flash_loader/Src/Dev_Inf.c#L28 this StorageInfo is the same as the FlashDevice symbol.
- Their functions are named differently but seem to be well defined here: https://github.com/STMicroelectronics/stm32-external-loader/blob/main/STM32F4x_boards/MT25QL512A_STM32469I-EVAL/Flash_loader/Src/Loader_Src.c#L47 so we can easily find those symbols in a binary and call them by putting the correct args on the stack and reading them back upon completion.

pkoevesdi, 2024-12-04:
Great findings, thank you! 
I don't really know much about it, but from a user's side and to make probe-rs universally usable and future proof, I could imagine that it makes sense to make the expected symbol and function names configurable as parameters into `target-gen elf`.
And we could then publish a working set of these parameters for ExternalLoaders of STM, and another set for algo-x from whoever-y, ...

Other option I'd throw in: https://github.com/ecnx/symrename or something similar - would it be enough to rename symbols inside the elf?

Yatekii, 2024-12-04:
I dont think the calling convention is exactly the same. So I believe renaming wont cut it.

kevswims, 2025-07-31:
For folks targeting STM32 parts, I have put together an example of a flash algorithm for an STM32H7RS MCU here: https://github.com/kevswims/stm32h7-flash-example 

This doesn't directly address being able to use the ST flash algorithms out of the box but it does provide a reasonable path forward for STM32 parts and doesn't require ever opening CubeIDE.