probe-rs/probe-rs #2818: ESP32-C3 bug recurs on latest `main`

lure23, 2024-09-18:
**Describe the bug**

This is something that affects my running of an I2C code. I've tracked it down to `probe-rs` versions (and `defmt` logging). Depending on which commit (of `probe-rs`) I use, I repeatedly get the following results:

|version / commit|status|date|
|---|---|---|
|`33bf231`|fail|Sep 17th (current `HEAD`)|
|...|fail|*will report in detail which is the first commit that fails*|
|[`0fb9395`](https://github.com/probe-rs/probe-rs/tree/0fb93950ed1e4ff507b6dd472d4634f0a9878ac8)|OK|Aug 25th|
|...|
|`0.24.0 crates.io`|fail|May 23rd|

The above table has been constructed by:

- installing `probe-rs` via:

   ```
  $ cargo install [--git https://github.com/probe-rs/probe-rs [--rev xxx]] probe-rs-tools --locked --force
    ```

- detaching and re-attaching the device (USB level), to make sure it's not left in a bad state
- running an example (long I2C blocks; see below)

**To Reproduce**

Steps to reproduce the behavior:

It's rather elaborate (sorry!); don't have a minimum repeatable case extracted. Also, it would need access to `vl53l5cx` time-of-flight sensor -- or something else that takes long I2C blocks.

The problem in question happens when loading the firmware to the sensor. This is a 80k download taking place in 252 byte (+ 2 index bytes) chunks. When failing, I get `TimedOut` response from the `esp-hal` I2C library. But it's neither the code, nor the library that causes this. It *only* happens on ESP32-C3, and depends on `probe-rs` revisions.

1. We assume you have access to ESP32-C3 and VL53L5CX SATEL board
1. clone `https://github.com/lure23/VL53L5CX_rust`
2. set up dependencies (and/or a VM)
3. `cd vl53l5cx_uld`
4. `make -f Makefile.2 z`

**Expected behavior**

Sensor found; firmware loads; results reported!: 

```
probe-rs run --speed=200 --log-format '{t:dimmed} [{L:bold}] {s}' target/riscv32imc-unknown-none-elf/release/examples/multiboard
      Erasing ✔ [00:00:04] [##########################] 256.00 KiB/256.00 KiB @ 59.43 KiB/s (eta 0s )
  Programming ✔ [00:00:41] [#################################] 105.54 KiB/105.54 KiB @ 2.53 KiB/s (eta 0s )
   Finished in 41.73853s
0.812560 [INFO ] Target powered off and on again.
0.816281 [DEBUG] Ping succeeded: 0xf0,0x02
3.557625 [INFO ] Init succeeded, driver version VL53L5CX_2.0.0
...

4.786394 [INFO ] Data #9 (sensor 33°C)
4.786427 [INFO ] .target_status:    [[[Valid(5), Valid(5), Valid(5), Valid(5)], [Valid(5), Valid(5), Valid(5), Valid(5)], [Valid(5), Valid(5), Valid(5), Valid(5)], [Valid(5), Valid(5), HalfValid(9), Valid(5)]]]
4.786545 [INFO ] .targets_detected: [[1, 1, 1, 1], [1, 1, 1, 1], [1, 1, 1, 1], [1, 1, 1, 1]]
4.786633 [INFO ] .distance_mm:      [[[1795, 1803, 21, 17], [1792, 1803, 22, 16], [1775, 1796, 20, 18], [1780, 1779, 1606, 13]]]
4.786747 [INFO ] End of ULD demo
```

*A clear and concise description of what you expected to happen.*

**Actual behaviour (when fails)**

```
probe-rs run --speed=200 --log-format '{t:dimmed} [{L:bold}] {s}' target/riscv32imc-unknown-none-elf/release/examples/multiboard
      Erasing ✔ [00:00:03] [##################################################################################################################] 256.00 KiB/256.00 KiB @ 71.87 KiB/s (eta 0s )
  Programming ✔ [00:00:42] [###################################################################################################################] 105.54 KiB/105.54 KiB @ 2.47 KiB/s (eta 0s )    Finished in 42.78763s
0.847639 [INFO ] Target powered off and on again.
0.851361 [DEBUG] Ping succeeded: 0xf0,0x02
1.314871 [ERROR] panicked at 'I2C write to 0x00fc (252 bytes) failed: TimeOut'
```

**Desktop (please complete the following information):**

- Linux Ubuntu

**Additional context**

An earlier me noted (Aug'24) that [this PR](https://github.com/probe-rs/probe-rs/pull/2748) would have removed the C3 problems I was facing then. So let's think that's the "level zero" when a fix was introduced.  `1c79838`

I recognize `0fb9395` (Aug 25th) as a "known good" commit that I have been using. 

That leaves 16 commits between `0fb9395` and now (`33bf231`). Somewhere out there the influence on long I2C writes on ESP32-C3 has been re-introduced.

bugadani, 2024-09-18:
Not surprising, and unless we figure out why some ESP32-C3s need to be halted, this won't be fixed. Avoid defmt if you have timing sensitive code.

lure23, 2024-09-18:
@bugadani Thanks for the clear description.

I am okay leaving it as-is. I have a C6 and can develop the application on that.

Should we now.. close this issue as "wont-fix" or keep it open?

bugadani, 2024-09-18:
Let's keep it, it's a problem, just a hard one.

closed 2026-01-12 by Yatekii