probe-rs/probe-rs #2826: Defmt logs not showing up after jump to application

lees19, 2024-09-25:
**Is your feature request related to a problem? Please describe.**
Logs with defmt seem to show up right until the jump to the application, but none of the application logs seem to show. 

**Describe the solution you'd like**
Logs from the application after the jump. 

**Describe alternatives you've considered**
By hard coding the paths to the elfs of the bootloader and application and creating an RttClient for each elf and then polling both every iteration seems to print the correct logs, but I am not sure simply adding more RttClients 

**Additional context**
Chip: STM32F411CEUx

[Alternative solution mentioned above, will have to set paths to boot/app elfs when reproducing](https://github.com/lees19/probe-rs) 

[Example bootloader jump here](https://github.com/lees19/logging-after-jump)

Yatekii, 2025-02-22:
This is actually an interesting usecase. We would need to attach to two different RTT blocks. This should easily be possible with two (or more) clients indeed.
I am not sure how a possible config interface would look like. Maybe @probe-rs/core has some input :)

Yatekii, 2025-02-22:
Can you follow this route here: https://github.com/probe-rs/rtt-target/pull/59? You can just link to the same spot from both application and bootloader and then you should be able to use the same buffer after boot :)

darkwater, 2025-02-22:
You'd still need to decode from two different ELF binaries, and actually there might not even be a way to tell which one you need if both RTT blocks are at the same location, there would need to be some kind of identifier.

Yatekii, 2025-02-22:
Oh, I guess it was a bit too early in the morning :D It would be cool if one could tell defmt somehow the start of IDs as well so nothing is duplicate.