probe-rs/probe-rs #2840: flash algorithm failed with code 1 on ESP32-C6 if flashing to 0xC10000, works fine up to 0x1B0000

dragonnn, 2024-10-01:
**Describe the bug**
When trying to flash an image on ESP32-C6 16MB to a custom address using `--base-address` option if the address is 0xC10000 it fails with `flash algorithm failed with code 1`

**To Reproduce**
Steps to reproduce the behavior:

1. Run `probe-rs download --chip esp32c6 --binary-format=bin update.bin --base-address=0xC10000`
2. To compare run for example `probe-rs download --chip esp32c6 --binary-format=bin update.bin --base-address=0x010000` this works fine
3. From my tests it starts breaking at 0x1C0000, 0x1B0000 works fine

**Expected behavior**
Flashing should end successfully. 

**Stacktrace**
[probe-rs.log.zip](https://github.com/user-attachments/files/17203255/probe-rs.log.zip)
Full stack trace when running probe-rs

**Desktop (please complete the following information):**

- Linux (Arch)

**Additional context**
Using espflash to write to 0xC10000 works fine:
```
espflash write-bin -c esp32c6 0xC10000 update.bin -p /dev/ttyACM0                                                                                                                                                      ─╯
[2024-10-01T10:09:31Z INFO ] 🚀 A new version of espflash is available: v3.1.1
[2024-10-01T10:09:31Z INFO ] Serial port: '/dev/ttyACM0'
[2024-10-01T10:09:31Z INFO ] Connecting...
[2024-10-01T10:09:31Z INFO ] Using flash stub
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:4b:14:b8
[00:00:02] [========================================]     183/183     0xC10000                                                                                                                                              [2024-10-01T10:09:34Z INFO ] Binary successfully written to flash!
```
Same goes for esptool.py:
```
esptool.py --after no_reset write_flash 0x00c10000 update.bin                                                                                                                                              ─╯
esptool.py v4.7.0

Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C6
Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 40:4c:ca:ff:fe:4b:14:b8
BASE MAC: 40:4c:ca:4b:14:b8
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00c10000 to 0x00c5efff...
Compressed 319600 bytes to 186940...
Wrote 319600 bytes (186940 compressed) at 0x00c10000 in 2.1 seconds (effective 1194.3 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
```
I was wondering if flash_size in IdfLoader is detect right but I did verify that probe-rs does pick up `FlashSize::_16Mb` with should be fine.

bugadani, 2024-10-01:
> From my tests it starts breaking at 0x1C0000, 0x1B0000 works fine

This is a good hint, thanks for including it.

dragonnn, 2024-11-28:
I tested it now again with https://github.com/probe-rs/probe-rs/pull/2925 and now it works fine! Thanks, great work :)

closed 2024-11-28 by dragonnn