probe-rs/probe-rs #2845: embed config file: Allow override from command line

chrysn, 2024-10-04:
**Is your feature request related to a problem? Please describe.**

When an application uses a larger framework for building and/or the Embed.toml content is to some extent invocation dependent, the Embedded.toml needs to be generated at a static location, whereas all other relevant paths (in particular the Cargo target directory) can be bent away, eg. from persistent disk to a RAM disk.

**Describe the solution you'd like**

Providing the path of the Embed.toml file in an argument (or an environment variable) would allow .

**Describe alternatives you've considered**

* Passing all the content of the Embed.yaml on the command line would in theory be nice (and consistent with how we pass programmer and chip to `probe-rs run`), but easily gets out of hand in terms of size.
* The Embed.local.yaml can be overwritten, which will not show up as a dirty git repository state when gitignored. However, with .

**Additional context**

I'm using cargo-embed from RIOT-rs. Its build system is about to be configured with different actions that, for example, may run the application with or without RTT.

**Next steps**

Unless people scream "no, bad approach" (or have a preference for environment variable vs. command-line argument), I'd start sketching a PR for this soon-ish.

chrysn, 2024-10-04:
A concrete example of generated content is when the build system starts a TCP listener on a random port, and from that generates the socket entry of a default.rtt.up_channel item.

bugadani, 2024-10-04:
I'm generally not opposed to `cargo embed --config <path>` or similar if you really need it.

> The Embed.local.yaml can be overwritten, which will not show up as a dirty git repository state when gitignored. However, with .

I think you might have an unfinished thought there :) And it makes me wonder what that caveat is.

Wouldn't [profiles](https://probe.rs/docs/tools/cargo-embed/#profiles) be sufficient for your use case?

chrysn, 2024-10-04:
> I think you might have an unfinished thought there :) And it makes me wonder what that caveat is.

That's what I get from doing too many things at the same time :-)

Should have been

* The Embed.local.yaml can be overwritten, which will not show up as a dirty git repository state when gitignored. However, when the same application is built and run on different boards at the same time -- which is something that is common on RIOT OS when assessing portability, and I expect to do it a lot in RIOT-rs too. That probably works when flashed one after another (`cargo embed` won't change its behavior once running just because its Embed.toml changes), but as our build system prides itself in parallel building of multiple variations of an application, and we plan hardware-in-the-loop testing, eventually we'll run into race conditions.

Profiles would help to some extent, but we'd still be placing Embed.local.toml in a build system where every other generated file is in an easy-to-clean-up place, we'd be appending a profile to the file for every invocation, so that's not too pretty either.

closed 2024-10-16 by bugadani

DanielJoyce, 2024-12-23:
This seems to have broken the --config option for cargo_embed

```
❯ cargo embed --config log
error: unexpected argument '--config' found

  tip: a similar argument exists: '--config-file'

```

DanielJoyce, 2024-12-23:
Ahhh, the documentation is out of date.

`cargo embed CONFIG_NAME` works. the --config flag is not needed,

https://probe.rs/docs/tools/cargo-embed/

> [Profiles](https://probe.rs/docs/tools/cargo-embed/#profiles)
The data in the Embed.toml is structured in two levels: The outer layer is a configuration profile name, inside each profile there are then a series of sections with different options. The default profile is called "default" ;) When invoking cargo-embed, a different profile name can be passed as an argument with --config <profile>, which will load the settings under that profile instead of default.

The docs are out of date.

chrysn, 2024-12-24:
That documentation is not part of the `probe-rs` sources as far as I can grep. AIU this is now being tracked in https://github.com/probe-rs/probe-rs/issues/2985, with nothing more to do around this issue here.