probe-rs/probe-rs #2848: Failed to open the debug probe.: The debug probe could not be created.: A USB error occurred.: Could not determine driver for interface

Azorak-Aynir, 2024-10-05:
Hey, I'm new to embedded programming. 
I tried to get a basic blinking led program written with the Embassy framework running on my STM32F401RE. 
It didn't work out :(   
>> The error I got is exposed in the title and json below. 
```json
{
  "meta": {
    "version": "0.24.0",
    "commit": "0.24.0 (git commit: crates.io)",
    "arch": "x86_64",
    "os": "windows"
  },
  "command": [
    "C:\\Users\\User\\.cargo\\bin\\probe-rs.exe",
    "run",
    "--chip",
    "STM32F401RETx",
    "target\\thumbv7em-none-eabihf\\debug\\blinkled_rust",
    "--report"
  ],
  "elf": "target\\thumbv7em-none-eabihf\\debug\\blinkled_rust",
  "log": "C:\\Users\\User\\AppData\\Roaming\\probe-rs\\probe-rs\\data\\1728095215526.log",
  "error": "Failed to open the debug probe.\n\nCaused by:\n    0: The debug probe could not be created.\n    1: A USB error occurred.\n    2: Could not determine driver for interface"
}
```

When I run ````probe-rs list```` in the terminal I get this:
```The following debug probes were found:
[0]: STLink V2-1 -- 0483:374b:066EFF575755846687183908 (ST-LINK)
```
Any other subcommands fails (info, ...) cargo flash and embed also fails. All fail with the same error.

I'm thinking that my problem might be because it is a STLink 2-1, and that it is not a supported probe ? (I only see STLink V2 and V3 supported in the documentation). 
Is this it ? If yes are there any plans to support it ? If not is there anything else I should try out or check ? Am I missing something ?

Thank you for your time :)

Azorak-Aynir, 2024-10-05:
I successfully uploaded and run a blinky C++ program via the keil studio mbed ide. So it shouldn't be a problem with my hardware at the very least.

kevinmehall, 2024-10-05:
(I'm the developer of nusb, the library used by probe-rs returning that error message, but not a probe-rs developer)

Your device is listed [here](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/stlink/usb_interface.rs#L23), and it's not the one with the comment "Unproven", so that seems like a good start.

The error means that it thinks the device has the Composite Group Parent driver installed to handle the device's interfaces separately, and the interface probe-rs is trying to use has no driver installed.  It's possible that the driver is not installed correctly, or that probe-rs is choosing the wrong interface of the device.

You could try reinstalling the ST driver package, or use [Zadig](https://zadig.akeo.ie/) to install  WinUSB, either for the whole device, or interface 0.

But before you do, if you have a few minutes to troubleshoot, I'd like to see its current driver configuration, because there clearly is a driver installed if it's working with another app, and it looks like the ST installer should be installing WinUSB. If so, nusb and probe-rs would ideally be able to find and use that driver without extra configuration.

In Device Manager, switch to "Devices by connection" on the View menu, and find the device, likely under a few levels of PCI Express, then under a USB Controller. There will be an entry representing the device as a whole, possibly labeled "USB Composite Device", with interfaces underneath it. For each of the interfaces, open its Properties, and on the Details tab, note the Service and Hardware IDs values.

Azorak-Aynir, 2024-10-05:
Thanks for the quick review and help @kevinmehall !

Here is the info you wanted (under the USB Composite Device) about the interfaces, in order:

-  ST-Link Debug
( 
Service values: not listed in dropdown property menu , 
Hardware IDs values: 
USB\VID_0483&PID_374B&REV_0100&MI_00 
USB\VID_0483&PID_374B&REV_0100&MI_00 
)
-  USB Mass Storage Device 
( 
Service values: USBSTOR, 
Hardware IDs values:
USB\VID_0483&PID_374B&REV_0100&MI_01 
USB\VID_0483&PID_374B&MI_01 
)
- Listed beneath the USB Mass Storage: MBED microcontroller USB Device 
( 
Service values: disk, 
Hardware IDs values: 
USBSTOR\DiskMBED____microcontroller_1.0_
USBSTOR\DiskMBED____microcontroller_
USBSTOR\DiskMBED____
USBSTOR\MBED____microcontroller_1
MBED____microcontroller_1
USBSTOR\GenDisk
GenDisk
)
- USB Serial Device (COM7)
(
Service values: usbser,
Hardware IDs values: 
USB\VID_0483&PID_374B&REV_0100&MI_02
USB\VID_0483&PID_374B&MI_02
)

Azorak-Aynir, 2024-10-05:
I'll install WinUSB with Zadig to see if that solves the problem **a bit later**, this way things stay the same if you need any other information to help you troubleshoot.

kevinmehall, 2024-10-05:
Ok, the interface 0 that probe-rs is trying to use does indeed seem like the right interface, and the lack of the Service value is why nusb thinks there is no driver. What does the Driver Details dialog from the Driver tab on that "ST-Link Debug" interface say?

Azorak-Aynir, 2024-10-05:
No drivers files required or loaded for this device.

The St-Link is the only one whose status says it's got a problem: 

The drivers for this device are not installed. (Code 28)
There are no compatible drivers for this device.
To find a driver for this device, click Update Driver.

Didn't find a driver with the windows tool.

Trying this https://www.st.com/en/development-tools/stsw-link009.html?dl=redirect

Azorak-Aynir, 2024-10-05:
Ok, St-Link Device says everything is good in device manager

kevinmehall, 2024-10-05:
My main question was how the Keil software was successfully accessing the device, but given that it had no driver, I have to assume it's using that Mass Storage interface instead of the debugger.

Hopefully that works with WinUSB installed, either via the ST tool or Zadig.

Azorak-Aynir, 2024-10-05:
Yep it work now the rust embassy blinky runs and all probe-rs commands work.

Azorak-Aynir, 2024-10-05:
Thanks for the help! closing the issue now !

closed 2024-10-05 by Azorak-Aynir