probe-rs/probe-rs #2849: STM334F334R8T6 failed to flash

Randommist, 2024-10-05:
I'm trying to flash a stm32nucleo board via probe-rs using
``` sudo probe-rs run --chip STM32F334R8T6 target/thumbv7m-none-eabi/release/rust_embeded_stm32```
the output I get is
``` 
WARN probe_rs::config::registry: Matching STM32F334R8T6 based on wildcard. Consider specifying the chip as STM32F334R8Tx instead.
WARN probe_rs::util::rtt: No RTT header info was present in the ELF file. Does your firmware run RTT?
      Erasing ✔ [00:00:06] [----------------------------------------------------------------------------------------------]      0 B/2.00 KiB @      0 B/s (eta 0s )
  Programming ✔ [00:00:06] [----------------------------------------------------------------------------------------------]      0 B/     0 B @      0 B/s (eta 0s )
Error: The flashing procedure failed for 'target/thumbv7m-none-eabi/release/rust_embeded_stm32'.

Caused by:
    0: Failed to erase flash sector at address 0x08000000.
    1: Something during the interaction with the core went wrong
    2: A timeout occurred.
```

Yatekii, 2025-02-22:
Is this still happening on master?

Randommist, 2025-03-26:
Unfortunately i don't have the opportunity to test anymore. The issue can be closed if necessary