probe-rs/probe-rs #2850: Connecting to the chip was unsuccessful.: An ARM specific error occurred.: The debug probe encountered an error.: An error which is specific to the debug probe in use occurred.: Command failed with status SwdDpWait

Azorak-Aynir, 2024-10-05:
Hey, so I got my Embassy blinky program to flash and run on my STM32F401RE nucleo board.
But when I made some modification to the files and wanted to re flash and run it, I got a ton of ```WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdDpWait
 WARN probe_rs::probe::stlink: got SwdDpWait/SwdApWait, retrying.``` then the error
 tried to run probe-rs reset and erase, cargo embed/flash ... 
got the same error 

```json
{
  "meta": {
    "version": "0.24.0",
    "commit": "0.24.0 (git commit: crates.io)",
    "arch": "x86_64",
    "os": "windows"
  },
  "command": [
    "probe-rs",
    "run",
    "--chip",
    "STM32F401RETx",
    "--report",
    "target\\thumbv7em-none-eabihf\\debug\\blinkled_rust"
  ],
  "elf": "target\\thumbv7em-none-eabihf\\debug\\blinkled_rust",
  "log": "C:\\Users\\Manu\\AppData\\Roaming\\probe-rs\\probe-rs\\data\\1728165338080.log",
  "error": "Connecting to the chip was unsuccessful.\n\nCaused by:\n    0: An ARM specific error occurred.\n    1: The debug probe encountered an error.\n    2: An error which is specific to the debug probe in use occurred.\n    3: Command failed with status SwdDpWait"
}
```

Somehow, it turned out that if I unplugged the USB and plugged it back in I could run any probe-rs command of my choice just fine, **BUT ONLY ONE**, then I would get the same error if I tried to run any other command subsequently

It is worth mentioning that I am still very unfamiliar with the embedded/rust/probe-rs ecosystem, so I might very well be doing something wrong.

I would appreciate some advice/insight into this error.

Thank you for your time

Azorak-Aynir, 2024-10-06:
I think this issue is the same as #2851. But they seem to have identified the problem. See also #2852 : their PR to solve the issue.

Azorak-Aynir, 2024-10-07:
For info resetting the board has the same effect as deconnecting and reconnecting the board. I can run one command then blocked by the same error.

CorkCrawbaugh, 2024-11-14:
I encountered the same problem trying to run "blinky" on a STM32F446RE Nucleo board. It wasn't possible to build then run for more than one cycle.

I discovered that after exiting my "run" (Ctl-C) within my terminal window (simple cmd prompt on windows ) I could connect to the board using the STM32 ST-LINK utility (https://www.st.com/en/development-tools/stsw-link004.html). Simply disconnecting from the "target" reset the board and allowed a subsequent run. 

That step is now a part of my development process.

Yatekii, 2025-02-22:
Is this still a problem on current main? I think we had some fixes around this.

mdelete, 2025-03-10:
Unfortunately, yes. I'm still having the same issues, but on MacOS, with STM32L4P5ZGTx and STM32F446RETx. But it is still rather strange and elusive. It works for long streaks, then suddenly stops working, and then starts working again after some time. I was suspecting flaky USB on the Mac, but seeing the same errors here, on different platforms, I'm not so sure anymore. For a long time, I was also using version 0.24.0, just switched to 0.27.0 yesterday, but still no improvement. It's not just the Nucleo-Boards, I also had the same issue, on very rare occasions, with an ST-LINK/V2 on STM32WLE5JCIx.

Resonanz, 2025-09-28:
I get a similar error for a blue pill with genuine STMF103C8 chip (probe-rs 0.29.1):

```
resonanz@Mac stm32f1 % cargo run --release --bin blinky    
   Compiling embassy-stm32f1-examples v0.1.0 (/Users/resonanz/GitHub/embassy_Sept2025/examples/stm32f1)
    Finished `release` profile [optimized + debuginfo] target(s) in 0.59s
     Running `probe-rs run --chip STM32F103C8 target/thumbv7m-none-eabi/release/blinky`
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: The debug probe encountered an error.
    2: An error which is specific to the debug probe in use occurred.
    3: Command failed with status JtagNoDeviceConnected.
```