probe-rs/probe-rs #2851: Stm32Armv7 debug sequence unconditionally clears DBGMCU_CR/DBG_SLEEP bit when stopping.

inazarenko, 2024-10-06:
**Describe the bug**

Every time probe-rs disconnects from the target MCU, this code:

https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/vendor/st/sequences/stm32_armv7.rs#L94-L96

clears the lower 3 bits of the DBGMCU_CR register, regardless of what its state was when the probe was attached.

If the target MCU firmware spends most of its time in WFE sleep, next time one tries to attach the probe, the attempt would fail with:

```
...
WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdDpWait
WARN probe_rs::probe::stlink: got SwdDpWait/SwdApWait, retrying.
WARN probe_rs::probe::stlink: too many retries, giving up
```

**To Reproduce**
Steps to reproduce the behavior:

1. Attach an STM32F3Discovery over USB and make sure probe-rs can see and connect.
2. Clone https://github.com/inazarenko/dbgmcu_cr_test and `cargo run` to load the firmware.
3. Observe green user LED marked LD7 turn on. LED is set by the firmware to match the DBGMCU_CR/DBG_SLEEP bit every second.
4. Ctrl-C to detach probe-rs session
5. Observe green the LED turn off.
6. `cargo run` again; observe the failure to connect.
7. Reset the MCU with RESET button to allow probe-rs to connect again.

**Expected behavior**
Embassy apps aggressively sleep the core with WFE and attempts to configure these STM MCUs in the "allow debug in sleep mode" . The default configuration sets it here: 

https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/lib.rs#L257

Probe-rs should preserve these control bits to allow disconnecting/connecting to the MCU more than once without having to manually reset the MCU.

**Desktop (please complete the following information):**

- Windows 10 (but not really relevant).

inazarenko, 2024-10-06:
See also https://github.com/probe-rs/probe-rs/pull/2852

Azorak-Aynir, 2024-10-06:
This issue might be the same problem I had #2850

schctl, 2024-11-19:
Can confirm that this issue occurs for me with my STM32F411 board, and is fixed by #2852

irina-nita, 2024-12-01:
This issue occurred for me with an STM32F429 board. The fix in #2852 seems to solve it for now. Are there any updates on when it might be merged into upstream? Thanks!

closed 2024-12-19 by Yatekii