probe-rs/probe-rs #2860: Confusing output when using ESP32C3 via EspJtag

Krensi, 2024-10-11:
Hi!
First of all, thanks for your great work! 

I am playing around with the esp32C3 and probe-rs 0.24.0.

After installing the correct WinUSB drivers with Zadig I can download a [sample firmware](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/AT_Binary_Lists/esp_at_binaries.html). 
[image]

I was a little confused if the connection is alright when I saw the output of `probe-rs info --chip esp32c3` which was like this:
[image]

Is this a possible enhancement to improve the output, or is this output expected as it is?

Thanks again and greetings,
Christian

bugadani, 2024-10-11:
That output is not quite right, but probe-rs info is also useless for you, so if everything works fine, this is a non-issue at the moment.

Yatekii, 2024-12-11:
Why does it print all zeroes? 🤔