probe-rs/probe-rs #2869: Can´t flash to STM32F103C8T6

MathieuFennema, 2024-10-20:
Hello there,

I tried to start my embedded project on a '[blue pill]( https://www.aliexpress.com/item/1005006788305638.html?spm=a2g0o.order_list.order_list_main.5.619a1802RlTso6 )' and followed '[Rusty Bit's](https://www.youtube.com/watch?v=TOAynddiu5M)'  suggestions, and tried mocking about. 

The setup I'm using is a cheap [J-link](https://www.aliexpress.com/item/1005006628169678.html?spm=a2g0o.order_list.order_list_main.10.619a1802RlTso6) flasher to flash the blue pill. So I tried a blink program via Platform I/O, and that seemed to upload fine. 
The Rust binary compiles correctly, and when I drag my output bin to the Platform I/O build folder, I can also upload my Rust bin using Platform I/O.
I've also used the official JFlashExe application, which could flash both binaries.

The error I'm getting is "Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: An error occurred in the communication with an access port or debug port.
    2: Target device did not respond to request." 

I've checked my OS device privlages, played around with the probe-rs configuration, but just can´t make it work.
Do you have any advice for me? (I tried all options, and combination of those, e.g. connect_under_reset, do_chip_erase, set the protocol to "Swd", etc... I've added my report as suggested on the page: 
 [report.zip](https://github.com/user-attachments/files/17451694/report.zip)

MathieuFennema, 2024-11-01:
Update: I've tried flashing it with my Segger J-trace, found a page on[ how to hook it up](https://wiki.segger.com/Blue_Pill) . This worked directly, without issue, no code/cargo setting modifications. The issue must be in the cheap flasher.

Yatekii, 2025-02-22:
Yeah china-provided JLink fakes are known to not work.

closed 2025-02-22 by Yatekii

bugadani, 2025-02-22:
Actually, that particular probe is pretty okay, I use multiple of those.