probe-rs/probe-rs #2888: Check if esp partition-table is honored when using embedded-test

t-moe, 2024-11-07:
In my project I use a custom esp-idf partition table. it has two ota partitions and also ota_data, and then some more config partitions.

I noticed, that when I flash my normal project using espflash, then later run an embedded-test (with probe-rs), both with the same partition file specified, my config partitions are gone after running embedded-test (with probe-rs).  Maybe probe-rs does not honor the partition table when running embedded-test?

t-moe, 2024-11-07:
Also, appending `--restore-unwritten` will just result in an error:

>  WARN probe_rs::probe::espusbjtag::protocol: Something went wrong in read_bulk Kind(TimedOut) when trying to read 64bytes - pending_in_bits: 550
> Reading flash ✔   0% [--------------------]
> Error: The flashing procedure failed for 'hello_world.elf'.
> 
> Caused by:
>     0: Something during the interaction with the core went wrong
>     1: An error with the usage of the probe occurred
>     2: USB Communication Error
>     3: timed out

bugadani, 2024-11-10:
> WARN probe_rs::probe::espusbjtag::protocol: Something went wrong in read_bulk Kind(TimedOut) when trying to read 64bytes - pending_in_bits: 550

Let's start here. I can reproduce it, and I have no idea yet why this happens. C6 and H2 are affected, C2 and Xtensas are not. I'm not sure if this is the probe driver (pending_in_bits: 550 is consistent) or the riscv layer, or needs both at the same time.

bugadani, 2024-11-23:
The `--restore-unwritten` issue is fixed by #2921