probe-rs/probe-rs #2909: nrf5340-dk Core 1 Locked

djovanoski, 2024-11-15:
**Describe the bug**
i have nrf5340-dk  rev 2.0.2 when i run simple hello_world example with info log i get Core 1 Locked , i have tried the erase and recover script from examples in the repo, also i have used nrfutil cmd from nordic where i recover the device but still this issue. In nrf5340-app-hal is metntioned about SNordic's [Secure Partition Manager](https://github.com/nrfconnect/sdk-nrf/tree/v1.7/samples/spm) from nRF SDK v1.7. SPM v1.7 is configured to expect your application at address 0x0005_0000 and so that is what memory.x must specify as the start of Flash. but i can not find it this to tested as is removed. 

**To Reproduce**
Steps to reproduce the behavior:

1.  cargo generate \
    --git https://github.com/knurling-rs/app-template \
    --branch main \
    --name my-app
2.  Added dependecies ( nrf5340-app-hal)
3. cargo rb hello 

**Stacktrace**
```
RUST_LOG=info cargo r 
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.01s
     Running `probe-rs run --chip nRF5340_xxAA --allow-erase-all target/thumbv8m.main-none-eabihf/debug/tz_nrf`
 INFO probe_rs::config::target: Using sequence Arm(Nrf5340(()))
 INFO probe_rs::probe::jlink: J-Link: Firmware version: J-Link OB-nRF5340-NordicSemi compiled Apr 11 2024 17:44:26
 INFO probe_rs::probe::jlink: J-Link: Hardware version: J-Link 1.0.0
 INFO probe_rs::probe::jlink: J-Link: Target voltage: 3.30 V
 INFO probe_rs::probe::jlink::speed: Selecting speed: 400 Hz
 INFO probe_rs::architecture::arm::sequences: Debug port Default is powered down, powering up
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Checking if core 0 is unlocked
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Core 0 is already unlocked
 INFO probe_rs::vendor::nordicsemi::sequences::nrf: Checking if core 1 is unlocked
 WARN probe_rs::vendor::nordicsemi::sequences::nrf: Core 1 is locked. Erase procedure will be started to unlock it.
 WARN probe_rs::architecture::arm::core::armv8m: The core is in locked up status as a result of an unrecoverable exception
 WARN probe_rs::architecture::arm::core::armv8m: The core is in locked up status as a result of an unrecoverable exception
 INFO probe_rs::flashing::download: Found loadable segment, physical address: 0x00050000, virtual address: 0x00050000, flags: 0x4
 INFO probe_rs::flashing::download: Matching section: ".vector_table"
 INFO probe_rs::flashing::download: Found loadable segment, physical address: 0x00050154, virtual address: 0x00050154, flags: 0x5
 INFO probe_rs::flashing::download: Matching section: ".text"
 INFO probe_rs::flashing::download: Found loadable segment, physical address: 0x000512e8, virtual address: 0x000512e8, flags: 0x4
 INFO probe_rs::flashing::download: Matching section: ".rodata"
 INFO probe_rs::flashing::download: Found loadable segment, physical address: 0x00051734, virtual address: 0x2003fbc0, flags: 0x6
 INFO probe_rs::flashing::download: Matching section: ".data"
 INFO probe_rs::flashing::loader: Found 4 loadable sections:
 INFO probe_rs::flashing::loader:     .vector_table at 0x00050000 (340 bytes)
 INFO probe_rs::flashing::loader:     .text at 0x00050154 (4500 bytes)
 INFO probe_rs::flashing::loader:     .rodata at 0x000512E8 (1100 bytes)
 INFO probe_rs::flashing::loader:     .data at 0x00051734 (56 bytes)
 INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20040000, cores: ["application"], access: None } INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20040000, cores: ["application"], access: None } INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack               INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002740                                                       ░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20040000, cores: ["application"], access: None }
 INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20040000, cores: ["application"], access: None }
 INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack
 INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002740
 INFO probe_rs::flashing::flasher: Erasing sector at address 0x00050000
 INFO probe_rs::flashing::flasher: Done erasing sector. Result is 0. This took 119.978131ms
 INFO probe_rs::flashing::flasher: Erasing sector at address 0x00051000
 INFO probe_rs::flashing::flasher: Done erasing sector. Result is 0. This took 119.978141ms
 INFO probe_rs::flashing::flasher: Took 150.428435ms to download 4096 byte page into ram
 INFO probe_rs::flashing::flasher: Took 150.153465ms to download 4096 byte page into ram
 WARN session_drop: probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    A timeout occurred during an operation.                                                                             INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                          INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                         Error: The flashing procedure failed for 'target/thumbv8m.main-none-eabihf/debug/tz_nrf'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: A timeout occurred during an operation.
```

**Desktop (please complete the following information):**
- Linux  Ubuntu 24.10

Yatekii, 2024-11-23:
I guess we need to find out where exactly it times out.

djovanoski, 2024-11-23:
Hey @Yatekii  i have  forked probe-rs and debug where it exit  so based on my understanding it breaks in run_flash_download(..) method in loader.commit(session, options) in iterate all flash algorithms we need to use and to the flashing so it process the algo on first app core and exit on second core as is locked . I hope this helps if you need additional info let me know.

igiona, 2024-12-01:
@djovanoski can you try to run your code with probe-rs v0.23.0 ?

I've been seeing this as well in 0.24.0 upwards, but not on v0.23.0

djovanoski, 2024-12-05:
@igiona just to confirm that probe-rs v0.23.0 solved the issue , sorry for the delay in answer but thanks a lot.

closed 2024-12-05 by djovanoski

igiona, 2024-12-07:
@djovanoski this is actually "bad", since `0.23.0` is an older version of the tool.

Can you give the following code another try to see weather it also works?

```bash
cargo install probe-rs-tools -f --git https://github.com/igiona/probe-rs.git --locked --branch revert-halting-on-wr
```

If it does work, that this issue is related to #2930

djovanoski, 2024-12-07:
@igiona  so to confirm again that your probe-rs worked with the app core and net core as i have worked with net core and with v0.23.0 was not able , as i was getting the Failed to attach to RTT but now app and net cores are working with your version , so just to thank you even this is not good but at least i can work on the code with your probe-rs :)

igiona, 2024-12-09:
> [@igiona](https://github.com/igiona) so to confirm again that your probe-rs worked with the app core and net core as i have worked with net core and with v0.23.0 was not able , as i was getting the [...]

Thank you for testing 👍 
Somewhat I'm glad to hear that it works, but now we need to figure out how to properly fix this without breaking other use-cases... or maybe how to properly fix the use-case that introduced the change in the first place 🤗 
Let's track this in #2930

igiona, 2024-12-12:
There have been a fix on master, I would give it a try, likely #2949 fixes this for you as well 👍