probe-rs/probe-rs #2930: Regression when of probe-rs run on nrf5340 (compared to v0.24.0)

igiona, 2024-12-01:
**Describe the bug**

When trying to flash & run the application core of the nrf5340, Erase&Flash works as expected but probe-rs fails to run/attach to the application as follow:

```bash
$ cargo run
    Finished `dev` profile [optimized + debuginfo] target(s) in 11.53s
     Running `probe-rs run --chip nRF5340_xxAA --probe '1209:4853' --log-format oneline 'target\thumbv8m.main-none-eabihf\debug\rs-watch'`
 WARN probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    A timeout occurred during an operation.                                                                                                 Error: The flashing procedure failed for 'target\thumbv8m.main-none-eabihf\debug\rs-watch'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: A timeout occurred during an operation.
error: process didn't exit successfully: `probe-rs run --chip nRF5340_xxAA --probe '1209:4853' --log-format oneline 'target\thumbv8m.main-none-eabihf\debug\rs-watch'` (exit code: 1)

```

**To Reproduce**
Steps to reproduce the behavior:
You need either a ZSWatch-HW or a nrf5340-DK

1. Clone https://github.com/igiona/rs-watch
2. Adapt/Remove the probe S/N in `.cargo/config.toml`
3. cargo run
4. watch it fail 

**Expected behavior**
The application starts, probe-rs attaches and defmt-logs are displayed.

**Desktop (please complete the following information):**

- Windows 11

**Additional context**
The same code runs as expected when using probe-rs v0.24.0 (installed via `cargo binstall`)

igiona, 2024-12-01:
@Dirbaio is this something you experience also on the `nrf54` ?

igiona, 2024-12-02:
Bisect result is point to:

>$ git bisect bad
a4dd765487bcfa7f25027bd7daffbb57cf531248 is the first bad commit
commit a4dd765487bcfa7f25027bd7daffbb57cf531248 (HEAD)
Author: Robin Mueller <31589589+robamu@users.noreply.github.com>
Date:   Thu Jul 11 12:48:28 2024 +0200
\
    Halt core when flashing RAM (#2672)
\
 changelog/added-core-halt-for-ram-flash.md | 1 +
 probe-rs/src/flashing/loader.rs            | 9 +++++++++
 2 files changed, 10 insertions(+)
 create mode 100644 changelog/added-core-halt-for-ram-flash.md

igiona, 2024-12-02:
Reverting the commit on `main` fixes this issue, so indeed this is causing troubles on the `nrf5340`.

Yatekii, 2024-12-03:
Seems to me like we cannot halt the core. Why though?

igiona, 2024-12-09:
> Seems to me like we cannot halt the core. Why though?

From #2909 I'm somehow keen to think that halting one core, does halt the other as well and probe-rs kind of don't like/expect it?

marcteys, 2024-12-09:
For reference, this also happens with nrf9160 and nfr9161 (which makes sense because they both have nrf5340 inside). 

Downgrading to v0.23.0 fixes the issue.

Tiwalun, 2024-12-09:
> For reference, this also happens with nrf9160 and nfr9161 (which makes sense because they both have nrf5340 inside).

These chips don't have a nRF5340 inside. On the nRF5340, there is an application and a radio core, which both can be accessed.
The nRF9160 and 61 only have an application core.

> Downgrading to v0.23.0 fixes the issue.

Does this mean that v0.24.0 is broken for you? Then this sounds like a different issue?

marcteys, 2024-12-09:
Thanks for the clarification.

I had the same issues of locked core when flashing with v0.24 (latest). I
had to 'manually' unlock it with --allow-erase-all  I downgraded to v0.23
and no more issue. So I suppose v0.24 is broken (Windows 10 native and
Windows 10 when flashing using WSL).

igiona, 2024-12-10:
> Thanks for the clarification.
> 
> I had the same issues of locked core when flashing with v0.24 (latest). I
> had to 'manually' unlock it with --allow-erase-all  I downgraded to v0.23
> and no more issue. So I suppose v0.24 is broken (Windows 10 native and
> Windows 10 when flashing using WSL).
> 
> 
> > For reference, this also happens with nrf9160 and nfr9161 (which makes
> > sense because they both have nrf5340 inside).
> >
> > These chips don't have a nRF5340 inside. On the nRF5340, there is an
> > application and a radio core, which both can be accessed.
> > The nRF9160 and 61 only have an application core.
> >
> > Downgrading to v0.23.0 fixes the issue.
> >
> > Does this mean that v0.24.0 is broken for you? Then this sounds like a
> > different issue?

Could you give this a try?

```
cargo install probe-rs-tools -f --git https://github.com/igiona/probe-rs.git --locked --branch revert-halting-on-wr
```

?

closed 2024-12-10 by bugadani