probe-rs/probe-rs #2932: Probe-rs unable to shut down core on CodeCell (esp32c3)

OleStrohm, 2024-12-01:
**Describe the bug**
I am using probe-rs (0.24.0) to flash and get logging from Rust code running on a [CodeCell](https://microbots.io/products/codecell?variant=49714638717261)
It's working for the most part, but logging is very inconsistent, and it seems to be caused by probe-rs not being able to shut down the core properly.
I'm guessing that since the core keeps running after shutting down probe-rs, it keeps filling up the rtt buffer, which breaks hte logging.

Here's an example output from the below program:
```
INFO  ESP initialized!
└─ esp_example::__risc_v_rt__main @ src/main.rs:12
INFO  ping
└─ esp_example::__risc_v_rt__main @ src/main.rs:14
INFO  ping
└─ esp_example::__risc_v_rt__main @ src/main.rs:14
INFO  ping
└─ esp_example::__risc_v_rt__main @ src/main.rs:14
INFO  ping
└─ esp_example::__risc_v_rt__main @ src/main.rs:14
INFO  ping
└─ esp_example::__risc_v_rt__main @ src/main.rs:14
ERROR probe_rs::util::rtt:
Error reading from RTT: The control block has been corrupted. write pointer is 43293265 while buffer size is 1024 for up channel 0 (defmt)
INFO  ESP initialized!
└─ esp_example::__risc_v_rt__main @ src/main.rs:12
INFO  ping
└─ esp_example::__risc_v_rt__main @ src/main.rs:14
```

**To Reproduce**
With the following program
```rust
#![no_std]
#![no_main]

use defmt::info;
use defmt_rtt as _;
use esp_backtrace as _;
use esp_hal::delay::Delay;
use esp_hal::prelude::*;

#[entry]
fn main() -> ! {
    info!("ESP initialized!");
    loop {
        info!("ping");
        Delay::new().delay_millis(1000);
    }
}
```
and run with `probe-rs run ./target/riscv32imc-unknown-none-elf/debug/esp_example --chip esp32c3`

**Expected behavior**
Should print without issue on all runs

**Stacktrace**
Warning when pressing ctrl-c on probe-rs
```
WARN probe_rs::session: Failed to deconfigure device during shutdown: A RISC-V specific error occurred.

Caused by:
    Error occurred during execution of an abstract command: HaltResume
```
And error when starting program:
```
ERROR probe_rs::util::rtt:
Error reading from RTT: The control block has been corrupted. write pointer is 43293265 while buffer size is 1024 for up channel 0 (defmt)
```

**Desktop:**
- Linux (NixOS)

**Additional context**
I am running this on a [CodeCell](https://microbots.io/products/codecell?variant=49714638717261), and while I don't have any other esp32c3 devices to test, I'm guessing this is likely an issue specific to this device.
I'm happy to run any potential patches, or investigate why this issue is happening myself, if someone could guide me to where this error likely originates in the code.

Yatekii, 2024-12-03:
cc @bugadani

bugadani, 2024-12-04:
Can you try probe-rs from git master, please?

OleStrohm, 2024-12-04:
It works as expected on master! Thank you! I'll see if I can figure out how to update to it temporarily so I don't have to wait on the next release :)

closed 2024-12-04 by OleStrohm