probe-rs/probe-rs #2935: Inconsistent Test Results on STM32H750

Dicklessgreat, 2024-12-03:
I am running tests on the STM32H750 and have encountered a very strange situation. Each time I run the tests, the results are different. I've labeled the patterns I observed as "A", "B," and "C."

- Pattern A: The test panics before it begins.
```
ERROR probe_rs::debug::debug_info: Other("UNWIND: Tried to unwind `RegisterRule` at CFA = None.")
Frame 0: HardFault <Cause: Escalated UsageFault <Cause: Undefined instruction>> : ERROR: While resolving function information for the program counter (0x00000002) that caused the exception. @ 0x00000002
Frame 1: <unknown function @ 0x00000002> : ERROR: UNWIND: Tried to unwind `RegisterRule` at CFA = None. @ 0x00000002
       /Users/dicklessgreat/.cargo/git/checkouts/embassy-9312dcb0ed774b29/b88dc13/embassy-time-queue-driver/src/lib.rs:44:14
Error: CPU halted unexpectedly.
error: test failed, to rerun pass `--test test`

Caused by:
  process didn't exit successfully: `probe-rs run --chip STM32H750IBKx /Users/dicklessgreat/Documents/Rust/stm32h750_embedded_test/target/thumbv7em-none-eabihf/debug/deps/test-459b163b2cf10477` (exit status: 1)
```
- Pattern B: The test also panics before it starts. But another error message.
```
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApFault
 WARN probe_rs::debug::debug_info: UNWIND: Error while checking for exception context. The stack trace will not include the calling frames. : UNWIND: Error reading FrameDescriptorEntry at PC=134247133 : Do not have unwind info for the given address.
Frame 0: UNWIND: Error while checking for exception context. The stack trace will not include the calling frames. : UNWIND: Error reading FrameDescriptorEntry at PC=134247133 : Do not have unwind info for the given address. @ 0x08006cb4
Error: CPU halted unexpectedly.
error: test failed, to rerun pass `--test test`

Caused by:
  process didn't exit successfully: `probe-rs run --chip STM32H750IBKx /Users/dicklessgreat/Documents/Rust/stm32h750_embedded_test/target/thumbv7em-none-eabihf/debug/deps/test-459b163b2cf10477` (exit status: 1)

0.000000 INFO  tests available: [Test { name: "tests::first_test", function: 0x8001165, should_panic: false, ignored: false, timeout: None }]
└─ embedded_test::export::run_tests @ /Users/dicklessgreat/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-test-0.4.0/src/fmt.rs:36  

```
- Pattern C: The test passes successfully.
```
running 1 test
test tests::first_test ... 0.000000 INFO  Running test: Test { name: "tests::first_test", function: 0x8001165, should_panic: false, ignored: false, timeout: None }
└─ embedded_test::export::run_tests @ /Users/dicklessgreat/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-test-0.4.0/src/fmt.rs:36  
0.000000 INFO  Test exited with () or Ok(..)
└─ embedded_test::export::check_outcome @ /Users/dicklessgreat/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-test-0.4.0/src/fmt.rs:36  
ok

test result: ok. 1 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.09s
```
To help with troubleshooting, I've created the simple sample project that reproduces the issue. You can find it [here](https://github.com/Dicklessgreat/stm32h750_embedded_test)

TBH, I don't know what the error message really means, but if anyone has any ideas or suggestions for resolving this, I would greatly appreciate it.

Dicklessgreat, 2024-12-03:
`probe-rs --version`:
```
probe-rs 0.24.0 (git commit: f8185426)
```
Then, sample project works fine!
Thanks!

closed 2024-12-03 by Dicklessgreat

t-moe, 2024-12-03:
Oh, ok.  I guess this one solved itself...
Thank you.

Dicklessgreat, 2024-12-03:
I tested `embedded-test` both `0.4.0` and `0.5.0`, and both work fine
The only difference from last try is `probe-rs`.
So, something between `40b83ff1` and `f8185426` fixed the problem?