probe-rs/probe-rs #2944: BMP does not run ARM targets

mfiumara, 2024-12-08:
**Describe the bug**
I cannot run using Black Magic debug Probe using probe-rs. 

**To Reproduce**
Steps to reproduce the behavior:

1. Compile an nrf91 example code from here (https://github.com/embassy-rs/embassy/blob/main/examples/nrf9160/src/bin/blinky.rs) 
2. `probe-rs run --chip=nRF9160_xxAA --always-print-stacktrace --protocol=swd`

**Expected behavior**
This should run the application on the target.

**Stacktrace**
```
RUST_BACKTRACE=1 cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.33s
     Running `probe-rs run --chip=nRF9160_xxAA --always-print-stacktrace --protocol=swd '--log-format={t} {[{L}]%bold} {s}  {{c} {ff}:{l:1}%dimmed}' target/thumbv8m.main-none-eabihf/debug/nrf-tracker`
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: The debug probe encountered an error.
    2: The selected probe does not support the 'ARM' interface.

Stack backtrace:
   0: std::backtrace::Backtrace::create
   1: anyhow::error::<impl core::convert::From<E> for anyhow::Error>::from
   2: probe_rs::cmd::run::Cmd::run
   3: probe_rs::main
   4: std::sys::backtrace::__rust_begin_short_backtrace
   5: std::rt::lang_start::{{closure}}
   6: std::rt::lang_start_internal
   7: main
   8: <unknown>
   9: __libc_start_main
  10: _start
```

**Desktop (please complete the following information):**

- Arch Linux

**Additional context**

There is no problem when using `probe-rs info` and the cores pop up:

```
probe-rs info --protocol swd
Probing target via SWD

ARM Chip with debug port Default:
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0x80000000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0x80000000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0x80000000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
Debug Port: DPv2, Designer: Holt Integrated Circuits, Part: 0x0, Revision: 0xf, Instance: 0x0f
├── 0 MemoryAP (AmbaAhb3)
│   └── ROM Table (Class 1), Designer: ARM Ltd
│       ├── Cortex-M33 SCS  (Coresight Component)
│       │   └── CPUID
│       │       ├── IMPLEMENTER: ARM Ltd
│       │       ├── VARIANT: 0
│       │       ├── PARTNO: Cortex-M33
│       │       └── REVISION: 2
│       ├── Cortex-M33 DWT  (Coresight Component)
│       ├── Cortex-M33 BPU  (Coresight Component)
│       ├── Cortex-M33 ITM  (Coresight Component)
│       ├── Cortex-M33 ETM  (Coresight Component)
│       └── Cortex-M33 CTI  (Coresight Component)
├── 1 MemoryAP (AmbaAhb3)
│   └── ROM Table (Class 1), Designer: ARM Ltd
│       ├── Cortex-M33 SCS  (Coresight Component)
│       │   └── CPUID
│       │       ├── IMPLEMENTER: ARM Ltd
│       │       ├── VARIANT: 0
│       │       ├── PARTNO: Cortex-M33
│       │       └── REVISION: 2
│       ├── Cortex-M33 DWT  (Coresight Component)
│       ├── Cortex-M33 BPU  (Coresight Component)
│       ├── Cortex-M33 ITM  (Coresight Component)
│       ├── Cortex-M33 ETM  (Coresight Component)
│       └── Cortex-M33 CTI  (Coresight Component)
├── 2 Unknown AP (Designer: Nordic VLSI ASA, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x1)
├── 3 MemoryAP (AmbaApb2Apb3)
│   └── Generic
├── 4 Unknown AP (Designer: Nordic VLSI ASA, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x1)
├── 5 Unknown AP (Designer: Nordic VLSI ASA, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x1)
└── 6 Unknown AP (Designer: Nordic VLSI ASA, Class: Undefined, Type: 0x1, Variant: 0xa, Revision: 0x1)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

mfiumara, 2024-12-09:
I debugged this and it seems to be caused due to `get_arm_communication_interface` is not implemented.

xobs, 2024-12-10:
I'm not able to reproduce this, but then again I don't have an nRF91 board to test it on.

Using the latest `master` and an STM32F103 board I have:

```
❯ cargo run --bin probe-rs -- list
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.13s
     Running `target/debug/probe-rs list`
The following debug probes were found:
[0]: Black Magic Probe v1.10.0-1426-g6cc8286e -- 1d50:6018:97B6DC14 (Black Magic Probe)

seancross in 🌐 bestways in probe-rs on  master [$?] is 📦 v0.24.0 via 🦀 v1.83.0
❯ cargo run --bin probe-rs -- run --chip=STM32F103RB --always-print-stacktrace --protocol=swd '--log-format={t} {[{L}]%bold} {s}  {{c} {ff}:{l:1}%dimmed}' zephyr.elf
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.13s
     Running `target/debug/probe-rs run --chip=STM32F103RB --always-print-stacktrace --protocol=swd '--log-format={t} {[{L}]%bold} {s}  {{c} {ff}:{l:1}%dimmed}' zephyr.elf`
      Erasing ✔ 100% [####################]  78.00 KiB @  25.94 KiB/s (took 3s)
  Programming ✔ 100% [####################]  78.00 KiB @  11.21 KiB/s (took 7s)
     Finished in 6.97s
```

xobs, 2024-12-10:
(Note that I did discover that BMP didn't support connecting under reset, since the `swj_pins()` function was a no-op. That's fixed in the latest PR.)

xobs, 2024-12-10:
It works if I use an nRF52840 as well:

```
❯ cargo run --bin probe-rs -- run --chip=nRF52840_xxAA --always-print-stacktrace --protocol=swd pwm-blinky-demo
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.13s
     Running `target/debug/probe-rs run --chip=nRF52840_xxAA --always-print-stacktrace --protocol=swd pwm-blinky-demo`
      Erasing ✔ 100% [####################]   4.00 KiB @  24.38 KiB/s (took 0s)
  Programming ✔ 100% [####################]   4.00 KiB @  12.90 KiB/s (took 0s)
     Finished in 0.33s
```

And I can verify that probe-rs successfully programmed the device:

```
❯ arm-none-eabi-gdb  -ex 'tar ext /dev/cu.usbmodem97B6DC141' -ex 'mon s' -ex 'att 1' -ex 'compare-sections' -ex 'disconnect' -ex 'exit' pwm-blinky-demo
GNU gdb (Arm GNU Toolchain 13.3.Rel1 (Build arm-13.24)) 14.2.90.20240526-git
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=aarch64-apple-darwin20.6.0 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.linaro.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from pwm-blinky-demo...
(No debugging symbols found in pwm-blinky-demo)
Remote debugging using /dev/cu.usbmodem97B6DC141
Target voltage: 3.2V
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port
Attaching to program: /Users/seancross/Code/probe-rs/pwm-blinky-demo, Remote target
0x000e4b9e in ?? ()
Section .vector_table, range 0x50000 -- 0x50100: matched.
Section .text, range 0x50100 -- 0x50a64: matched.
Section .rodata, range 0x50a64 -- 0x50c18: matched.
Ending remote debugging.

seancross in 🌐 bestways in probe-rs on  bmp-connect-under-reset [$?] is 📦 v0.24.0 via 🦀 v1.83.0
❯
```

xobs, 2024-12-10:
The nRF91 is very different from the nRF52:

```
❯ probe-rs info --protocol swd
Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv1, DP Designer: ARM Ltd
├── 0 MemoryAP (AmbaAhb3)
│   └── ROM Table (Class 1), Designer: Nordic VLSI ASA
│       ├── Cortex-M4 SCS   (Generic IP component)
│       │   └── CPUID
│       │       ├── IMPLEMENTER: ARM Ltd
│       │       ├── VARIANT: 0
│       │       ├── PARTNO: Cortex-M4
│       │       └── REVISION: 1
│       ├── Cortex-M3 DWT   (Generic IP component)
│       ├── Cortex-M3 FBP   (Generic IP component)
│       ├── Cortex-M3 ITM   (Generic IP component)
│       ├── Cortex-M4 TPIU  (Coresight Component)
│       └── Cortex-M4 ETM   (Coresight Component)
└── 1 Unknown AP (Designer: Nordic VLSI ASA, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x0)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

…ss in 🌐 bestways in probe-rs on  bmp-connect-under-reset [$?] is 📦 v0.24.0 via 🦀 v1.83.0
❯
```

xobs, 2024-12-10:
Actually, what version is your BMP running? Can you run `probe-rs list`?

mfiumara, 2024-12-10:
> Actually, what version is your BMP running? Can you run `probe-rs list`?

[0]: Black Magic Probe v1.10.2 -- 1d50:6018:98B72495 (Black Magic Probe)

mfiumara, 2024-12-10:
Had some extended discussions on https://app.element.io/#/room/#probe-rs:matrix.org to triage this bug.
It looks like the debug sequence was not correctly implemented for any debug probe that did not implement `get_arm_communication_interface`.

I have opened a PR that fixes these debug sequences, check the linked PR #2948. I will close this issue once merged.

PR #2948 (2024-12-11): fix: nrf91 sequence on BMP
This fixes the nrf91 sequence for the BMP probe by removing of accessing
the memory_interface. Accessing this interface before the cores results
in a non-function debug sequence.

Fixes #2944
closed 2024-12-11 by Tiwalun