probe-rs/probe-rs #2946: ITM trace through ETF/ETM/ETB is silent

jordens, 2024-12-09:
**Describe the bug**

As per discussion with @ryan-summers 

ITM trace through ETF/ETM/ETB is silent.

**To Reproduce**

```
rj@murray:~/src/stabilizer (main)$ probe-rs itm --chip STM32H743ZITx --probe 0483:3754:0016001B5553500A20393256 memory 400000000
 WARN probe_rs::config::sequences::stm32h7: Enabling tracing for STM32H7
rj@murray:~/src/stabilizer (main)$ 
```

**Expected behavior**

Before (a couple versions back) it printed a nice timed trace of interrupt enter and exit.

**Stacktrace**

N/A

**Desktop (please complete the following information):**

- Ubuntu GNU/Linux

**Additional context**

- Version: current probe-rs cli

Yatekii, 2024-12-10:
Could you provide a log with `RUST_LOG=debug`?

Also, maybe with a fix for this we should come up with a way to test ITM.

jordens, 2024-12-11:
`RUST_LOG=debug probe-rs itm --chip STM32H743ZITx --probe 0483:3754:0016001B5553500A20393256 --log-file /dev/null memory 400000000`

bad: [probe-rs-itm-0.24-debug.log](https://github.com/user-attachments/files/18093260/probe-rs-itm-0.24-debug.log)
good: [probe-rs-itm-0.20-debug.log](https://github.com/user-attachments/files/18093259/probe-rs-itm-0.20-debug.log)

c.f. #1688

Yatekii, 2024-12-11:
Would it be possible to git bisect this?

ryan-summers, 2025-02-04:
@jordens I'm not seeing anything wrong with ITM tracing to the console. However, I am seeing Stabilizer hitting an internal panic (front panel LEDs drop to red) when attempting to start ITM tracing. Attaching to an RTT console for a dump is indicating that there's a DAC overflow:

```
rsummers@Lazarus:~/repositories/quartiq/stabilizer$ probe-rs attach target/thumbv7em-none-eabihf/release/dual-iir --chip STM32H743ZITx
16:27:47.498: panicked at /home/rsummers/repositories/quartiq/stabilizer/src/hardware/dac.rs:272:1:
16:27:47.498: called `Result::unwrap()` on an `Err` value: Overflow                    
Frame 0: __bkpt @ 0x0804e004 inline                 
       /home/rsummers/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/../asm/inline.rs:14:5                                                                                                
Frame 1: bkpt @ 0x000000000804e004 inline                                                                                                                                                                                 /home/rsummers/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/call_asm.rs:11:43
Frame 2: panic @ 0x000000000804e004                                                                                                                                                                                
       /home/rsummers/repositories/quartiq/stabilizer/src/hardware/mod.rs:175:9                                                                                                                                    
Frame 3: pad_integral @ 0x080370a8                                                                       
       /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/core/src/fmt/mod.rs:1343:1                                                                                                                          
Error: CPU halted unexpectedly.  
```

Given that this is highly specific to Stabilizer and not probe-rs, I'm going to close this one and open an issue in the Stabilizer repo.

ryan-summers, 2025-02-04:
When i drop down the sample rate on Stabilizer (as mentioned in https://github.com/quartiq/stabilizer/issues/1012), I can get a full trace out using the CLI:

```
rsummers@Lazarus:~/repositories/probe-rs/probe-rs$ probe-rs itm --chip STM32H743ZITx memory 400000000                                                                                                              
 WARN probe_rs::vendor::st::sequences::stm32h7: Enabling tracing for STM32H7                                                                                                                                       
Err(MalformedPacket(InvalidHardwareDisc { disc_id: 3, size: 3 }))                                                                                                                                                  
Ok(TimestampedTracePackets { timestamp: Sync(3ns), packets: [], malformed_packets: [], consumed_packets: 1 })                                                                            
Ok(TimestampedTracePackets { timestamp: Sync(321.618µs), packets: [], malformed_packets: [], consumed_packets: 1 })                                                                      
Ok(TimestampedTracePackets { timestamp: Sync(327.478µs), packets: [ExceptionTrace { exception: Interrupt { irqn: 15 }, action: Exited }], malformed_packets: [], consumed_packets: 2 })   
Ok(TimestampedTracePackets { timestamp: Sync(327.521µs), packets: [ExceptionTrace { exception: ThreadMode, action: Returned }], malformed_packets: [], consumed_packets: 2 })    
// .... packets continue for many hundreds of lines
```

ryan-summers, 2025-02-04:
I'll look into whether or not the CPU is being halted. It's also possible that the debug probe is using the wrong bus matrix, so the communication to retrieve the ETB data is causing a contention between DMA and the debug probe.

jordens, 2025-02-04:
Ack. I would not have thought bus matrix contention can play a role here as those matrices are in general multi-master. Are you thinking it's the (single) link between the matrices?

Another pointer: during/after detaching with ctrl-c (in `run` or `attach --connect-under-reset`) it panics for the same reason. That's also annoying and similar but it may not be the same root cause. I do think though that it should need to do that.