probe-rs/probe-rs #2950: probe-rs incompatible(?) with rpi debug probe

derenrich, 2024-12-11:
**Describe the bug**
When I try to use probe-rs it works the first time but then leaves the probe in a state where it doesn't work until I power cycle it. I can successfully debug and program my pico using openocd (though only after I power cycle the probe after trying probe-rs).

**To Reproduce**
Steps to reproduce the behavior:

1. install `probe-rs 0.24.0 (git commit: 6fc653a)`
2. run `probe-rs list` -> it detects the probe
3. run `probe-rs info --chip RP2040 --protocol swd` -> it prints an error (see below) but detects the pico
4. re-run `probe-rs info --chip RP2040 --protocol swd` -> now it errors with a new error
5. try to connect with openocd (e.g. `sudo openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000"`) -> errors with "Error: CMSIS-DAP command CMD_INFO failed."
6. Need to power cycle to get openocd to work. But now it works

First error:

```
Error showing ARM chip information for Debug Port Default: An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.
```

Second error:

```
Error: Failed to open the debug probe.

Caused by:
    0: An error which is specific to the debug probe in use occurred.
    1: Error handling CMSIS-DAP command Info
    2: Error in the USB access
```

**Expected behavior**
It should work. It should connect. It should not break openocd

**Stacktrace**

Typically on its second go it produces an error like (no matter what command I use other than list):

```
oss/rp2040-project-template - (main) > probe-rs reset --chip RP2040
Error: Failed to open the debug probe.

Caused by:
    0: An error which is specific to the debug probe in use occurred.
    1: Error handling CMSIS-DAP command Info
    2: Error in the USB access
```

**Desktop (please complete the following information):**

- macOS Mojave

**Additional context**

- Open On-Chip Debugger 0.12.0
- dumping report logs doesn't show anything interesting to me

Yatekii, 2024-12-11:
Seems like we have an issue with getting the device to wake up 🤔

The second error is the same as openocd.

Tiwalun, 2024-12-11:
Does another command (i.e. not `info`) work? The `info` command is not really representative.

It would also be helpful to try the current master version of probe-rs, you can install it using `cargo install --locked --git https://github.com/probe-rs/probe-rs`.

derenrich, 2024-12-11:
I can try again tonight with another command. I also ran `probe-rs reset --chip RP2040` which gave the same USB error (though maybe only after running `info` so it was already borked)

I never got to trying to program anything because it would always stop responding after the first request.

Other people online seem to be reporting success with probe-rs + the pico probe so maybe this is OSX related? Or maybe just nobody else bothered to try to run `info` first? I can try just programming/debugging.

I was able to work around the issue by just swapping the `runner` with a shell-script calling out to openocd but i'd prefer not to.

derenrich, 2024-12-11:
ok running `probe-rs debug --chip RP2040 --protocol swd` after a power cycle does work. 

running `halt` in that debugger does halt the program and 

`probe-rs reset --chip RP2040 --protocol swd` brings it back.

running `probe-rs run --chip RP2040 --protocol swd target/thumbv6m-none-eabi/release/rp2040-project-template` works as expected.

so yeah seems to just be an issue with `info` that leaves it in some broken state (none of the above work after running info).

derenrich, 2024-12-11:
installing from source doesn't fix the issue. if anything it's worse.

```
oss/rp2040-project-template - (main) > ~/.cargo/bin/probe-rs  --version
probe-rs 0.24.0 (git commit: 5b0f5db3)

oss/rp2040-project-template - (main) > ~/.cargo/bin/probe-rs  list
The following debug probes were found:
[0]: Debug Probe _CMSIS_DAP_ -- 2e8a:000c:E663AC91D351832E (CMSIS-DAP)

oss/rp2040-project-template - (main) >  ~/.cargo/bin/probe-rs info --chip RP2040 --protocol swd
Probing target via SWD

ARM Chip with debug port Default:
No access ports found on this chip.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

oss/rp2040-project-template - (main) >  ~/.cargo/bin/probe-rs info --chip RP2040 --protocol swd
Error: Failed to open the debug probe.

Caused by:
    0: An error which is specific to the debug probe in use occurred.
    1: Error handling CMSIS-DAP command Info.
    2: Error in the USB access.
```

elfmimi, 2025-01-03:
to probe RP2040 with _info_ command, it is better done along with --target-sel option for now.
`probe-rs info --target-sel 0x01002927`

solimike, 2025-01-20:
I just hit what I think is the same problem and did a bit of investigation looking into the probe-rs logs and getting my head around the Debug Probe source and _think_ the issue may be a now-fixed problem with Debug Probe that can be resolved by re-flashing your Debug Probe.

I could consistently hit issues that caused my just-delivered Debug Probe to lock up and then require a power-cycle to recover either using `probe-rs info` or `probe-rs reset` without having any target connected. In both cases the logs showed reasonable CMSIS Request/Response exchanges with the Debug Probe until closing the connection. The common factor seemed to be a DAP_Disconnect to sent to the Debug Probe that went unacknowledged and then nothing further from the Debug Probe until power-cycled.

In the case of `probe-rs info` a DAP_Disconnect gets sent after the DAP_Info to get the probe's capabilities, but there has never been a DAP_Connect.

In the case of `probe-rs reset` two DAP_Disconnects get sent, the first of which is acknowledged OK and the second of which is unacknowledged and kills the Debug Probe.

So my thinking was that something about "disconnect-while-disconnected" is pathological for the Debug Probe. Cue some code reading in the [Debug Probe repo](https://github.com/raspberrypi/debugprobe). As far as I can tell sending DAP_Disconnect while disconnected shouldn't be an issue: the CMSIS library treats it as a no-op and the main message loop looks ok, albeit with some suspicious disabling and re-enabling of the task scheduler at various points to create critical sections that look ripe for causing deadlocks.

At this point I thought it sensible to check the code I was reading was sensibly close to the code running on my Debug Probe. `picotool info -a` reported that I was running a "1.4.1-develop SDK" from 1 Feb 2023.  Hmmm! I have just reflashed the probe with [Debug Probe 2.2.1](https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.2.1) from the start of January and now, magically, the lock-ups are resolved. (I hesitate to say `probe-rs` does _actually support_ Debug Probe because my target is RP2350-based and so I'm now off to investigate #3005...)

Hope that's useful background and may help move this issue forward.

ewenmcneill, 2025-01-25:
> At this point I thought it sensible to check the code I was reading was sensibly close to the code running on my Debug Probe. picotool info -a reported that I was running a "1.4.1-develop SDK" from [...]

Excellent find by @solimike!

I found the same situation myself, with a [Raspberry Pi Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html) purchased this month (January 2025).  From factory it:
* shipped with firmware built February 2023; and
* locked up after `probe-rs info` was run (after failing out) so that, eg `probe-rs run ...` then also failed (as described in this issue) until the probe was power cycled

After updating to the [latest Debug Probe firmware](https://github.com/raspberrypi/debugprobe/releases/latest) (still 2.2.1, built 2025-01-10), `probe-rs info` no longer errors out in the way it used to, *and* `probe-rs run ...` *works* after running `probe-rs info` \o/

My guess is that most (all?) Raspberry Pi Debug Probes are probably shipped with the same old firmware from factory (since it's unlikely the factory programming jig has been updated to flash anything newer), and thus are likely to have this problem.  Whereas anyone using a second Pico dev board as a probe presumably flashed a 2024/2025 version of the probe firmware.

For anyone else wanting to try [updating their Raspberry Pi Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#updating-the-firmware-on-the-debug-probe):
*  open the debug probe case (it's just press fit, no clips, so easy to open with a finger nail at one end)
* hold down the "bootsel" button (the only button on the probe board), to put it into programming mode
* copy the `debugprobe.uf2`of the firmware version you want onto the `RPI-RP2` drive that auto-mounts (at least on macOS / Windows; it'll auto-disconnect as soon as the copy finishes)

If you want to check the firmware versions before/after, you'll need [`picotool`](https://github.com/raspberrypi/picotool), either as a binary from somewhere, or built from source (and to put the Debug Probe into "bootsel" mode as above before running `picotool`).

But if you just want to YOLO the Debug Probe update via a U2F upload, in case it helps, [that can be done without `picotool`](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#updating-the-firmware-on-the-debug-probe).

Ewen

PS: The `1.4.1-develop` mentioned above is the Pico SDK version, rather than the Debug Probe firmware version; but it too is rather old, as the Pico SDK version is now around `2.1.0`.

<details><summary>picotool output for Debug Probe *before* firmware update</summary>

```
ewen@rustdev:~$ sudo picotool info -a
Program Information
 name:              picoprobe
 binary start:      0x10000000
 binary end:        0x10009bb0

Fixed Pin Information
 0:                 UART0 TX
 25:                LED

Build Information
 sdk version:       1.4.1-develop
 pico_board:        pico
 boot2_name:        boot2_w25q080
 build date:        Feb  1 2023
 build attributes:  Release

Device Information
 type:              RP2040
 revision:          B2
 flash size:        2048K
 flash id:          [... omitted ...]

Metadata Blocks
 none
ewen@rustdev:~$

```
</details>

<details><summary>picotool output for Debug Probe *after* firmware update</summary>

```
ewen@rustdev:~$ sudo picotool info -a
Program Information
 name:              debugprobe
 features:          PROBE UART BAUDRATE 115200
                    PROBE UART INTERFACE uart1
                    UART stdin / stdout
 binary start:      0x10000000
 binary end:        0x1000a6ec

Fixed Pin Information
 0:                 UART0 TX, UART0 TX
 1:                 UART0 RX, UART0 RX
 4:                 PROBE UART TX
 5:                 PROBE UART RX
 12:                PROBE SWCLK
 13:                PROBE SWDI
 14:                PROBE SWDIO 
 25:                LED

Build Information
 sdk version:       2.1.0
 pico_board:        pico
 boot2_name:        boot2_w25q080
 build date:        Jan  9 2025
 build attributes:  Release

Device Information
 type:              RP2040
 revision:          B2 
 flash size:        2048K
 flash id:          [... omitted ...]

Metadata Blocks
 none
ewen@rustdev:~$ 
```
</details>

ewenmcneill, 2025-01-25:
FWIW, the [Debug Probe firmware 2.1.0 release](https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.1.0) (November 2024) has this intriguing release note:

* Put SWD and nRESET pins into HiZ mode when the DAP is deactivated

which seems like it might be relevant.  There's also DAP changes in [picoprobe-cmsis-v1.1](https://github.com/raspberrypi/debugprobe/releases/tag/picoprobe-cmsis-v1.1) from January 2024, but they feel more like feature support.  (FTR the rename from `picoprobe` to `debugprobe` happened with the [2.0 firmware release](https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.0) in March 2024; but I can't find a way to see that without `picotool` and being in "bootsel" mode.)

Given the early February 2023 release date ("build date:        Feb  1 2023"), my guess is that the factory shipped firmware mage is built from the [picoprobe-cmsis-v1.01](https://github.com/raspberrypi/debugprobe/releases/tag/picoprobe-cmsis-v1.0.1) release (November 2022).  Which seems to be their first CMSIS implementation, in which case it's unsurprising there were bugs (the release notes even say it  should be considered BETA" until they updated documentation).

FTR, the factory firmware is *not* [picoprobe-cmsis-v1.02](https://github.com/raspberrypi/debugprobe/releases/tag/picoprobe-cmsis-v1.02) in the factory image as (a) that's the end of February 2023, not the start; and (b) it explicitly says "Bump SDK to 1.5.0", which is already newer than what was in the factory shipped firmware.

I suspect the `probe-rs` documentation for the Raspberry Pi Debug Probe should probably encourage users to [upgrade to the latest firmware on the debug probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#updating-the-firmware-on-the-debug-probe) before using `probe-rs`, eg on the probe setup documentation page:

https://probe.rs/docs/getting-started/probe-setup/

Ewen

ewenmcneill, 2025-01-27:
Copying over a very relevant comment by @konkers from https://github.com/probe-rs/probe-rs/pull/3005#issuecomment-2614579162 so it's easier to find:

> If you're looking to add a probe specific quirk, it might be better to detect the debugprobe version and error out with information on how to upgrade instead of trying to workaround it in the core ADI support.
> 
> We saw lockup issues with old firmware on the rp2040 as well and opted to do detection in our flash/run infrastructure: https://cs.opensource.google/pigweed/pigweed/+/main:targets/rp2040/py/rp2040_utils/device_detector.py;l=361

The referenced code (in the link quoted above) is a Python example for detecting the Raspberry Pi Debug Probe, and getting its firmware version information, and bailing out with an error that the user should update the Debug Probe firmware.

Of note, that code tells the user to "Update to version 2.0.1 or later", which suggests the issue was fixed in the probe firmware before the 2.1.0 I was guessing earlier as the fix.

(FTR, relevant fix in the GitHub pigweed repo -- https://github.com/google/pigweed/commit/20f3739c55ea8600caa6ededd61ac05fcd80b4d2 -- seems to have been added June 2024)

Ewen

ewenmcneill, 2025-01-27:
This also gives a way (Linux) to check if the debug probe has been updated from the command line, without having to get into "bootsel" mode, since it seems the firmware version is being encoded into `bcdDevice` in the USB properties:

Fresh out of box (also purchased January 2025; fortunately I had bought two at once so one hasn't been updated):

```
ewen@rustdev:~$ sudo lsusb -v -d 2e8a:000c | egrep '2e8a:000c|idProduct|idVendor|bcdDevice|iManufacturer|iProduct'
Bus 002 Device 029: ID 2e8a:000c Raspberry Pi Debug Probe (CMSIS-DAP)
  idVendor           0x2e8a 
  idProduct          0x000c 
  bcdDevice            1.01
  iManufacturer           1 Raspberry Pi
  iProduct                2 Debug Probe (CMSIS-DAP)
ewen@rustdev:~$ 
```

Updated to 2.2.1 a couple of days ago, as above:

```
ewen@rustdev:~$ sudo lsusb -v -d 2e8a:000c | egrep '2e8a:000c|idProduct|idVendor|bcdDevice|iManufacturer|iProduct'
Bus 002 Device 028: ID 2e8a:000c Raspberry Pi Debug Probe (CMSIS-DAP)
  idVendor           0x2e8a 
  idProduct          0x000c 
  bcdDevice            2.21
  iManufacturer           1 Raspberry Pi
  iProduct                2 Debug Probe (CMSIS-DAP)
ewen@rustdev:~$ 
```

The code example in the previous comment suggests the version is being encoded as X.Y.Z (each 4 bits, ie BCD == Binary Coded Decimal) by Raspberry Pi, but `lsusb` is printing it as just major/minor.

That seems to [confirm my guess](https://github.com/probe-rs/probe-rs/issues/2950#issuecomment-2613816950) that the factory shipped version is the [picoprobe-cmsis-v1.01](https://github.com/raspberrypi/debugprobe/releases/tag/picoprobe-cmsis-v1.0.1) release (November 2022).

Ewen

PR #3252 (2025-06-22): Reject old RPi debugprobes
Closes #2950 as the 2.22 firmware seems to work just fine*.
closed 2025-06-22 by bugadani