probe-rs/probe-rs #2953: There should be more error info

0xorial, 2024-12-12:
**Describe the bug**
I got a promicro nrf52840 board and daplink board. soldered, connected. They are detected, but anything else just gives a generic error:

```
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.13s
     Running `probe-rs run --chip nrf52840_xxAA target/thumbv7em-none-eabihf/debug/central`
Error: Failed to open the debug probe.

Caused by:
    0: An error which is specific to the debug probe in use occurred.
    1: Error handling CMSIS-DAP command Info
    2: Error in the USB access
```

**To Reproduce**
see above

**Expected behavior**
Any hints whatsoever to what's going on.

**Stacktrace**
I wish there was...

**Desktop (please complete the following information):**
- macOS Mojave

Yatekii, 2024-12-12:
Something seems to be off with your probe. Can you enable `RUST_LOG=trace` and upload the trace to this issue?

0xorial, 2024-12-12:
@Yatekii 

[logs.txt](https://github.com/user-attachments/files/18118386/logs.txt)

0xorial, 2024-12-12:
In fairness, the issue I described is solved because I do see some error information now, after setting RUST_LOG=true. But would be still to have an idea what's going wrong...

Yatekii, 2024-12-14:
Seems that the first USB transfer just fails with no further info. We would have to print some more error info around there.

0xorial, 2024-12-14:
@Yatekii thank you for having a look. Any quick leads where I can already start looking myself?

Yatekii, 2024-12-14:
So this happens in the CMSIS-DAP `Info` command (https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/cmsisdap/commands/general/info.rs) but I am not sure exactly which variant. I think it is the one to determine the packet size.
You could check where that is sent to the wire (USB) and add additional printing of eventual return values of the USB stack. If there is none, you might have to patch the USB stack (`nusb`). But my guess is that the probe hangs itself somehow when we send a command it does not like earlier and then gets stuck and does not reply to our command.
Maybe we need to just reset the USB device before we do things. I thought we had a mechanism like that in place in early days but with more stability we might have removed it.

kevinmehall, 2024-12-14:
The

```
 INFO async_io::block_on: nusb::platform::macos_iokit::transfer: Cancelled all transfers on endpoint 82. status=0    
 INFO nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600003ae7460, status=e00002eb, len=0 
```

looks like your timeout triggering and canceling the transfer.

Could add a `#[from]` [here](https://github.com/probe-rs/probe-rs/blob/ecb095c8cdf39d0a13dc9156de6c49765fce6483/probe-rs/src/probe/cmsisdap/commands/mod.rs#L65) for more (but not necessarily better) error detail.

0xorial, 2024-12-25:
After some debugging I narrowed the issue down to the `nusb` library calling `ReadPipeAsync`, but never receiving a callback all the way until timeout fires and `AbortPipe` is called because the handles are dropped. After `AbortPipe` is called I see the packet from device to host in wireshark, AND the callback in the code is called (but of course the handle is abandoned by now). Anyone has any idea what could be going wrong here?

kevinmehall, 2024-12-26:
That's consistent with the device not sending anything before the timeout -- either the device is stuck or the timeout is too short.

Assuming you're using a software USB analyzer, seeing the transfer complete immediately after `AbortPipe` is expected -- it will have an error status that is probably `kIOReturnAborted`.

0xorial, 2024-12-26:
Ah, thanks, that saved me from the effort of trying to reproduce it directly via nusb.

At this point I am not sure what to explore further, I see a total of 21 pairs of exchanges in wireshark when i connect the device (descriptor and such), and then 3 pairs when I try anything with probe-rs, like `probe-rs info`. Out of last 3, 2 indeed have a status kIOReturnAborted, and 1 has kIOReturnSuccess. I attach the wireshark capture here, in case any would be so kind to have a look.[usb_capture.pcapng.gz](https://github.com/user-attachments/files/18255073/usb_capture.pcapng.gz)
[usb_capture.pcapng.zip](https://github.com/user-attachments/files/18255074/usb_capture.pcapng.zip)

elfmimi, 2024-12-31:
I suppose PR #3002 will change the situation a bit for this issue.
Without the fix, it was bailed out by the first timeout with no retrying. and the cause was displayed as a rather ambiguous message "Error in the USB access".

With the fix, if it actually fails after waiting for some 16 seconds, it will show this message.
```
Error: Failed to open the debug probe.

Caused by:
    0: An error which is specific to the debug probe in use occurred.
    1: Could not determine a suitable packet size for this probe.
```