probe-rs/probe-rs #2969: 0.25 STM32U0 regression

diondokter, 2024-12-18:
### Describe the bug

Hi all,

this used to work before just fine, but I'm now getting a `SwdApFault` error after flashing is finished on version 0.25.
This is for the U0 nucleo64 dev board with the on-board STLink v2.1 programmer.
Updating the Stlink firmware did not change anything.

It seems to do with reading data for RTT.
Maybe a timeout was shortened or some process got faster so it's now catching the AP at a bad time?

### To Reproduce

1. Run `probe-rs run --chip stm32u083rctx` as the runner in `config.toml`
2. Call `cargo run --release`
3. See the erase and flash succeed
4. Error:
```
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApFault
Error: Error communicating with the probe.

Caused by:
    0: An ARM specific error occurred.
    1: The debug probe encountered an error.
    2: An error which is specific to the debug probe in use occurred.
    3: Command failed with status SwdApFault.
```

This can be fixed by lowering the speed:
1. `probe-rs run --chip stm32u083rctx --speed 950`
2. Everything succeeds and RTT/Defmt logs can be seen

### Stacktrace

```shell
Erasing ✔ 100% [####################]  52.00 KiB @  38.24 KiB/s (took 1s)
  Programming ✔ 100% [####################]  51.00 KiB @  13.88 KiB/s (took 4s)                                                          
     Finished in 3.68s
DEBUG probe_rs::architecture::arm::ap::memory_ap: reading IDR: IDR { REVISION: 8, DESIGNER: JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")), CLASS: MemAp, _RES0: 0, VARIANT: 0, TYPE: AmbaAhb3 }
DEBUG probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: true, HNONSEC: false, MasterType: true, Allocate: false, Cacheable: true, Bufferable: false, Privileged: true, Data: true, SPIDEN: true, TrInProg: false, DeviceEn: true, AddrInc: Single, Size: U32, _reserved_bits: 0 }
DEBUG reset_and_halt: probe_rs::architecture::arm::core::armv6m: Reason for halt has changed, old reason was Halted(Breakpoint(Unknown)), new reason is Exception timeout=100ms
DEBUG probe_rs::architecture::arm::ap::memory_ap: reading IDR: IDR { REVISION: 8, DESIGNER: JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")), CLASS: MemAp, _RES0: 0, VARIANT: 0, TYPE: AmbaAhb3 }
DEBUG probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: true, HNONSEC: false, MasterType: true, Allocate: false, Cacheable: true, Bufferable: false, Privileged: true, Data: true, SPIDEN: true, TrInProg: false, DeviceEn: true, AddrInc: Single, Size: U32, _reserved_bits: 0 }
DEBUG probe_rs::rtt: Scanning at exact address: 0x20000000
DEBUG probe_rs::util::rtt::client: RTT control block corrupted (write pointer is 1217181298 while buffer size is 1174648080 for channel 0 (☺$$♦ @))
DEBUG probe_rs::cmd::run: Cleared RTT header
DEBUG probe_rs::architecture::arm::ap::memory_ap: reading IDR: IDR { REVISION: 8, DESIGNER: JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")), CLASS: MemAp, _RES0: 0, VARIANT: 0, TYPE: AmbaAhb3 }
DEBUG probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: true, HNONSEC: false, MasterType: true, Allocate: false, Cacheable: true, Bufferable: false, Privileged: true, Data: true, SPIDEN: true, TrInProg: false, DeviceEn: true, AddrInc: Single, Size: U32, _reserved_bits: 0 }
DEBUG probe_rs::architecture::arm::core::armv6m: Cached halt reason: Halted(Exception)
DEBUG probe_rs::architecture::arm::core::armv6m: Cached halt reason: Halted(Exception)
DEBUG run: probe_rs::architecture::arm::core::armv6m: Reason for halt has changed, old reason was Halted(Exception), new reason is Request
DEBUG probe_rs::rtt: Scanning at exact address: 0x20000000
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApFault
DEBUG session_drop: probe_rs::architecture::arm::ap::memory_ap: reading IDR: IDR { REVISION: 8, DESIGNER: JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")), CLASS: MemAp, _RES0: 0, VARIANT: 0, TYPE: AmbaAhb3 }
DEBUG session_drop: probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: true, HNONSEC: false, MasterType: true, Allocate: false, Cacheable: true, Bufferable: false, Privileged: true, Data: true, SPIDEN: true, TrInProg: false, DeviceEn: true, AddrInc: Single, Size: U32, _reserved_bits: 0 }
 INFO session_drop: probe_rs::session: Halting core 0...
 INFO session_drop: probe_rs::session: Clearing breakpoints for core 0
DEBUG session_drop: probe_rs::architecture::arm::ap::memory_ap: reading IDR: IDR { REVISION: 8, DESIGNER: JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")), CLASS: MemAp, _RES0: 0, VARIANT: 0, TYPE: AmbaAhb3 }
DEBUG session_drop: probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: true, HNONSEC: false, MasterType: true, Allocate: false, Cacheable: true, Bufferable: false, Privileged: true, Data: true, SPIDEN: true, TrInProg: false, DeviceEn: true, AddrInc: Single, Size: U32, _reserved_bits: 0 }
DEBUG session_drop: probe_rs::session: Resuming core...
DEBUG session_drop: probe_rs::architecture::arm::ap::memory_ap: reading IDR: IDR { REVISION: 8, DESIGNER: JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")), CLASS: MemAp, _RES0: 0, VARIANT: 0, TYPE: AmbaAhb3 }
DEBUG session_drop: probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: true, HNONSEC: false, MasterType: true, Allocate: false, Cacheable: true, Bufferable: false, Privileged: true, Data: true, SPIDEN: true, TrInProg: false, DeviceEn: true, AddrInc: Single, Size: U32, _reserved_bits: 0 }
DEBUG session_drop:run: probe_rs::architecture::arm::core::armv6m: Reason for halt has changed, old reason was Halted(Request), new reason is Request
DEBUG session_drop: probe_rs::architecture::arm::ap::memory_ap: reading IDR: IDR { REVISION: 8, DESIGNER: JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")), CLASS: MemAp, _RES0: 0, VARIANT: 0, TYPE: AmbaAhb3 }
DEBUG session_drop: probe_rs::architecture::arm::ap: Writing register CSW, value=CSW { DbgSwEnable: true, HNONSEC: false, MasterType: true, Allocate: false, Cacheable: true, Bufferable: false, Privileged: true, Data: true, SPIDEN: true, TrInProg: false, DeviceEn: true, AddrInc: Single, Size: U32, _reserved_bits: 0 }
DEBUG probe_rs::probe::stlink: Current device mode: Jtag
Error: Error communicating with the probe.
```

### Operating System

Windows

### Additional context

Project on which I'm encountering this: https://github.com/diondokter/s2lp/tree/351a00b3f5fc0289e7ed07f2d1c5ff3e253f2a3d/examples

Yatekii, 2024-12-19:
Could you try and bisect this please? :)

matoushybl, 2024-12-19:
Did you try `--connect-under-reset`? A colleague of mine reported a similar regression with H7 to me, but I haven't been able to reproduce, but I was only able to flash when using `--connect-under-reset` for the first time, then everything worked well without the flag on both `0.24.0` and `0.25.0`.

diondokter, 2024-12-19:
> Did you try `--connect-under-reset`? A colleague of mine reported a similar regression with H7 to me, but I haven't been able to reproduce, but I was only able to flash when using `--connect-under-reset` for the first time, then everything worked well without the flag on both `0.24.0` and `0.25.0`.

Yes,  connect under reset doesn't change anything. Also note that flashing works just fine!

diondokter, 2024-12-19:
> Could you try and bisect this please? :)

I'll do when I have time!
Is there any specific probe-rs tooling for this? Or is doing it manually the best option?

Yatekii, 2024-12-19:
I don't know of any tooling :( Maybe we should add some tooling for this :) But truth be told, dependent on command you might need to change a lot.
I think git bisect should be able to take a shellscript and dependent on status code find out by itself. Since the command surface in probe-rs did not change since 0.24.0 for what you do, you should be able to do it this way :)

tlevora, 2024-12-20:
Having pretty similar issue with STM32H7 and a particular binary. After `probe-rs erase --chip STM32H753ZITx` followed with powercycle, flash and run my binary is successfull. Another flashing is OK, but then I got the same error message:

```
WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApFault
Error: Error communicating with the probe.
```

If I do not powercycle after reset and do just SoC hw reset, the issue persists.

My app has following memory:
```
MEMORY
{
  FLASH  : ORIGIN = 0x08000000, LENGTH = 2048K
  DTCM   : ORIGIN = 0x20000000, LENGTH = 128K
  RAM    : ORIGIN = 0x24000000, LENGTH = 512K
  ...
}

_stack_start = ORIGIN(DTCM) + LENGTH(DTCM);
_stack_end = ORIGIN(DTCM);
```

Basically all the `data` including `defmt` section is located within `RAM` memory. After moving all the data into `DTCM` memory region, `probe-rs` the issue disappears.

tlevora, 2024-12-20:
Adding another observation. My app implements a custom dynamic memory allocator which is not used at all yet, it is implemented only. Commenting just a single line out makes my app running.

```rust
#[global_allocator] // Commenting this line out makes the probe-rs execute my app
static ALLOCATOR: BumpAllocator<SIZE> = BumpAllocator::new(); // my implementation of allocator
```

diondokter, 2024-12-20:
Interesting! However the firmware I'm finding this in is quite simple so I'm not sure if this is related.
It could be if it's timing related though

diondokter, 2024-12-20:
Alright, a git bisect later and the single commit in this PR is the first bad version: https://github.com/probe-rs/probe-rs/pull/2883

diondokter, 2024-12-20:
Probe-rs still compiles if that commit is reverted on the current master. If I do that, I don't see any problems anymore.

I'm not familiar with this code section in probe-rs, so it's hard for me to say why the new PR is wrong...

bugadani, 2024-12-20:
The theory of it was fine, so I guess it's beyond me as well :) We can revert it, there's nothing to gain by removing 7 words of magic.

diondokter, 2024-12-20:
The change probe-rs made is not exactly the same as why pyOCD did. Probe-rs made it a double bkpt instruction while pyOCD did a bkpt and then branch back to the previous instruction.

I tried out what pyOCD does, but it makes it hang. So BKPT+BKPT = crash, BKPT+b.-2 = hang.
Something isn't working here... Maybe pyOCD does something extra?

I'm gonna play around with this a bit.

diondokter, 2024-12-20:
Hmmm, ok weird realization. I removed the blob alltogether so no instructions are run. That makes it time out. I guess probe-rs waits on the breakpoint signal to know the flash program has started. Fair. Logical.

But when it goes wrong for me, that's after flashing. Why would a header for the flashing algo affect the running of the program?
Maybe the flash algo for the U0 resets to the start after flashing?

diondokter, 2024-12-20:
Ha, if I replace it all with 1 bkpt instruction and then 15 nops it also works :D
Now let's see if it's the size that is important or the delay

diondokter, 2024-12-20:
Ok, smallest I can get it is 3 words:

```asm
bkpt
nop
nop
nop
nop
nop
```

I have no idea why adding the nops would do anything.

---
*After more testing*

The nops aren't even executed. I replaced them with 'udf' and everything works.

bugadani, 2024-12-20:
Just for fun, what happens if you remove the `load_address`of the flash algo?

diondokter, 2024-12-20:
Oh yes! That works!

The load_address was set to 0x2000_0020. Without it set it picks 0x2000_0000.

When it's set in the yaml, it subtracts the size of the header from it.
So why would it _**not**_ work when the used load_address is 0x2000_0018, but does work when it's 0x2000_0014 or lower?

diondokter, 2024-12-20:
Hmmmm, the return address that's set for the called flash functions is always set to the load address: https://github.com/probe-rs/probe-rs/blob/5e006737e70206a9852c9f3bcc656294f245987a/probe-rs/src/flashing/flasher.rs#L691-L700

So I don't see why a change in load address would matter...

bugadani, 2024-12-21:
If your MCU ends up halted after a flash, you can try `probe-rs read`-ing back the first few bytes, just to see if they are still the expected value. Could be that we miscalculate something, and IDK, accidentally set the stack pointer to the original load_address, or some such. Unlikely, but I haven't touched this code in a while... Some debug logging would help us to see if this is a possibility or not.

diondokter, 2024-12-21:
Ok, other observation:

The header is stored anywhere inbetween 0x2000_0000 and 0x2000_0020 depending on the size of the header.
Stuff seems to break around the RTT mechanism.
In my firmware the RTT block is located at 0x2000_0000.

On the master branch this is logged:
```
DEBUG probe_rs::rtt: Scanning at exact address: 0x20000000
DEBUG probe_rs::util::rtt::client: RTT control block corrupted (write pointer is 1217181298 while buffer size is 1174648080 for channel 0 (☺$$♦ @))
DEBUG probe_rs::cmd::run: Cleared RTT header
...
DEBUG probe_rs::rtt: Scanning at exact address: 0x20000000
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApFault
```

If I 'fix' the header by making it longer, it changes to:
```
DEBUG probe_rs::rtt: Scanning at exact address: 0x20000000
DEBUG probe_rs::util::rtt::client: RTT control block corrupted (Unexpected array sizes at 0x20000000: max_up_channels=1 max_down_channels=3187719680)
DEBUG probe_rs::cmd::run: Cleared RTT header
...
DEBUG probe_rs::rtt: Scanning at exact address: 0x20000000
DEBUG probe_rs::util::rtt::client: RTT control block corrupted (Unexpected array sizes at 0x20000000: max_up_channels=1 max_down_channels=3187719680)
DEBUG probe_rs::rtt: Scanning at exact address: 0x20000000
0.000000 INFO  Initializing microcontroller  ///// <= Defmt logs start
```

In both cases the RTT is corrupted, which is expected I guess.
But WHY it's corrupted is different. Also, `3187719680` is the machine code for `BKPT`.

It seems like the response to both types of corruption is the same by clearing the RTT header.
On master the second try crashes. On my fix version it still reads corrupted even though it should've been set to all 00's. But on the fix version I think the mcu program starts at some point which writes the proper RTT header which probe-rs can then attach to.

Any idea what could be happening here? Why does the header clear not work?

closed 2024-12-21 by bugadani

tlevora, 2024-12-22:
The problem still persists under some conditions.

My SoC has flashing algorithm header loaded from address `2400_0008h`, that header is immediately followed with the flashing algorithm. This memory region is in a collision with RTT header of my program, which means that flashing partially rewrites that RTT header.

Method `RttClient::clear_control_block` called from run cmd ends with an error because of a rewritten pointer to rtt channel when trying to construct the channel in `rtt::Channel::from`.

I guess we need to invalidate RTT because of following cases:

- The probe will look for it between SoC reset and `.data` section initialization
- Newly flashed program does not contain RTT at all

Since the RTT is being identified based on `'SEGGER RTT\0\0\0\0\0\0'` magic, I think there are basically two solutions to this issue:

- Clear the `SEGGER RTT` magic only if found
- Clear the RTT header before flashing (potentially replacing it with flashing algo)

Honestly I would prefer just to search for the magic id and clear it if found because we are pretty sure the control block is corrupted when calling `clear_control_block`. But not to mess up the code I proposed a [fix](https://github.com/probe-rs/probe-rs/pull/2981) which propagates proper errors from `Rtt::from` plus I have moved the RTT block clear before flashing procedure not to access wrong memory. If you would prefer different solution, just give me know, will try to fix it.

PR #2978 (2024-12-21): Added extra checks for RTT control block integrity
Added extra checks for RTT control block integrity so partially trashed blocks are now more likely to be rejected as corrupted.

Fixes #2969 

This was a fun one to figure out! Learned a lot about the flashing parts of probe-rs.
It was a coincidence that 0.24 worked. Here's the rundown:

- Device had already run and so has an RTT block at 0x2000_0000
- Target yaml specifies a load_address of 0x2000_0020
- Flash header and instructions are written to RAM
  - In 0.24 arm flash header was 32 bytes and was thus put at 0x2000_0000
  - In 0.25 arm f [... cut]