probe-rs/probe-rs #2977: Regression using BMP on nRF52

chrysn, 2024-12-21:
### Describe the bug

Updating from the original Black Magic Probe PR branch (#2791) to 0.25 I found that my setup did not work any more.

Bisection showed that the regression was introduced in 5b0f5db39d5e13b55b1568e5c8695721e18a48de (#2924).

### To Reproduce

1. Attach a Black Magic Probe via USB
2. Attach an nRF52840 to the BMP (I'm using a particle-xenon board, but that should not make any difference)
3. `probe-rs run --protocol=swd --chip nrf52840_xxAA ../../build/bin/particle-xenon/coap/cargo/thumbv7em-none-eabi/release/coap` (I don't have a minimal binary to show, but given things don't get very far, I trust that you'll have something sufficiently suitable at hand)

Result:
```
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: Failed to find or attach to the target. Please check the wiring before retrying.
```

### Expected behavior

Progress bars "Erasing", "Programming" etc.

### Stacktrace

```shell

```

### Operating System

Linux (Debian)

### Additional context

Across the version where running fails, there is no difference in the output of `probe-rs info`; in both cases, it is

```
Probing target via JTAG

ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv1, DP Designer: ARM Ltd
├── 0 MemoryAP (AmbaAhb3)
│   └── ROM Table (Class 1), Designer: Nordic VLSI ASA
│       ├── Cortex-M4 SCS   (Generic IP component)
│       │   └── CPUID
│       │       ├── IMPLEMENTER: ARM Ltd
│       │       ├── VARIANT: 0
│       │       ├── PARTNO: Cortex-M4
│       │       └── REVISION: 1
│       ├── Cortex-M3 DWT   (Generic IP component)
│       ├── Cortex-M3 FBP   (Generic IP component)
│       ├── Cortex-M3 ITM   (Generic IP component)
│       ├── Cortex-M4 TPIU  (Coresight Component)
│       └── Cortex-M4 ETM   (Coresight Component)
└── 1 Unknown AP (Designer: Nordic VLSI ASA, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x0)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

chrysn, 2024-12-21:
The good thing is that the "breaking" commit is a very simple one -- the bad thing is that looking at it I still don't know *why* it breaks.

I could confirm that of the only place introduced there that could really make a difference, which is

```patch
+        if probe.scan_chain().iter().len() > 0 {
+            for core in &cores {
+                probe.select_jtag_tap(core.interface_idx())?;
+            }
+        }
```

we *do* really leave the flow there at the `?`, but I don't know enough of the context to understand *why* that is. I'm a bit confused why we even get into anything JTAG related, given that I explicitly selected SWD … maybe that check should only run if JTAG is used. The error in `probe-rs info` indicates that something is really afoul on the JTAG side -- maybe that's a bug with BMP, but even then, the code might be a bit defensive here. (Even if it *is* a bug in the BMP implementation, maybe probing JTAG on some target could fail on some targets even with a perfect debugger?).

Yatekii, 2024-12-21:
I think its a bug that we do this when you are on SWD. Seems like some check fails somewhere.

drahnr, 2024-12-24:
I can confirm the above issue and can repro if more data is needed

chrysn, 2025-01-13:
Maybe @robamu (author of #2924 that introduced the regression) could help?

robamu, 2025-01-14:
I'd have to look at it in more detail, but I  can tell you that I added this specific code so the correct JTAG tap is selected in case a custom JTAG tap index should be used.

robamu, 2025-01-14:
One option worth trying would be this change:

```
        if target.jtag.is_some() && probe.scan_chain().iter().len() > 0 {
            for core in &cores {
                probe.select_jtag_tap(core.interface_idx())?;
            }
        }
```

Other than that, I don´t have the hardware.. What I did to debug problems like this is that I build a custom application and use a locally cloned probe-rs library. That way you could step through the code and figure out what exactly is going wrong.
Some docs can be found here: https://docs.rs/probe-rs/latest/probe_rs/flashing/index.html (older version though, but you could build the docs locally with the cloned version)

xobs, 2025-01-18:
This is fixed in #3030

xobs, 2025-01-19:
The issue is that it was unconditionally selecting a JTAG tap, even in SWD (or unspecified) mode.

The BMP implementation would return `&[]` as the scan chain in such a case, whereas the CMSIS-DAP would return an error. One solution would be to not select a JTAG TAP if it's not in JTAG mode, but I fixed it by having BMP behave the same as CMSIS-DAP.

Please give the latest master branch a try.

Yatekii, 2025-02-22:
I will mark this as done, please reopen if needed.

closed 2025-02-22 by Yatekii