probe-rs/probe-rs #2988: vscode plugin prompts no-semihosting error

eighteen-k-gold-malow, 2024-12-25:
### Describe the bug

I am using st-link. Why does the probe-rs plugin of vscode use F5 debugging and prompt a `no-semihosting` error when it encounters `hprintln!`, but I can print text normally when I use `probe-rs run --chip stm32f103c8` to execute `hprintln!`

### my code
```rust
hprintln!("I2C write failed: {:?}", e)
```
### config
```json
{
    "version": "0.2.0",
    "configurations": [

        {
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "probe-rs Test",
            "cwd": "${workspaceFolder}",
            "connectUnderReset": true,
            "chip": "stm32f103c8",
            "probe": "0483:3748",
            "flashingConfig": {
                "flashingEnabled": true,
                "haltAfterReset": false
            },
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "./target/thumbv7m-none-eabi/debug/${workspaceFolderBasename}"
                }
            ]
        }
    ]
}
```

### Stacktrace

```shell

```

### Operating System

None

bugadani, 2024-12-28:
Why did you close this issue, did you figure something out?

eighteen-k-gold-malow, 2024-12-28:
> Why did you close this issue, did you figure something out?

I raised this issue with the vscode plugin project, but they don't seem to be willing to deal with it.

bugadani, 2024-12-28:
Ah okay, thanks. Just for tracking purposes, the plugin-side issue is https://github.com/probe-rs/vscode/issues/109

Right now, the recommendation against semihosting is a valid one, we mostly added semihosting prints for embedded-test (`probe-rs run` printing the messages is a side-effect of that) and the code around it is pretty much terrible. But it would definitely be nice if we could print output to `cargo embed` and the debugger.

Yatekii, 2024-12-28:
I guess it's an UX question how we can both support it but also nudge the user nicely towards defmt/RTT.

Sorry for the lack of clarity in my comment @eighteen-k-gold-malow.

closed 2025-05-26 by Yatekii