probe-rs/probe-rs #2989: Can't flash or read full memory for STM32L452RETx

viperidae, 2024-12-27:
### Describe the bug

Anything but reading small amounts of memory consistently fails.
Using the built in ST Link on a Nucleo-64 L452RE board.
Used ST Link application on Windows to update STLink to latest firmware.

probe-rs runs on ubuntu in a VM
```
boxuser@linux:~/projects/my-embedded-rust-project$ cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.79s
     Running `probe-rs run --chip STM32L452RETx target/thumbv7em-none-eabi/debug/my-embedded-rust-project`
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: The debug probe encountered an error.
    2: An error which is specific to the debug probe in use occurred.
    3: USB error.
    4: timed out
```
It's not always the same error
```
vboxuser@linux:~/projects/my-embedded-rust-project$ cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.05s
     Running `probe-rs run --chip STM32L452RETx target/thumbv7em-none-eabi/debug/my-embedded-rust-project`
 WARN probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    0: The debug probe encountered an error.
    1: An error which is specific to the debug probe in use occurred.
    2: USB error.
    3: timed out                                                                                                                                                                        Error: The flashing procedure failed for 'target/thumbv7em-none-eabi/debug/my-embedded-rust-project'.

Caused by:
    0: The initialization of the flash algorithm failed.
    1: Something during the interaction with the core went wrong
    2: Failed to write CPU register R2/a3.
    3: An ARM specific error occurred.
    4: The debug probe encountered an error.
    5: An error which is specific to the debug probe in use occurred.
    6: USB error.
    7: timed out
```

Reading small amounts of memory is fine
```
vboxuser@linux:~/projects/my-embedded-rust-project$ probe-rs read --protocol swd --chip stm32l452retx b8 0x08000000 50
00 00 02 20 95 01 00 08 33 7f 00 08 e9 be 00 08 33 7f 00 08 33 7f 00 08 33 7f 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 33 7f 00 08 33 7f 
```
Most times reading 5kb works
Most times reading 50kb fails
```
vboxuser@linux:~/projects/my-embedded-rust-project$ probe-rs read --protocol swd --chip stm32l452retx b8 0x08000000 50000
Error: An ARM specific error occurred.

Caused by:
    0: The debug probe encountered an error.
    1: An error which is specific to the debug probe in use occurred.
    2: USB error.
    3: timed out
```

Flashing also fails
```
vboxuser@linux:~/projects/my-embedded-rust-project$ cargo flash --chip stm32l452retx
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.05s
    Flashing /home/vboxuser/projects/my-embedded-rust-project/target/thumbv7em-none-eabi/debug/my-embedded-rust-project
 WARN probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    0: The debug probe encountered an error.
    1: An error which is specific to the debug probe in use occurred.
    2: USB error.
    3: timed out                                                                                                                                                                         WARN probe_rs::session: Failed to deconfigure device during shutdown: Arm(Probe(ProbeSpecific(BoxedProbeError(Usb(Kind(TimedOut))))))                                                         Error The initialization of the flash algorithm failed.

  Caused by:  
          0: The initialization of the flash algorithm failed.
          1: Failed to start running code on the CPU
          2: An ARM specific error occurred.
          3: The debug probe encountered an error.
          4: An error which is specific to the debug probe in use occurred.
          5: USB error.
          6: timed out
```

It's usually always some kind of timed out error

### Stacktrace

```shell

```

### Operating System

Linux (Ubuntu)

Yatekii, 2024-12-27:
I don't have such a chip here, but you are running this in a VM which can theoretically introduce additional delays. Can you try outside of the VM?

viperidae, 2024-12-28:
> I don't have such a chip here, but you are running this in a VM which can theoretically introduce additional delays. Can you try outside of the VM?

While I don't currently have any bare metal running linux to try this out on, I have just tried the linux version of STM32CubeProgrammer in the VM and it works perfectly fine.

Also tried a Nucleo with an STM32F303RE , and it has the same USB timeout problems with probe-rs.

STM32CubeProgrammer
```
  04:20:47 : STM32CubeProgrammer API v2.18.0 | Linux-64Bits 
  04:20:55 : UR connection mode is defined with the HWrst reset mode
  04:20:56 : ST-LINK SN  : 066FFF564948897767133527
  04:20:56 : ST-LINK FW  : V2J37M26
  04:20:56 : Board       : NUCLEO-F303RE
  04:20:56 : Voltage     : 3.24V
  04:20:56 : SWD freq    : 4000 KHz
  04:20:56 : Connect mode: Normal
  04:20:56 : Reset mode  : Software reset
  04:20:56 : Device ID   : 0x446
  04:20:56 : Revision ID : Rev Y
  04:20:56 : Debug in Low Power mode is not supported for this device.
  04:20:56 : UPLOADING OPTION BYTES DATA ...
  04:20:56 :   Bank          : 0x00
  04:20:56 :   Address       : 0x1ffff800
  04:20:56 :   Size          : 16 Bytes
  04:20:56 : UPLOADING ...
  04:20:56 :   Size          : 1024 Bytes
  04:20:56 :   Address       : 0x8000000
  04:20:56 : Read progress:
  04:20:56 : Data read successfully
  04:20:56 : Time elapsed during the read operation is: 00:00:00.024
  04:23:22 : Opening and parsing file: my-embedded-rust-project.elf
  04:23:22 : Memory Programming ...
  04:23:22 :   File          : my-embedded-rust-project.elf
  04:23:22 :   Size          : 49.89 KB 
  04:23:22 :   Address       : 0x08000000 
  04:23:22 : Erasing memory corresponding to segment 0:
  04:23:23 : Erasing internal memory sectors [0 24]
  04:23:25 : Download in Progress:
  04:23:28 : File download complete
  04:23:28 : Time elapsed during download operation: 00:00:05.438
```

probe-rs on the same VM
```
$ cargo run
   Compiling stm32-metapac v15.0.0
   Compiling my-embedded-rust-project v0.1.0 (/home/vboxuser/projects/my-embedded-rust-project)
   Compiling embassy-stm32 v0.1.0
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 8.61s
     Running `probe-rs run --chip STM32F303RE target/thumbv7em-none-eabi/debug/my-embedded-rust-project`
      Erasing ✔   0% [--------------------]
  Programming ✔ 100% [--------------------]                                                                                                             Error: The flashing procedure failed for 'target/thumbv7em-none-eabi/debug/my-embedded-rust-project'.

Caused by:
    0: Failed to erase flash sector at address 0x08000000.
    1: Failed to read the core status.
    2: An ARM specific error occurred.
    3: The debug probe encountered an error.
    4: An error which is specific to the debug probe in use occurred.
    5: USB error.
    6: timed out
```

viperidae, 2024-12-28:
Installed probe-rs on Windows, it works.

Interesting that STM32CubeProgrammer works perfectly fine in the VM but probe-rs doesn't. Maybe some of the timeouts could be tweaked a bit?

Yatekii, 2025-02-22:
Indeed, we can tweak some of the timeouts. Do you have the capacity to try and tweak this on your system? Otherwise I will try and do this soon-ish.