probe-rs/probe-rs #2995: Error: Failed to open the debug probe.

elfmimi, 2024-12-29:
### Describe the bug

There are some cases where probe-rs fails to open a connection to CMSIS-DAP debug probes.

when it fails it gives result like this.
```
Error: Failed to open the debug probe.

Caused by:
    0: An error which is specific to the debug probe in use occurred.
    1: Error handling CMSIS-DAP command Info.
    2: Error in the USB access.
```

The probability of failure seems to vary widely.

### To Reproduce

As far as I can tell, it is observable only with specific setup.
like this.
1. ASMedia USB host controller ASM1142 (in my case)
2. Use a USB Hub
3. USB Full-Speed CMSIS-DAP debug probe

`probe-rs info` will give the above result once in several attempts or more.

the behavior is not limited to `info` command.
for example `probe-rs reset --chip cortex-m0` will also give the same result with presumably the same failure rate.

Debug probes tested:
RaspberryPi Pico + [DebugProbe v2.2.0](https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.2.0)  debugprobe_on_pico.uf2
LPC11u35 Board + [DAPLink v257](https://github.com/ARMmbed/DAPLink/releases/tag/v0257) 0257_lpc11u35__0x0000.bin

No target device needed to be connected.

OS: Windows 10 Pro 10.0.19045
Intel based PC

### Expected behavior

no failure.

### Stacktrace

```shell

```

### Operating System

Windows

### Additional context

probe-rs 0.22.0 do not exhibit this behavior.

elfmimi, 2024-12-29:
I've already found a solution to this bug in two ways.

**one**) straightforward quick patch
https://github.com/elfmimi/probe-rs/commit/e75d3c4c834ace16e6f662cb4b57e0b9b124c687

**two**) trickier but in line with what it was before
https://github.com/elfmimi/probe-rs/commit/fc4dcb859029a66909d9a6c56b609dbb13fef5b4

which is preferred? shall I create a pull request?

bugadani, 2024-12-29:
I think I prefer 2 and a PR would be appreciated :)

PR #3002 (2024-12-31): Fix regression in USB timeout handling
Fixes #2995

The regression was introduced with PR #1842, when we moved from **rusb** to **nusb**.

I have only tested it on Windows.
closed 2024-12-31 by Yatekii