probe-rs/probe-rs #2996: "info" command not working 99% of the time with J-Link + RP2040 + Windows 11

DualTachyon, 2024-12-29:
### Describe the bug

Most of the time it either finds nothing and sometimes reports errors. For example:

```
C:\Storage\GIT\probe-rs\target\debug>probe-rs.exe info --protocol swd --chip RP2040
Probing target via SWD

ARM Chip with debug port Default:
No access ports found on this chip.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

However if I run "gdb" first and then Control C, the first invocation of "info" will work partially (only 1 core is reported) and the next invocation will report nothing as above.

I have had it report the 2 cores maybe once every "zillion" attempts. There seems to be some bad state persisting on the J-Link.

### To Reproduce

1. "probe-rs.exe info --protocol swd --chip RP2040"
2. Observe nothing is found
3. "probe-rs.exe gdb --protocol swd --chip RP2040"
4. In a separate window, run arm-none-eabi-gdb and connect with "target remote localhost:1337"
5. Control-C on probe-rs.exe
6. "probe-rs.exe info --protocol swd --chip RP2040"
7. Observe one core is found
8. "probe-rs.exe info --protocol swd --chip RP2040"
9. Observe nothing is found.

### Expected behavior

I would expect both cores to be found every time.

### Stacktrace

```shell

```

### Operating System

Windows

### Additional context

I attached 4 captures of probe-rs with RUST_LOG=probe_rs::probe=trace. I used the latest git tree as some updates were added for RP2040.
My J-Link is an EDU unit based on V10 hardware, with the latest firmware.

log1.txt is running "info" without success
log2.txt is running "gdb" and having gdb.exe doing "target remote localhost:1337" and quitting immediately
log3.txt is running "info" with partial success
log4.txt is running "info" again, without success.

Log 1 to 4 were all produced in sequence within seconds of each other.

I was hoping to use probe-rs with a STM32MP257 development board. While probe-rs doesn't have a YAML profile for it, I was also hoping to write one myself and decided to try using a RP2040 first. The STM32MP257 is also a multi-drop SWD device and the results there are often worse. I had hoped "info" would be able to at least enumerate the ROM table.

[log1.txt](https://github.com/user-attachments/files/18269709/log1.txt)
[log2.txt](https://github.com/user-attachments/files/18269708/log2.txt)
[log3.txt](https://github.com/user-attachments/files/18269711/log3.txt)
[log4.txt](https://github.com/user-attachments/files/18269710/log4.txt)

Tiwalun, 2024-12-29:
Multi-drop SWD is not really meant to be discoverable, the targetsel values for the RP2040 are hardcoded in the info command, so they somewhat work.

If you select the protocol explicitly using `--protocol swd`, and specify the `TARGETSEL` value using `--target-sel`, do you get better results?

Tiwalun, 2024-12-29:
The `--chip` command is unfortunately ignored by the `info` command, the info command was mostly meant to discover information about "unknown" chips.

DualTachyon, 2024-12-29:
> Multi-drop SWD is not really meant to be discoverable, the targetsel values for the RP2040 are hardcoded in the info command, so they somewhat work.
> 
> If you select the protocol explicitly using `--protocol swd`, and specify the `TARGETSEL` value using `--target-sel`, do you get better results?

No, I get errors:

```
C:\Storage\GIT\probe-rs>target\debug\probe-rs.exe info --protocol swd --target-sel 0
Probing target via SWD

Error showing ARM chip information for Debug Port Multidrop(0): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

C:\Storage\GIT\probe-rs>target\debug\probe-rs.exe info --protocol swd --target-sel 1
Probing target via SWD

Error showing ARM chip information for Debug Port Multidrop(1): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

C:\Storage\GIT\probe-rs>
```

If I do the GDB dance, then --target-sel X doesn't work, but without it, it works as described previously (only once):

```
C:\Storage\GIT\probe-rs>target\debug\probe-rs.exe gdb --protocol swd --chip RP2040
Firing up GDB stub for Armv6m cores at [[::1]:1337, 127.0.0.1:1337]
^C
C:\Storage\GIT\probe-rs>target\debug\probe-rs.exe info --protocol swd --target-sel 1
Probing target via SWD

Error showing ARM chip information for Debug Port Multidrop(1): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

C:\Storage\GIT\probe-rs>target\debug\probe-rs.exe gdb --protocol swd --chip RP2040
Firing up GDB stub for Armv6m cores at [[::1]:1337, 127.0.0.1:1337]
^C
C:\Storage\GIT\probe-rs>target\debug\probe-rs.exe info --protocol swd --target-sel 1
Probing target via SWD

Error showing ARM chip information for Debug Port Multidrop(1): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

C:\Storage\GIT\probe-rs>target\debug\probe-rs.exe info --protocol swd --target-sel 0
Probing target via SWD

Error showing ARM chip information for Debug Port Multidrop(0): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

C:\Storage\GIT\probe-rs>target\debug\probe-rs.exe info --protocol swd
Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv2, MINDP, Designer: Raspberry Pi Trading Ltd, Part: 0x1002, Revision: 0x0, Instance: 0x00
└── 0 MemoryAP (AmbaAhb3)
    └── ROM Table (Class 1), Designer: ARM Ltd
        ├── Cortex-M0 SCS   (Generic IP component)
        │   └── CPUID
        │       ├── IMPLEMENTER: ARM Ltd
        │       ├── VARIANT: 0
        │       ├── PARTNO: Cortex-M0+
        │       └── REVISION: 1
        ├── Cortex-M0 DWT   (Generic IP component)
        └── Cortex-M0 BPU   (Generic IP component)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

C:\Storage\GIT\probe-rs>target\debug\probe-rs.exe info --protocol swd
Probing target via SWD

ARM Chip with debug port Default:
No access ports found on this chip.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

C:\Storage\GIT\probe-rs>
```

I don't remember the exact sequence of events or even commands, but I seem to remember "info" outputting the 2 cores, once or twice. Should that have never happened?

Tiwalun, 2024-12-29:
0 is wrong for the `targetsel` value, if you want to talk to a RP2040. 0x01002927 or 0x11002927 would be valid values. 

But I wouldn't focus too much on the info command and the RP2040, if you want to support a new chip. Most likely, your chip doesn't need multidrop at all, so far the RP2040 is the only chip to ever need it. The RP2040 is kind of the worst possible chip to use as a comparison, basically all other chips (expect the RP2350) are easier to use.

DualTachyon, 2024-12-29:
> 0x01002927

OK I didn't know that (currently learning as I go!). I tried it and it works with these 2. Thanks for that.

I went to check the IDs for the STM32 board and I'm having no luck. According to ARM Development Studio, these were the IDs it discovered with automatic discovery (using an ST-LINK v3):

```
[28/12/24 22:44:28] Enumerating AP devices for DAP at scanchain index 0:
[28/12/24 22:44:28] APIDR of access port 0 = 0x54770002 (APB-AP)
[28/12/24 22:44:28] APIDR of access port 1 = 0x84770001 (AHB-AP)
[28/12/24 22:44:28] Failed to read AP_IDR register for AP index 4
[28/12/24 22:44:28] APIDR of access port 4 = 0x44770004 (AXI-AP)
[28/12/24 22:44:28] Failed to read AP_IDR register for AP index 16
[28/12/24 22:44:28] Number of AP buses detected: 3
[28/12/24 22:44:28] AP types:
[28/12/24 22:44:28] 		APB-AP
[28/12/24 22:44:28] 		AHB-AP
[28/12/24 22:44:28] 		AXI-AP
```

Are the above 3 APIDR values what I should be using with target-sel?

The output with them is:

```
C:\Apps\Probe>probe-rs.exe info --protocol swd --target-sel 0x54770002
Probing target via SWD

Error showing ARM chip information for Debug Port Multidrop(1417084930): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

C:\Apps\Probe>probe-rs.exe info --protocol swd --target-sel 0x84770001
Probing target via SWD

Error showing ARM chip information for Debug Port Multidrop(2222391297): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

C:\Apps\Probe>probe-rs.exe info --protocol swd --target-sel 0x44770004
Probing target via SWD

Error showing ARM chip information for Debug Port Multidrop(1148649476): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

If I do a more generic "info", I get:

```
C:\Apps\Probe>probe-rs.exe info --protocol swd
Probing target via SWD

ARM Chip with debug port Default:
 WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
 WARN probe_rs::probe::arm_debug_interface: DAP FAULT
 WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
 WARN probe_rs::probe::arm_debug_interface: DAP FAULT
 WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
Error showing ARM chip information for Debug Port Default: An error occurred in the communication with an access port or debug port.

Caused by:
    Target device responded with a FAULT response to the request.

Trying alternate multi-drop debug ports
Error showing ARM chip information for Debug Port Multidrop(16787751): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Error showing ARM chip information for Debug Port Multidrop(285223207): An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

C:\Apps\Probe>
```

If I use a ST-Link v2 on ARM Dev Studio, it complains Multi-drop SWD is not supported, but on the v3 there are no such complaints. Segger's J-Link software also complains this is a Multi-drop SWD device and only does bare minimal with the first AP it finds.

I can confirm that I can access all APs with J-Link on OpenOCD, so this proves the SWD connection is wired up correctly.
```
Info : J-Link V10 compiled Jan 30 2023 11:28:07
Info : Hardware version: 10.10
Info : VTarget = 3.336 V
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x6ba02477
Info : [stm32mp25x.ap0] Examination succeed
Info : stm32mp25x.a35_0: hardware has 6 breakpoints, 4 watchpoints
Info : stm32mp25x.a35_1: hardware has 6 breakpoints, 4 watchpoints
Info : [stm32mp25x.m33] Cortex-M33 r1p0 processor detected
Info : [stm32mp25x.m33] target has 8 breakpoints, 4 watchpoints
Info : [stm32mp25x.m0p] Cortex-M0+ r0p1 processor detected
Info : [stm32mp25x.m0p] target has 4 breakpoints, 2 watchpoints
Info : [stm32mp25x.axi] Examination succeed
```

The .cfg used is from ST direct.

Should I turn this into a new bug focusing on STM32MP2x, or maybe I should rename this one?

I will try writing the yaml for it in the meantime.

bugadani, 2024-12-29:
Since you now don't have an RP2040-related issue, please open a new one.

closed 2024-12-29 by bugadani

Tiwalun, 2024-12-29:
@DualTachyon , I've opened #2997 , and responded there.