probe-rs/probe-rs #2997: Support for STM32MP257

Tiwalun, 2024-12-29:
> 0x01002927
> 
> OK I didn't know that (currently learning as I go!). I tried it and it works with these 2. Thanks for that.
> 
> I went to check the IDs for the STM32 board and I'm having no luck. According to ARM Development Studio, these were the IDs it discovered with automatic discovery (using an ST-LINK v3):
> 
> ```
> [28/12/24 22:44:28] Enumerating AP devices for DAP at scanchain index 0:
> [28/12/24 22:44:28] APIDR of access port 0 = 0x54770002 (APB-AP)
> [28/12/24 22:44:28] APIDR of access port 1 = 0x84770001 (AHB-AP)
> [28/12/24 22:44:28] Failed to read AP_IDR register for AP index 4
> [28/12/24 22:44:28] APIDR of access port 4 = 0x44770004 (AXI-AP)
> [28/12/24 22:44:28] Failed to read AP_IDR register for AP index 16
> [28/12/24 22:44:28] Number of AP buses detected: 3
> [28/12/24 22:44:28] AP types:
> [28/12/24 22:44:28] 		APB-AP
> [28/12/24 22:44:28] 		AHB-AP
> [28/12/24 22:44:28] 		AXI-AP
> ```
> 
> Are the above 3 APIDR values what I should be using with target-sel?
> 
> The output with them is:
> 
> ```
> C:\Apps\Probe>probe-rs.exe info --protocol swd --target-sel 0x54770002
> Probing target via SWD
> 
> Error showing ARM chip information for Debug Port Multidrop(1417084930): An ARM specific error occurred.
> 
> Caused by:
>     0: An error occurred in the communication with an access port or debug port.
>     1: Target device did not respond to request.
> 
> Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
> Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
> 
> 
> C:\Apps\Probe>probe-rs.exe info --protocol swd --target-sel 0x84770001
> Probing target via SWD
> 
> Error showing ARM chip information for Debug Port Multidrop(2222391297): An ARM specific error occurred.
> 
> Caused by:
>     0: An error occurred in the communication with an access port or debug port.
>     1: Target device did not respond to request.
> 
> Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
> Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
> 
> 
> C:\Apps\Probe>probe-rs.exe info --protocol swd --target-sel 0x44770004
> Probing target via SWD
> 
> Error showing ARM chip information for Debug Port Multidrop(1148649476): An ARM specific error occurred.
> 
> Caused by:
>     0: An error occurred in the communication with an access port or debug port.
>     1: Target device did not respond to request.
> 
> Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
> Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
> ```
> 
> If I do a more generic "info", I get:
> 
> ```
> C:\Apps\Probe>probe-rs.exe info --protocol swd
> Probing target via SWD
> 
> ARM Chip with debug port Default:
>  WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
>  WARN probe_rs::probe::arm_debug_interface: DAP FAULT
>  WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
>  WARN probe_rs::probe::arm_debug_interface: DAP FAULT
>  WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
> Error showing ARM chip information for Debug Port Default: An error occurred in the communication with an access port or debug port.
> 
> Caused by:
>     Target device responded with a FAULT response to the request.
> 
> Trying alternate multi-drop debug ports
> Error showing ARM chip information for Debug Port Multidrop(16787751): An ARM specific error occurred.
> 
> Caused by:
>     0: An error occurred in the communication with an access port or debug port.
>     1: Target device did not respond to request.
> 
> Error showing ARM chip information for Debug Port Multidrop(285223207): An ARM specific error occurred.
> 
> Caused by:
>     0: An error occurred in the communication with an access port or debug port.
>     1: Target device did not respond to request.
> 
> Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
> Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
> 
> 
> C:\Apps\Probe>
> ```
> 
> If I use a ST-Link v2 on ARM Dev Studio, it complains Multi-drop SWD is not supported, but on the v3 there are no such complaints. Segger's J-Link software also complains this is a Multi-drop SWD device and only does bare minimal with the first AP it finds.
> 
> I can confirm that I can access all APs with J-Link on OpenOCD, so this proves the SWD connection is wired up correctly.
> ```
> Info : J-Link V10 compiled Jan 30 2023 11:28:07
> Info : Hardware version: 10.10
> Info : VTarget = 3.336 V
> Info : clock speed 5000 kHz
> Info : SWD DPIDR 0x6ba02477
> Info : [stm32mp25x.ap0] Examination succeed
> Info : stm32mp25x.a35_0: hardware has 6 breakpoints, 4 watchpoints
> Info : stm32mp25x.a35_1: hardware has 6 breakpoints, 4 watchpoints
> Info : [stm32mp25x.m33] Cortex-M33 r1p0 processor detected
> Info : [stm32mp25x.m33] target has 8 breakpoints, 4 watchpoints
> Info : [stm32mp25x.m0p] Cortex-M0+ r0p1 processor detected
> Info : [stm32mp25x.m0p] target has 4 breakpoints, 2 watchpoints
> Info : [stm32mp25x.axi] Examination succeed
> ```
> 
> The .cfg used is from ST direct.
> 
> Should I turn this into a new bug focusing on STM32MP2x, or maybe I should rename this one?
> 
> I will try writing the yaml for it in the meantime.
>  

 _Originally posted by @DualTachyon in [#2996](https://github.com/probe-rs/probe-rs/issues/2996#issuecomment-2564764382)_

Tiwalun, 2024-12-29:
@DualTachyion, I'm continuing the discussion here, to make it clearer.

The APIDR values are something different, based on the output it seems that the chip has the standard layout, where you have a single debug port (DP), which gives you access to multiple access ports (APs). In this setup, you usually don't need to specify a targetsel value.

You could try connecting over JTAG, the STM chips usually support both JTAG and SWD.

DualTachyon, 2024-12-29:
Thanks for migrating to this bug!

Unfortunately, the development board only exposes SWD. Link: https://www.myirtech.com/list.asp?id=780

Just running "info" with SWD protocol yields these errors. Are they something worth pursuing?

```
ARM Chip with debug port Default:
 WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
 WARN probe_rs::probe::arm_debug_interface: DAP FAULT
 WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
 WARN probe_rs::probe::arm_debug_interface: DAP FAULT
 WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
Error showing ARM chip information for Debug Port Default: An error occurred in the communication with an access port or debug port.
```

I wrote a new and very basic yaml based on the BCM2712 file and I can confirm that running just "gdb" works. I will look into adding the other cores and more detailed memory map.

```
name: STM32MP2 Series
variants:
- name: STM32MP257
  cores:
  - name: CPU0_0
    type: armv8a
    core_access_options: !Arm
      ap: 0
      debug_base: 0x80210000
      cti_base: 0x80220000
  memory_map:
  - !Ram
    name: SRAM
    range:
      start: 0x20000000
      end: 0x20040000
    cores:
    - CPU0_0
```

[image]

DualTachyon, 2024-12-29:
I added all 4 cores using information from the ST OpenOCD config file, but I'm getting an error related to the Cortex M cores:
```
C:\Storage\GIT\probe-rs>cargo run -- gdb --chip STM32MP257
   Compiling probe-rs v0.25.0 (C:\Storage\GIT\probe-rs\probe-rs)
   Compiling probe-rs-tools v0.25.0 (C:\Storage\GIT\probe-rs\probe-rs-tools)
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 11.85s
     Running `target\debug\probe-rs.exe gdb --chip STM32MP257`
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: The AP with address FullyQualifiedApAddress { dp: Default, ap: V1(8) } does not exist.
error: process didn't exit successfully: `target\debug\probe-rs.exe gdb --chip STM32MP257` (exit code: 1)
```

I don't know if it's a mistake or an omission I made, or whether the SoC has some quirkiness or if I hit a bug with probe-rs.

```
name: STM32MP2 Series
variants:
- name: STM32MP257
  cores:
  - name: CPU1_0
    type: armv8a
    core_access_options: !Arm
      ap: 0
      debug_base: 0x80210000
      cti_base: 0x80220000
  - name: CPU1_1
    type: armv8a
    core_access_options: !Arm
      ap: 0
      debug_base: 0x80310000
      cti_base: 0x80320000
  - name: CPU2
    type: armv8m
    core_access_options: !Arm
      ap: 8
      cti_base: 0xe0042000
  - name: CPU3
    type: armv6m
    core_access_options: !Arm
      ap: 2
      cti_base: 0xf0000000
  memory_map:
  - !Ram
    name: SRAM
    range:
      start: 0x20000000
      end: 0x20040000
    cores:
    - CPU0_0
    - CPU1_0
    - CPU2
    - CPU3
```

I added a "cti_base" to the 2 Cortex M cores as OpenOCD defines them, but it seems probe-rs is not expecting them? I tried without including the CTI bases and it made no difference.

The relevant information from OpenOCD:
```
# define AXI & APB Memory Access Ports
# NOTE: do not change the order of target create
target create $_CHIPNAME.ap0 mem_ap -dap $_CHIPNAME.dap -ap-num 0
target create $_CHIPNAME.axi mem_ap -dap $_CHIPNAME.dap -ap-num 4
target create $_CHIPNAME.ap2 mem_ap -dap $_CHIPNAME.dap -ap-num 2 -defer-examine
target create $_CHIPNAME.ap8 mem_ap -dap $_CHIPNAME.dap -ap-num 8 -defer-examine

# define the first Cortex-A35
cti create $_CHIPNAME.cti.a35_0 -dap $_CHIPNAME.dap -ap-num 0 -baseaddr 0x80220000
target create $_CHIPNAME.a35_0 aarch64 -dap $_CHIPNAME.dap -dbgbase 0x80210000 -ap-num 0 -cti $_CHIPNAME.cti.a35_0 -defer-examine

# define the second Cortex-A35
cti create $_CHIPNAME.cti.a35_1 -dap $_CHIPNAME.dap -ap-num 0 -baseaddr 0x80320000
target create $_CHIPNAME.a35_1 aarch64 -dap $_CHIPNAME.dap -dbgbase 0x80310000 -ap-num 0 -cti $_CHIPNAME.cti.a35_1 -defer-examine

# define the Cortex-M33
target create $_CHIPNAME.m33 cortex_m -dap $_CHIPNAME.dap -ap-num 8 -defer-examine
cti create $_CHIPNAME.cti.m33 -dap $_CHIPNAME.dap -ap-num 8 -baseaddr 0xe0042000

# define the Cortex-M0+
target create $_CHIPNAME.m0p cortex_m -dap $_CHIPNAME.dap -ap-num 2 -defer-examine
cti create $_CHIPNAME.cti.m0p -dap $_CHIPNAME.dap -ap-num 2 -baseaddr 0xf0000000

# define the system CTIs
cti create $_CHIPNAME.cti.sys0 -dap $_CHIPNAME.dap -ap-num 0 -baseaddr 0x80090000
cti create $_CHIPNAME.cti.sys1 -dap $_CHIPNAME.dap -ap-num 0 -baseaddr 0x800a0000
```

I suspect this portion is needed to make the Cortex Ms appear:
```
# Uses AP0 and AXI
proc _enable_debug {} {
	# set debug enable bits in DBGMCU_CR to get ap2/cm0+ and ap8/cm33 visible
	# set DBG_SWD_SEL_N bit in DBGMCU_CR to get ap2/cm0+ on main debug interface
	$::_CHIPNAME.ap0 mww 0x80010004 0x17
```

Is there a way to make this write from the yaml, or is a change in the Rust code necessary?

Tiwalun, 2024-12-29:
There are "sequences", which are used for this. See e.g probe-rs/src/vendor/st/sequences/stm32h7.rs