probe-rs/probe-rs #3009: Support `target_reset_assert` for FTDI probes

jr1221, 2025-01-06:
### Is your feature request related to a problem? Please describe.

I have to reset my target every time I flash, probably related to the STM32 WFI stuff which is often discussed here.  I have a NRST line running to a FTDI probe.  However, to reset the target with `--connect-under-reset` `target_reset_assert` must be supported, which it is not for FTDI probes.

### Describe the solution you'd like

I would like the `--connect-under-reset` to work.

### Describe alternatives you've considered

I have considered changing STM32 WFI bits to prevent the sleep that causes the reset from occuring, but I ran into issues out of the scope of this ticket.

### Additional context

I am knowledgable with rust but not necessarily the FTDI spec, I was wondering if this could be a situation for a new member to contribute? Is this ticket easily solvable, or does it require some big changes to `probe-rs`?  Any context would be much appreciated.

jr1221, 2025-01-06:
According to this the pin is configurable, so we must pass the pin to be reset to the ftdi
https://github.com/probe-rs/probe-rs/blob/2bb002b68e6ba4e3d2a15837c0884639d3a196cf/probe-rs/src/probe/ftdi/mod.rs#L364.

However Im starting to realize how many settings I was taking advantage of to get the result I wanted in OpenOCD.  Is there a RFC or roadmap which has/will discuss the limitations of statically configured FTDI?  I feel like introducing a customizable property either at compile time via macros or at runtime would reduce a whole class of these issues.

bugadani, 2025-01-06:
There are no plans currently to make probes configurable at all, mainly because nobody needs it badly enough to do it. We most likely shouldn't pollute the CLI parameters with `--ftdi-xy-pin` flags or such, instead we should figure out some standard-ish way. This would unblock SWD over FTDI, too.

I think the best would be to push https://github.com/probe-rs/probe-rs/pull/2994 and add probe configuration to it. In a config file we should be able to specify configuration for a probe type, or a particular serial number. We should be able to provide different configuration structs for each probe type. I don't know how the current probe `speed` fits into this picture, however.

Yatekii, 2025-02-22:
I don't get what keeps us from just using a static in-code map like we do for some vid:pid and probe version pairs already. I think this is something that can be dealt with in code without the need for any additional config.

bugadani, 2025-02-22:
Now that we have a config file, maybe we can add probe config to it, with pin maps.

Yatekii, 2025-02-22:
Ah, I think I misunderstood that the pid:vid pair is defined by the FTDI used and not by the probe. I guess some probe specific config bits are in order indeed.