probe-rs/probe-rs #3037: "ESP32-S3 with CH343 Converter Not Recognized by probe-rs Despite Being Visible in /dev/ttyACM0"

nan-mu, 2025-01-19:
### Describe the bug

I am using an ESP32-S3 development board with a CH343 converter connected to my Linux system. The device is displayed as /dev/ttyACM0 under the /dev directory. The issue is that when I run probe-rs list, the device is not recognized.

[image]

I believe the issue is not with the USB cable, as the same cable works fine with CMSIS-DAP. Furthermore, the development board itself is not the problem, as I can successfully perform flashing and RTT communication using ESP-IDF's Python scripts.

I suspect the issue lies with the device driver, as the device does not appear under /sys/bus/usb/devices, and /dev/ttyACM0 is shown as a character device rather than a symlink. However, I can see the device listed under /sys/class/tty, with the path: 

```
/sys/devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1:1.0/tty/ttyACM0.
```

My device is a Raspberry Pi 5, which has two USB2 and two USB3 ports. I tried two different ports, but probe-rs still doesn't work properly.

### To Reproduce

If you can see /dev/ttyACM0 but cannot find the device under /sys/bus/usb/devices, you will likely encounter the same issue that I am experiencing.

### Expected behavior

I expect probe-rs list to correctly detect the device and display the corresponding information, and for the device to be listed under /sys/bus/usb/devices. Additionally, /dev/ttyACM0 should appear as a symlink rather than a character device. The USB device should also function properly through the Raspberry Pi's ports, especially when used with probe-rs. My expectation is that the behavior aligns with the output seen in ESP-IDF, as shown below:

Here is the output from running the ESP-IDF commands for flashing and viewing RTT:

```bash
$ idf.py flash
Executing action: flash
Serial port /dev/ttyAMA10
Connecting.......................
/dev/ttyAMA10 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... ESP32-S3
Running ninja in directory /home/nan/esp/esp-idf/examples/get-started/hello_world/build
Executing "ninja flash"...
[1/5] cd /home/nan/esp/esp-idf/examples/get-started/hello_world/build/e.../nan/esp/esp-idf/examples/get-started/hello_world/build/hello_world.bin
hello_world.bin binary size 0x327f0 bytes. Smallest app partition is 0x100000 bytes. 0xcd810 bytes (80%) free.
[1/1] cd /home/nan/esp/esp-idf/examples/get-started/hello_world/build/b...sp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x5210 bytes. 0x2df0 bytes (36%) free.
[4/5] cd /home/nan/esp/esp-idf/components/esptool_py && /usr/bin/cmake ...ld -P /home/nan/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32s3 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_1v8)
Crystal is 40MHz
MAC: 68:b6:b3:3d:93:7c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00042fff...
Flash will be erased from 0x00008000 to 0x00008fff...
SHA digest in image updated
Compressed 21008 bytes to 13370...
Writing at 0x00000000... (100 %)
Wrote 21008 bytes (13370 compressed) at 0x00000000 in 0.7 seconds (effective 227.2 kbit/s)...
Hash of data verified.
Compressed 206832 bytes to 110416...
Writing at 0x0003da3b... (100 %)
Wrote 206832 bytes (110416 compressed) at 0x00010000 in 2.6 seconds (effective 630.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
[... 85 lines cut ...]
Restarting in 0 seconds...
Restarting now.
```

This output shows that the ESP32-S3 development board is correctly detected and communicates with ESP-IDF for both flashing and RTT monitoring.

### Stacktrace

```shell
I used `lsusb -v` and `udevadm info -a -p $(udevadm info -q path -n /dev/ttyACM0)` to check the device, but `probe-rs` was unable to correctly detect the device. The detailed outputs from these tools are provided in the attachment for further analysis.
```

### Operating System

Linux (Debian)

### Additional context

Maybe help: [Taobao link](https://item.taobao.com/item.htm?abbucket=14&id=737615148866&ns=1&pisk=gOzq3-G-GZQqkjVUFy0w8tNLbKgxpVWCuPMss5ViGxDmh5ci7821MxNGHRzZE5EfMc6v_IUzLFTfHiFM7VgGd9_CRSFjWVXIDc-2aKhIiAbsnFVokfiAF_95RSFxWIJGOubBQPUgxKxiIR0oqXhiShmmS_ooOf0MocYDqLD-EA0iifYoEfGHiEYgILDotXGDmmDMZQcEOVDgIVfzZYhZBUzguUkqm10pBEtPQEijKSDySeRKTmXTahTH-zlm0PDIUbG7zjoqL0EyEQU0F54bk0IJVVF7YRrgKTtjnklUUDZVLEyUHf2rlyfOD0ng7Pl82O-ZLlVbvlmPiZVruxogP8AB2qrgHPoY0CCLZq28vvnc2ThzlyiZp0Jln7NrnDrahTLihkPuUDaWFeg3v8zZx2jPwdhuV7aT0dxZmbhrd_5zEEbMmdPUiz-9X0f-av1FThKtmbhrd_5yXhnlwbkCT65..&priceTId=2150426217373110480297438e64e5&spm=a21n57.1.hoverItem.2&utparam=%7B%22aplus_abtest%22%3A%225031325e6f3a972bacb72308be1d0711%22%7D&xxc=taobaoSearch)

[lsusb.log](https://github.com/user-attachments/files/18469955/lsusb.log)
[probe-rs.log](https://github.com/user-attachments/files/18469953/probe-rs.log)
[udevadm.log](https://github.com/user-attachments/files/18469954/udevadm.log)

[image]

bugadani, 2025-01-19:
probe-rs does not support UART to flash/debug. And idf.py does not support RTT.

nan-mu, 2025-01-19:
> probe-rs does not support UART to flash/debug. And idf.py does not support RTT.

Alright, then this might turn into a feature request? And isn't the output of `idf.py monitor` already RTT?

Vollbrecht, 2025-01-19:
if you want just flashing capabilities than `espflash` is the rust tool to do it. i don't speak for probe-rs, but consider that its primarily a tool to get the jtag/svd debug connections up and running, its not really its primary goal.

nan-mu, 2025-01-19:
> if you want just flashing/monitor capabilities than `espflash` is the rust tool to do it. i don't speak for probe-rs, but consider that its primarily a tool to get the jtag/svd debug connections up and running, its not really its primary goal.

Thank you, I came from the template project in the ESP Rust book. They’ve updated a lot of APIs, and in the latest `cargo run` alias, they use `probe-rs`. Ok, thanks.

But to be honest, I think these two tools seem quite redundant? I’m surprised that `probe-rs`, as a Rust embedded version of OpenOCD, doesn’t support this functionality.

nan-mu, 2025-01-19:
Alright, it seems that [https://github.com/esp-rs/esp-generate/blob/main/template/.cargo/config.toml](https://github.com/esp-rs/esp-generate/blob/main/template/.cargo/config.toml) indicates I chose the wrong configuration when creating the project.

closed 2025-01-19 by nan-mu