probe-rs/probe-rs #3045: Missing / incomplete defmt output with v0.26.0 and v0.25.0

jschneider-bensch, 2025-01-23:
### Describe the bug

For the example reproducer setup described below, I am getting no defmt output using probe-rs v0.26.0 and incomplete output for v0.25.0:

 v0.26.0:
```
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.02s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/hello`
      Erasing ✔ 100% [####################]   8.00 KiB @  19.32 KiB/s (took 0s)
  Programming ✔ 100% [####################]   8.00 KiB @  16.46 KiB/s (took 0s)                                                                   Finished in 0.90s
```

v0.25.0:
```
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.02s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/hello`
      Erasing ✔ 100% [####################]   8.00 KiB @  19.31 KiB/s (took 0s)
  Programming ✔ 100% [####################]   8.00 KiB @   8.89 KiB/s (took 1s)                                                                   Finished in 0.90s
Hello, world!
```

### To Reproduce

1. Set up an example project, e.g. using [this template](https://github.com/knurling-rs/app-template). I have an nRF52840, so I can follow the instructions exactly.
2. Install probe-rs v0.26.0 or v0.25.0
3. Run one of the examples with output, e.g. `cargo rb hello`.

### Expected behavior

Using probe-rs v.0.24.0, I get the following behaviour:
``` 
Finished `dev` profile [optimized + debuginfo] target(s) in 0.02s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/hello`
      Erasing ✔ [00:00:00] [######################################################################] 8.00 KiB/8.00 KiB @ 25.71 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [######################################################################] 8.00 KiB/8.00 KiB @ 16.30 KiB/s (eta 0s )    Finished in 0.843s
<lvl> Hello, world!
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:8```

### Stacktrace

```shell

```

### Operating System

Other Linux Distribution

### Additional context

Observed on nRF52840, but also on STM32L4R5ZI, so maybe a Cortex-M issue?

kaidokert, 2025-02-02:
I just spent couple hours trying to figure out why i got no output on `defmt` on a `ATSAMD21` project that worked fine on another development host where i hadn't updated things.
 
Turned out downgrade from `0.26` to `cargo install probe-rs-tools@0.25.0` "fixed" the problem.

Using `rtt_target` for output works fine in both cases, it seems specific to `demt`. 

Versions used:
```
├── defmt-rtt v0.4.1
│   ├── critical-section v1.2.0
│   └── defmt v0.3.10 (*)
```

sebastianbergt, 2025-02-05:
Based on what you recommended, I just downgraded probe-rs with
`curl --proto '=https' --tlsv1.2 -LsSf https://github.com/probe-rs/probe-rs/releases/download/v0.25.0/probe-rs-tools-installer.sh | sh`

Yatekii, 2025-02-22:
Could this be because of an outdated defmt-rtt?

ia0, 2025-02-28:
Could this be related to https://github.com/knurling-rs/defmt/issues/825#issuecomment-2624109853 where probe-rs seems to not handle high RTT throughput?

ia0, 2025-02-28:
Also the issue is still present with 0.27.0.

bugadani, 2025-02-28:
Any chance you could provide us a reproducer? I have a number of devices to try, although none of them is an nrf52840.

ia0, 2025-02-28:
https://github.com/knurling-rs/defmt/issues/825#issuecomment-2124663112 is probably the most simple demonstration of the issue. You can easily modify `app-template` like I did in this branch: https://github.com/ia0/app-template/tree/bug

I get a similar output as in the comment. I get numbers starting from 622 and I'm missing 768 and 914. But I expect those numbers to be non-deterministic.

bugadani, 2025-03-16:
The only data loss I see happens during boot, before we configure the RTT channel to be BlockIfFull. We can only reconfigure the channels after the control block has been initialized, and we are not really notified about that, so what happens to data printed between the init event and our first poll depends on what the target firmware sets as the default channel mode. On rtt-target that is `NoBlockSkip`, on defmt-rtt it is `NoBlockTrim`. I don't see anything we can do about this, without watchpoints that we don't support currently (to detect when the channel mode is written, halt, and force it to our mode).

Maybe we could make the target firmware debugger-aware and initialize channels in blocking mode if probe-rs is connected. However, I'm not sure if we want to make rtt-target platform dependent in such a way, and this could also be achieved in user code I believe.

After probe-rs has set `BlockIfFull` the target loops until probe-rs has freed up enough space in the RTT buffer to continue, so data really, really should not get lost. I can reproduce the first issue just fine, but not any other data loss.

ia0, 2025-03-16:
> The only data loss I see happens during boot

Weird... I've added a `bug.sh` script to the branch above. Here's my output running `./bug.sh`. This really looks to me that I get the `NoBlockTrim` behavior up to some number (here 171) and then I get occasional drops.

```
Starting at 171
Expected 307 got 308
Expected 595 got 596
Expected 872 got 873
Finished with 999
```

I've also pushed a https://github.com/ia0/app-template/tree/issue/probe-rs/3045-2 branch using `rtt-target` with `BlockIfFull` and I get this:

```
Starting at 0
Expected 170 got 171
Expected 460 got 461
Expected 746 got 747
Expected 893 got 894
Finished with 999
```

This really suggests that we get drop regardless of `BlockIfFull`.

What output do you get on both those branches?

bugadani, 2025-03-16:
I had to make a following changes:
- My device is an nrf52833 (a microbit v2)
- I'm on windows, so I'm printing all lines, and basically checking in an editor if the lines and line numbers line up (in a line).

With current git master:
- Using the default channel mode, numbers 170-919 are missing, the rest are there.
- With `BlockIfFull` everything is printed

ia0, 2025-03-16:
Ok so this may help corner the issue, it could be specific to any of:
- nrf52840 vs nrf52833
- linux vs windows

I don't have any other board than nrf52840 so can't test more platforms. I have windows though for games, but I have no clue how to use it for development. Would be good if others could test their setups (on both branches).

ValouBambou, 2026-03-11:
I cannot reproduce it with 0.31 version it might have been fully fixed !