probe-rs/probe-rs #3053: nrf5340 core1 locked

papadkostas, 2025-01-27:
### Describe the bug

I've been trying to connect to a board with NRF5340 but seems always the core1 is locked.
While looks like it's attempting to erase, it fails to connect.
```
WARN probe_rs::vendor::nordicsemi::sequences::nrf: Core 1 is locked. Erase procedure will be started to unlock it.
Error: Connecting to the chip was unsuccessful.
```
I've tried to recover the mcu with other tools and try to program again, still did not help.
I've been using a pico debug probe and the JLink OB probe, none of these worked.
While they are both able to detect the MCU, going any further was not possible.
```
❯ probe-rs info
Available Probes:
0: Debug Probe (CMSIS-DAP) -- 2e8a:000c:E6614103E70D5A2F (CMSIS-DAP)
1: J-Link -- 1366:0105:000760208498 (J-Link)
Selection: 0
Probing target via JTAG

Error identifying target using protocol JTAG: The probe does not support the JTAG protocol.

Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv2, Designer: Nordic VLSI ASA, Part: 0x70, Revision: 0x5, Instance: 0x00
├── 0 MemoryAP (AmbaAhb3)
│   └── ROM Table (Class 1), Designer: Nordic VLSI ASA
│       ├── ROM Table (Class 1), Designer: ARM Ltd
│       │   ├── Cortex-M33 SCS  (Coresight Component)
│       │   │   └── CPUID
│       │   │       ├── IMPLEMENTER: ARM Ltd
│       │   │       ├── VARIANT: 0
│       │   │       ├── PARTNO: Cortex-M33
│       │   │       └── REVISION: 4
│       │   ├── Cortex-M33 DWT  (Coresight Component)
│       │   ├── Cortex-M33 BPU  (Coresight Component)
│       │   ├── Cortex-M33 ITM  (Coresight Component)
│       │   ├── Cortex-M33 ETM  (Coresight Component)
│       │   └── Cortex-M33 CTI  (Coresight Component)
│       └── Cortex-M33 TPIU (Coresight Component)
├── 1 MemoryAP (AmbaAhb3)
│   └── Memory AP is not accessible, DeviceEn bit not set
├── 2 Unknown AP (Designer: Nordic VLSI ASA, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x1)
└── 3 Unknown AP (Designer: Nordic VLSI ASA, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x1)
```
I've tried 0.25, 0.24, 0.23 with no success.

The debug probes work with other tools without any issues.

### To Reproduce

either
probe-rs download ./build/merged.hex --verify --chip nRF5340_xxAA
or
probe-rs erase --chip nRF5340_xxAA

### Stacktrace

```shell

```

### Operating System

Linux (Debian)

Yatekii, 2025-02-22:
Could you attach a trace (`RUST_LOG=DEBUG`)? :)

papadkostas, 2025-02-24:
Sure, but does this require building from source? if yes, is there a place I can get it without building localy?

Yatekii, 2025-02-25:
Nope, you don't have to rebuild from source :)

papadkostas, 2025-03-12:
> Nope, you don't have to rebuild from source :)

could you describe me how-to? 🤔

0e4ef622, 2025-05-15:
I've found that probe-rs works perfectly fine as long as the net core is unlocked and running.

papadkostas, 2025-05-16:
> I've found that probe-rs works perfectly fine as long as the net core is unlocked and running.

Correct, in my case net core is not utilized at the moment. Good catch!