probe-rs/probe-rs #3058: RTT drops defmt frames

jonathanpallant, 2025-01-30:
### Describe the bug

probe-rs has been observed to drop defmt frames. The same firmware using Ozone and defmt-print does not drop frames. The frame drop rate appears to be correlated with RTT buffer size, so I suspect it's a buffer wrapping issue.

I note that https://github.com/probe-rs/probe-rs/pull/3049 says

>  I'm also fixing a bug where the defmt processor couldn't deal with data arriving split into multiple reads

which sounds like it might be the underlying issue

### To Reproduce

1. Write a program that waits for RTT to go into blocking mode and then defmt prints a lot of stuff
2. Some stuff will be missing
3. Run it again using Ozone and defmt-print and nothing goes missing
 

### Expected behavior

Frames to not go missing when the RTT buffer is in blocking mode

### Stacktrace

```shell

```

### Operating System

None

### Additional context

We discussed this on probe.rs matrix chat.

jannic, 2025-01-30:
See https://github.com/probe-rs/probe-rs/blob/849498d6ec2f7196baa8c3218fe9c1ffb9b039ba/probe-rs-tools/src/bin/probe-rs/util/rtt.rs#L211-L214

jonathanpallant, 2025-01-30:
uh yeah, that seems bad. A defmt frame is definitely wrapped across the end of the buffer. Even the small components within a frame can be wrapped.

jonathanpallant, 2025-01-30:
Yes except #3049 is broken for me because it needs #3056

jannic, 2025-01-30:
I think wrapping across the end of the buffer is not an issue - that is handled by ./probe-rs/src/rtt/channel.rs IIRC. However, the implementation can at most return one full RTT buffer. It may contain some incomplete defmt frame at the end.

Yatekii, 2025-02-22:
Both of the aforementioned PRs landed. Please reopen this if the issue is still present.

closed 2025-02-22 by Yatekii