probe-rs/probe-rs #3066: Error with semihosting `exit()` on rp235x

kaspar030, 2025-02-02:
### Describe the bug

I'm seeing this on an rpi-pico-2:

```
    Finished `release` profile [optimized + debuginfo] target(s) in 0.40s
     Running `probe-rs run --protocol=swd --chip RP235x --preverify ../../build/bin/rpi-pico2/cargo/thumbv8m.main-none-eabi/release/hello-world`
      Erasing ✔ 100% [####################]  16.00 KiB @  59.94 KiB/s (took 0s)
  Programming ✔ 100% [####################]  16.00 KiB @  13.63 KiB/s (took 1s)                                                                                
     Finished in 2.13s
INFO  Hello World!
Frame 0: syscall_readonly @ 0x10002bda inline
       /home/kaspar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.18/src/sys/arm_compat/syscall/arm.rs:88:9
Frame 1: sys_exit_extended @ 0x0000000010002bd8 inline
       /home/kaspar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.18/src/sys/arm_compat/mod.rs:195:9
Frame 2: exit @ 0x0000000010002bcc inline
       /home/kaspar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.18/src/sys/arm_compat/mod.rs:164:5
Frame 3: exit @ 0x0000000010002bcc
       /home/kaspar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.18/src/process.rs:57:5
Frame 4: init @ 0x10000240
       /home/kaspar/src/own/rust/ariel-os/src/ariel-os-embassy/src/lib.rs:121:1
Error: CPU halted unexpectedly.
```

This is on `probe-rs 0.26.0 (git commit: v0.1.0-345-gb2562d9b)`.

### Expected behavior

This should catch the exit properly like this: (output from **rp2040**)

```
   Finished `release` profile [optimized + debuginfo] target(s) in 7.39s
     Running `probe-rs run --protocol=swd --chip RP2040 --preverify ../../build/bin/rpi-pico/cargo/thumbv6m-none-eabi/release/hello-world`
 WARN probe_rs::architecture::arm::core::armv6m: The core is in locked up status as a result of an unrecoverable exception
      Erasing ✔ 100% [####################]  20.00 KiB @  79.61 KiB/s (took 0s)
  Programming ✔ 100% [####################]  20.00 KiB @  45.69 KiB/s (took 0s)                                                                                
     Finished in 0.85s
INFO  Hello World!
 WARN probe_rs::architecture::arm::communication_interface: Failed to switch to DP Multidrop(11002927): An error occurred in the communication with an access port or debug port.

### Stacktrace

```shell

```

### Operating System

Other Linux Distribution

jannic, 2025-02-03:
Tracing this issue showed a message `Semihosting check pc=0x10000aca instruction=0x180x18` where `instruction=0xbe0xab` was expected.

Disassembly of this code location showed that the expected `beab` actually is at this location:
```
10000ac8:       2018            movs    r0, #24
10000aca:       beab            bkpt    0x00ab
```
So something went wrong reading this value. Which lead me to #3070: The 0x18 seen in the log message is being read because the 8-bit read is actually doing a 32-bit read and one needs to select the correct bit lane manually.

kaspar030, 2025-02-03:
I can confirm that #3070 fixes this issue! :+1:

PR #3070 (2025-02-03): Fix read_8, read_16 for ap_v2
Fixes #3066
closed 2025-02-03 by bugadani