probe-rs/probe-rs #3075: Problem with access port on STM32F205

jwagen, 2025-02-06:
### Describe the bug

Hi

I am unable to connect to STM32F205 with a ST-link V2. I get this error.
When running probe-rs info i get this.

```
Debug Port: DPv0, DP Designer: <unknown>
└── 0 MemoryAP (AmbaAhb3)
    └── Error during access: The selected probe does not support the 'ARM' interface.
```

I have bisected it to [this commit](https://github.com/probe-rs/probe-rs/commit/0fa885a47e32966a87479b93f6c51edd3cfdc7fa#diff-a489871a99b42c40097b5750fae3794f6be7d7de8e740254cfb1c969ae997412)

The probe works fine with STMCube tools.

This is the output from v0.25.0 which seems to be working.
```
Probing target via JTAG

ARM Chip with debug port Default:
Debug Port: DPv0, DP Designer: <unknown>
└── 0 MemoryAP (AmbaAhb3)
    └── ROM Table (Class 1), Designer: STMicroelectronics
        ├── Cortex-M3 SCS   (Generic IP component)
        │   └── CPUID
        │       ├── IMPLEMENTER: ARM Ltd
        │       ├── VARIANT: 2
        │       ├── PARTNO: Cortex-M3
        │       └── REVISION: 0
        ├── Cortex-M3 DWT   (Generic IP component)
        ├── Cortex-M3 FBP   (Generic IP component)
        ├── Cortex-M3 ITM   (Generic IP component)
        ├── Cortex-M3 TPIU  (Coresight Component)
        └── Cortex-M3 ETM   (Coresight Component)
```

<details>
<summary>
Broken trace log
</summary>

``` 

DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: interface 0 has no string, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: interface 0 has no string, skipping
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 0 CMSIS-DAP probes using nusb, searching HID
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 0 CMSIS-DAP probes total
DEBUG list_wlink_devices: probe_rs::probe::wlink: Searching for WCH-Link(RV) probes
DEBUG list_wlink_devices: probe_rs::probe::wlink: Found 0 WCH-Link probes total
TRACE probe_rs::probe::cmsisdap::tools: Attempting to open device matching 0483:3748:55C3BF6A0666C285485531511767
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 1, device_address: 1, vendor_id: 0x1D6B, product_id: 0x0002, device_version: 0x0515, class: 0x09, subclass: 0x00, protocol: 0x01, speed: Some(High), manufacturer_string: Some("Linux 5.15.153.1-microsoft-standard-WSL2 vhci_hcd"), product_string: Some("USB/IP Virtual Host Controller"), serial_number: Some("vhci_hcd.0"), sysfs_path: SysfsPath("/sys/bus/usb/devices/usb1"), interfaces: [InterfaceInfo { interface_number: 0, class: 0x09, subclass: 0x00, protocol: 0x00, interface_string: None }] }
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 1, device_address: 4, vendor_id: 0x0483, product_id: 0x3748, device_version: 0x0100, class: 0x00, subclass: 0x00, protocol: 0x00, speed: Some(Full), manufacturer_string: Some("STMicroelectronics"), product_string: Some("STM32 STLink"), serial_number: Some("Uÿj\u{6}f\u{85}HU1Q\u{17}g"), sysfs_path: SysfsPath("/sys/bus/usb/devices/1-1"), interfaces: [InterfaceInfo { interface_number: 0, class: 0xFF, subclass: 0xFF, protocol: 0xFF, interface_string: Some("ST-Link Debug") }] }
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 2, device_address: 1, vendor_id: 0x1D6B, product_id: 0x0003, device_version: 0x0515, class: 0x09, subclass: 0x00, protocol: 0x03, speed: Some(Super), manufacturer_string: Some("Linux 5.15.153.1-microsoft-standard-WSL2 vhci_hcd"), product_string: Some("USB/IP Virtual Host Controller"), serial_number: Some("vhci_hcd.0"), sysfs_path: SysfsPath("/sys/bus/usb/devices/usb2"), interfaces: [InterfaceInfo { interface_number: 0, class: 0x09, subclass: 0x00, protocol: 0x00, interface_string: None }] }
DEBUG probe_rs::probe::cmsisdap::tools: Attempting to open 0483:3748 in CMSIS-DAP v1 mode
DEBUG probe_rs::probe::stlink::usb_interface: Aquired handle for probe
DEBUG probe_rs::probe::stlink::usb_interface: Claimed interface 0 of USB device.
DEBUG probe_rs::probe::stlink::usb_interface: Succesfully attached to STLink.
DEBUG probe_rs::probe::stlink: Initializing STLink...
TRACE probe_rs::probe::stlink: Getting current mode of device...
TRACE probe_rs::probe::stlink::usb_interface: Sending command [f5] to STLink, timeout: 1s
TRACE probe_rs::probe::stlink::usb_interface: Read 2 bytes, 0 bytes remaining
DEBUG probe_rs::probe::stlink: Current device mode: MassStorage
TRACE probe_rs::probe::stlink::usb_interface: Sending command [f1] to STLink, timeout: 1s
TRACE probe_rs::probe::stlink::usb_interface: Read 6 bytes, 0 bytes remaining
DEBUG probe_rs::probe::stlink: STLink version: (2, 45)
Probing target via JTAG

[... 121 lines cut ...]
Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
DEBUG probe_rs::probe::stlink: Detaching from STLink.
TRACE probe_rs::probe::stlink: Getting current mode of device...
TRACE probe_rs::probe::stlink::usb_interface: Sending command [f5] to STLink, timeout: 1s
TRACE probe_rs::probe::stlink::usb_interface: Read 2 bytes, 0 bytes remaining
DEBUG probe_rs::probe::stlink: Current device mode: Jtag
TRACE probe_rs::probe::stlink::usb_interface: Sending command [f2, 21] to STLink, timeout: 1s

TRACE probe_rs::probe::stlink: Getting current mode of device...
TRACE probe_rs::probe::stlink::usb_interface: Sending command [f5] to STLink, timeout: 1s
TRACE probe_rs::probe::stlink::usb_interface: Read 2 bytes, 0 bytes remaining
DEBUG probe_rs::probe::stlink: Current device mode: MassStorage
```
</details>

### Stacktrace

```shell

```

### Operating System

Linux (Ubuntu)

JanBerktold, 2025-02-06:
I ran into the same issue with the https://www.st.com/en/evaluation-tools/nucleo-h533re.html:

```
PS C:\Users\jberktold> probe-rs info --chip STM32H533RE --protocol swd
Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4780, Revision: 0x1, Instance: 0x00
├── 0 MemoryAP (AmbaApb2Apb3)
│   └── Error during access: The selected probe does not support the 'ARM' interface.
└── 1 MemoryAP (AmbaAhb3)
    └── Error during access: The selected probe does not support the 'ARM' interface.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

With v0.25:

```
PS C:\Users\jberktold> probe-rs info --protocol swd
Probing target via SWD

ARM Chip with debug port Default:
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApWdataError
Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4780, Revision: 0x1, Instance: 0x00
├── 0 MemoryAP (AmbaApb2Apb3)
│   └── Error during access: The debug probe encountered an error.
└── 1 MemoryAP (AmbaAhb3)
    └── ROM Table (Class 1), Designer: STMicroelectronics
        ├── ROM Table (Class 1), Designer: ARM Ltd
        │   ├── Cortex-M33 SCS  (Coresight Component)
        │   │   └── CPUID
        │   │       ├── IMPLEMENTER: ARM Ltd
        │   │       ├── VARIANT: 0
        │   │       ├── PARTNO: Cortex-M33
        │   │       └── REVISION: 4
        │   ├── Cortex-M33 DWT  (Coresight Component)
        │   ├── Cortex-M33 BPU  (Coresight Component)
        │   ├── Cortex-M33 ITM  (Coresight Component)
        │   ├── Cortex-M33 ETM  (Coresight Component)
        │   └── Cortex-M33 CTI  (Coresight Component)
        └── Cortex-M33 TPIU (Coresight Component)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

bugadani, 2025-02-06:
Do you have the same problem when you are trying to do something actually useful, or just with `info`?

jwagen, 2025-02-07:
It looks like `run` is working on v0.26.0. 

The pr #2972 I bisected it to says it added an additional check for the `info` command

```
> probe-rs.exe --version
probe-rs 0.26.0 (git commit: e92fd37)
```

```
> probe-rs.exe run Firmware.elf --chip STM32F205VGTx
    Erasing ✔ 100% [####################] 384.00 KiB @  64.26 KiB/s (took 6s)
Programming ✔ 100% [####################] 289.00 KiB @  31.66 KiB/s (took 9s)
     Finished in 15.12s
```

```
> probe-rs.exe info
Probing target via JTAG

ARM Chip with debug port Default:
Debug Port: DPv0, DP Designer: <unknown>
└── 0 MemoryAP (AmbaAhb3)
    └── Error during access: The selected probe does not support the 'ARM' interface.

Unable to debug RISC-V targets using the current probe. RISC-V specific information cannot be printed.
Unable to debug Xtensa targets using the current probe. Xtensa specific information cannot be printed.

Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv1, DP Designer: ARM Ltd
└── 0 MemoryAP (AmbaAhb3)
    └── Error during access: The selected probe does not support the 'ARM' interface.

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```