probe-rs/probe-rs #3088: `STM32H747XI` encountered DAP faults and cannot access cores via assigned APs

ryan-summers, 2025-02-10:
### Describe the bug

It appears that when the AP access IDs for the cores were updated in #3036 or #2643, access to the STM32H747XI chip broke. Flashing and info dumps indicate a large number of errors.

```
probe-rs --version
probe-rs 0.26.0 (git commit: 4537144b)

```

```
probe-rs info --chip STM32H747XI
Probing target via JTAG

ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

 WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
 WARN probe_rs::probe::arm_debug_interface: DAP FAULT
 WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
ARM Chip with debug port Default:
Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4500, Revision: 0x0, Instance: 0x00
├── 0 MemoryAP
│   └── ROM Table (Class 1), Designer: STMicroelectronics
│       ├── ROM Table (Class 1), Designer: ARM Ltd
│       │   ├── Cortex-M4 SCS   (Generic IP component)
│       │   │   └── CPUID
│       │   │       ├── IMPLEMENTER: ARM Ltd
│       │   │       ├── VARIANT: 1
│       │   │       ├── PARTNO: Cortex-M7
│       │   │       └── REVISION: 1
│       │   ├── Cortex-M3 DWT   (Generic IP component)
│       │   ├── Cortex-M7 FBP   (Generic IP component)
│       │   └── Cortex-M3 ITM   (Generic IP component)
│       ├── Cortex-M7 ETM   (Coresight Component)
│       └── Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
├── 1 MemoryAP
│   └── Error during access: Error using access port.
├── 2 MemoryAP
│   └── Error during access: Error using access port.
└── 3 Unknown AP (Designer: <unknown>, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x2)

 WARN probe_rs::probe::arm_debug_interface: DAP FAULT
 WARN probe_rs::probe::arm_debug_interface: Error reading CTRL/STAT register. This should not happen...
Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

### Stacktrace

```shell

```

### Operating System

None

badrbouslikhin, 2025-02-10:
Sorry, I got mixed up. That log is from v0.24.0.
Here is the one from `probe-rs 0.26.0 (git commit: 4537144b)`
````
Probing target via JTAG

ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

ARM Chip with debug port Default:
ERROR probe_rs::architecture::arm::memory::romtable:    Failed to read component information at 0x0.
Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4500, Revision: 0x0, Instance: 0x00
├── 0 MemoryAP (AmbaAhb3)
│   ├── 0xe0000000 Cortex-M3 ITM   (Generic IP component)
│   ├── 0xe0001000 Cortex-M3 DWT   (Generic IP component)
│   ├── 0xe0002000 Cortex-M7 FBP   (Generic IP component)
│   ├── 0xe000e000 Cortex-M4 SCS   (Generic IP component)
│   │   └── CPUID
│   │       ├── IMPLEMENTER: ARM Ltd
│   │       ├── VARIANT: 1
│   │       ├── PARTNO: Cortex-M7
│   │       └── REVISION: 1
│   ├── 0xe0041000 ETM architecture (Coresight Component)
│   ├── 0xe0043000 Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
│   ├── 0xe00fe000 ROM Table (Class 1), Designer: STMicroelectronics
│   └── 0xe00ff000 ROM Table (Class 1), Designer: ARM Ltd
├── 1 MemoryAP (AmbaAhb3)
│   └── 0x0000 Generic IP component
├── Memory AP is not accessible, DeviceEn bit not set
└── 3 Unknown AP (Designer: <unknown>, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x2)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

ryan-summers, 2025-02-10:
The current theory right now is that the 747XI should be using AP0 for the custom sequences, whereas right now it's hard-coded to use AP1:
https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/vendor/st/sequences/stm32h7.rs#L52

ryan-summers, 2025-02-11:
[image]

AP2 is actually the correct access point to use for accessing debug resources, so there's something else causing the flashing failures.

I'm currently under the impression the `info` command output above is a bit of a red-herring - it looks like we were looking at the wrong enable bit for the APs, which was fixed in #3089. This is why AP2 was being displayed as "Inaccessible".

#3089 also updated the DBGMCU definitions to support enabling the D2 power domain on debugger connection, which should ensure that the Cortex-M4 domain is powered up when the debugger attaches. This might resolve the flashing issues reported. For context, the original issue was that flashing reported failure to connect to the chip:
```
cargo run --release --features debug
    Finished `release` profile [optimized + debuginfo] target(s) in 0.32s
     Running `probe-rs run --chip STM32H747XIHx target/thumbv7em-none-eabihf/release/efficio-usb-dfu-bootloader`
Error: Connecting to the chip was unsuccessful.
```

badrbouslikhin, 2025-02-21:
I tried today with 8a5ab2f6, unfortunately I still have the same issue. Here are the full logs:

```
❯ probe-rs info --chip STM32H747XI

Probing target via JTAG
-----------------------

ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Failed to identify target using protocol JTAG: An error with the usage of the probe occurred

Caused by:
    0: An error occurred handling the JTAG scan chain.
    1: Invalid IR scan chain

Probing target via SWD
----------------------

ERROR probe_rs::architecture::arm::memory::romtable:    Failed to read component information at 0x0.

ARM Chip with debug port Default:

Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4500, Revision: 0x0, Instance: 0x00
├── V1(0) MemoryAP
│   └── 0 MemoryAP (AmbaAhb3)
│       ├── 0xe00fe000 ROM Table (Class 1), Designer: STMicroelectronics
│       ├── 0xe00ff000 ROM Table (Class 1), Designer: ARM Ltd
│       ├── 0xe0041000 ETM architecture (Coresight Component)
│       └── 0xe0043000 Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
├── V1(1) MemoryAP
│   └── 1 MemoryAP (AmbaAhb3)
├── V1(2) MemoryAP
│   └── 2 MemoryAP (AmbaApb2Apb3)
│       ├── 0xe00e0000 ROM Table (Class 1), Designer: STMicroelectronics
│       ├── 0xe00e3000 CoreSight SWO   (Coresight Component)
│       ├── 0xe00e4000 CoreSight TraceFunnel (Coresight Component)
│       ├── 0xe00e5000 System TSGEN    (Core Link / Prime Cell / System component)
│       ├── 0xe00f0000 ROM Table (Class 1), Designer: STMicroelectronics
│       ├── 0xe00f1000 Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
│       ├── 0xe00f3000 CoreSight TraceFunnel (Coresight Component)
│       ├── 0xe00f4000 CoreSight TMC   (Coresight Component)
│       └── 0xe00f5000 CoreSight TPIU  (Coresight Component)
└── V1(3) Unknown AP (Designer: <unknown>, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x2)
```

mstaack, 2025-05-20:
seeing something similar:

```
probe-rs info --chip STM32H747XI
Probing target via JTAG

 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4500, Revision: 0x0, Instance: 0x00
├── 0 MemoryAP (AmbaAhb3)
│   └── Error during access: The debug probe encountered an error.
├── 1 MemoryAP (AmbaAhb3)
│   └── Error during access: The debug probe encountered an error.
├── 2 MemoryAP (AmbaApb2Apb3)
│   └── Error during access: The debug probe encountered an error.
└── 3 Unknown AP (Designer: <unknown>, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x2)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed
```

```
probe-rs -V
probe-rs 0.27.0
```

mstaack, 2025-05-20:
did some git bisect, maybe this helps finding the issue:

that output looks a little better than the one before.

```
➜  probe-rs git:(0094be9d) cargo run -p probe-rs-tools -- info
   Compiling probe-rs-tools v0.25.0 (/Users/max/probe-rs/probe-rs-tools)
   Compiling probe-rs v0.25.0 (/Users/max/probe-rs/probe-rs)
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 19.93s
     Running `target/debug/probe-rs info`
Probing target via JTAG

 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

ARM Chip with debug port Default:
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApWdataError
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApWdataError
Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4500, Revision: 0x0, Instance: 0x00
├── 0 MemoryAP (AmbaAhb3)
│   └── ROM Table (Class 1), Designer: STMicroelectronics
│       ├── ROM Table (Class 1), Designer: ARM Ltd
│       │   ├── Cortex-M4 SCS   (Generic IP component)
│       │   │   └── CPUID
│       │   │       ├── IMPLEMENTER: ARM Ltd
│       │   │       ├── VARIANT: 1
│       │   │       ├── PARTNO: Cortex-M7
│       │   │       └── REVISION: 1
│       │   ├── Cortex-M3 DWT   (Generic IP component)
│       │   ├── Cortex-M7 FBP   (Generic IP component)
│       │   └── Cortex-M3 ITM   (Generic IP component)
│       ├── Cortex-M7 ETM   (Coresight Component)
│       └── Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
├── 1 MemoryAP (AmbaAhb3)
│   └── Error during access: The debug probe encountered an error.
├── 2 MemoryAP (AmbaApb2Apb3)
│   └── Error during access: The debug probe encountered an error.
└── 3 Unknown AP (Designer: <unknown>, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x2)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

➜  probe-rs git:(0094be9d) git bisect good
0fa885a47e32966a87479b93f6c51edd3cfdc7fa is the first bad commit
commit 0fa885a47e32966a87479b93f6c51edd3cfdc7fa
Author: Dominik Boehi <dboehi@fastmail.com>
Date:   Thu Dec 19 01:51:43 2024 +0100

    bugfix: Don't try to access disabled MemoryAPs in the info command (#2972)

 changelog/fixed-memory-port-enable-check.md                   |  1 +
 probe-rs-tools/src/bin/probe-rs/cmd/info.rs                   | 12 +++++++++++-
 probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb3.rs       |  5 +----
 probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb5.rs       |  5 +----
 probe-rs/src/architecture/arm/ap/memory_ap/amba_ahb5_hprot.rs |  5 +----
 probe-rs/src/architecture/arm/ap/memory_ap/amba_apb2_apb3.rs  |  5 +----
 probe-rs/src/architecture/arm/ap/memory_ap/amba_apb4_apb5.rs  |  5 +----
 probe-rs/src/architecture/arm/ap/memory_ap/amba_axi3_axi4.rs  |  5 +----
 probe-rs/src/architecture/arm/ap/memory_ap/amba_axi5.rs       |  5 +----
 probe-rs/src/architecture/arm/ap/memory_ap/mod.rs             | 36 +++++++++++++++++++++++++++++++++---
 probe-rs/src/architecture/arm/ap/mod.rs                       |  6 +++++-
 probe-rs/src/architecture/arm/communication_interface.rs      | 11 +++++++++--
 12 files changed, 66 insertions(+), 35 deletions(-)
 create mode 100644 changelog/fixed-memory-port-enable-check.md
```

bugadani, 2025-05-20:
Do you actually have an issue using probe-rs, or do you just have a problem with the info output showing you errors?

mstaack, 2025-05-21:
just a quick update.

removing:
```
        // Check if the AP is accessible
        let (interface, ap) = memory.try_as_parts()?;
        let csw = ap.generic_status(interface)?;
        if !csw.DeviceEn {
            return Ok(Tree::new(
                "Memory AP is not accessible, DeviceEn bit not set".to_string(),
            ));
        }
``` from https://github.com/probe-rs/probe-rs/commit/0fa885a47e32966a87479b93f6c51edd3cfdc7fa?diff=split#diff-b61d7da87ac178f36c2aebf1a765a051f118d2352c4f6f44173257db2ff32d87

gives this output, which i expect to be better:

```
➜  probe-rs git:(0fa885a4) cargo run -p probe-rs-tools -- info --chip STM32H747XI
   Compiling probe-rs-tools v0.25.0 (/Users/max/probe-rs/probe-rs-tools)
warning: unused import: `MemoryApType`
 --> probe-rs-tools/src/bin/probe-rs/cmd/info.rs:8:27
  |
8 |             ap::{ApClass, MemoryApType},
  |                           ^^^^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

warning: `probe-rs-tools` (bin "probe-rs") generated 1 warning
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 5.39s
     Running `target/debug/probe-rs info --chip STM32H747XI`
Probing target via JTAG

 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

ARM Chip with debug port Default:
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApWdataError
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApWdataError
Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4500, Revision: 0x0, Instance: 0x00
├── 0 MemoryAP (AmbaAhb3)
│   └── ROM Table (Class 1), Designer: STMicroelectronics
│       ├── ROM Table (Class 1), Designer: ARM Ltd
│       │   ├── Cortex-M4 SCS   (Generic IP component)
│       │   │   └── CPUID
│       │   │       ├── IMPLEMENTER: ARM Ltd
│       │   │       ├── VARIANT: 1
│       │   │       ├── PARTNO: Cortex-M7
│       │   │       └── REVISION: 1
│       │   ├── Cortex-M3 DWT   (Generic IP component)
│       │   ├── Cortex-M7 FBP   (Generic IP component)
│       │   └── Cortex-M3 ITM   (Generic IP component)
│       ├── Cortex-M7 ETM   (Coresight Component)
│       └── Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
├── 1 MemoryAP (AmbaAhb3)
│   └── Error during access: The debug probe encountered an error.
├── 2 MemoryAP (AmbaApb2Apb3)
│   └── Error during access: The debug probe encountered an error.
└── 3 Unknown AP (Designer: <unknown>, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x2)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

jseketa, 2026-11-08:
I tried to reproduce this on a Nucleo-H755ZI-Q (same debug setup as the H747: CM7 on AP0, CM4 on AP3) with an ST-LINK V3.

With the BCM4 option bit cleared, so the CM4 does not boot, connecting fails the way it does here:
- 0.25.0, 0.26.0: `The AP has the wrong type for the operation.`
- 0.32.0, master: `The AP with address FullyQualifiedApAddress { dp: Default, ap: V1(3) } does not exist.`
- 0.24.0 connects, because its target data put the CM4 on AP0.

#4664 makes probe-rs treat the CM4 as disabled in that case, and connecting works.

But in all outputs in this issue, AP3 reads as `Unknown AP (... Revision: 0x2)`, also with 0.24.0. On the H755 with BCM4 cleared it does not: master reads its IDR as 0, and 0.24.0 shows it as a memory AP. So I don't know yet whether your boards are in the same state.

I also tried a CM4 program that enters STOP right away. Connecting then fails differently: 0.32.0 and master time out halting the CM4, and 0.24.0 connects.

Could you add:
1. Whether the CM4 boots on your board: the BCM4 option bit (STM32CubeProgrammer, Option bytes, User Configuration), or `FLASH_OPTSR_CUR` at 0x5200201C, bit 22.
2. What the CM4 runs: nothing (blank bank 2), your own program, or ST's dual-core template, which puts the CM4 into STOP.
3. Whether #4664 connects for you.