probe-rs/probe-rs #3094: Problem with CMSIS-DAPLink

szg251, 2025-02-12:
### Describe the bug

I cannot connect my CMSIS-DAPLink debug probe to my microcontrollers (I tried with an STM32F103C8T6 and PICO R2040, both with the same error. I managed to connect, flash and debug with `pyOcd`, so this seems to be a problem with probe-rs, and not with my device.

The error message is the following

`probe-rs` can see the device:
```
❯ probe-rs list
The following debug probes were found:
[0]: WCH_Link -- 2a86:8011:0001A0000001 (CMSIS-DAP)
```

But when I try `info` or any other command, I get:

```
❯ probe-rs info --chip STM32F103C8
Error: Failed to open the debug probe.

Caused by:
    0: The debug probe could not be created.
    1: An error specific with the selected probe occurred.
    2: Usb endpoint not found.
```

### To Reproduce

1. connect device
2. try running `probe-rs info`

### Stacktrace

```shell
❯ RUST_LOG=debug probe-rs info --chip STM32F103C8
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Probing device 4295020872
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Serial Number`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `kUSBString`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Probing device 4295020888
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `kUSBString`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `kUSBString`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Probing device 4295020919
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Probing device 4295021797
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Serial Number`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `kUSBString`
DEBUG list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using nusb, searching HID
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Probing device 4295020872
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Serial Number`
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `kUSBString`
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Probing device 4295020888
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `kUSBString`
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `kUSBString`
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Probing device 4295020919
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Probing device 4295021797
DEBUG list_ftdi_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Serial Number`
[... 179 lines cut ...]
I added a few extra tracing::debug! lines, and found out that there is only 1 USB endpoint in the config:

                    for endpoint in altsetting.endpoints() {
                        tracing::debug!("Endpoint count: {}", altsetting.num_endpoints());
                        tracing::debug!("{:?}", endpoint);
                        if endpoint.address() == ENDPOINT_OUT {
                            endpoint_out = true;
                        } else if endpoint.address() == ENDPOINT_IN {
                            endpoint_in = true;
                        }
                    }

Result:

DEBUG nusb::platform::macos_iokit::device: Active config from single descriptor is 1
DEBUG probe_rs::probe::wlink::usb_interface: Endpoint count: 1
DEBUG probe_rs::probe::wlink::usb_interface: Endpoint { address: 0x84, direction: In, transfer_type: Interrupt, max_packet_size: 64, packets_per_microframe: 1, interval: 2 }
DEBUG nusb::platform::macos_iokit::events: event loop thread exited

This is where I gave up for now.
```

### Operating System

macOS

Yatekii, 2025-02-22:
Seems like WCH-Link does not adhere to the spec, but I guess we can make this a bit more resilient!

szg251, 2025-02-24:
> Seems like WCH-Link does not adhere to the spec, but I guess we can make this a bit more resilient!

Yes, I'm not exactly sure if this is even a genuine WCH debugger. I'm quite new to this embedded development thing, so I bought the first thing that looked viable. Nevertheless, it works with pyOCD, so it's not totally garbage :)

Let me know, if I can help you with anything else.

bugadani, 2025-04-24:
pyocd also needs 2 or 3 endpoints (https://github.com/pyocd/pyOCD/blob/90ee63a57a73ccca6da10b4ad5e3ebd5a6775e58/pyocd/probe/pydapaccess/interface/pyusb_v2_backend.py#L315) so if it works, there's some hope for the probe yet.