probe-rs/probe-rs #3095: JLink speed setting scaling is off

Yatekii, 2025-02-12:
### Describe the bug

Currently the JLink takes values like `400` or `1000` which result in `4 MHz` and `10 MHz` respectively because of an internal factor of 10. It would be good to make this unified with the other probes.

### Stacktrace

```shell

```

### Operating System

None

bugadani, 2025-04-29:
I don't see an issue here. --speed 400 gave me a 400kHz clock output, --speed 1000 1MHz. The cheap blue j-link clone couldn't keep up with this, but that's not probe-rs's fault - we do what the documentation asks us to do to set up the clock speed.

closed 2025-04-29 by bugadani

Yatekii, 2025-04-29:
Something here is wrong. I did not measure the clock speed but at least the output in probe-rs is not correct with J-Link EDU.

Yatekii, 2025-04-29:
When I give it a speed of 4000 or 10000 respectively it errors. The only speed I can set is 1000 which corresponds to 10000 MHz on the J-Link afaik, because that's the max on EDU. I tried many different thresholds and the EDU refuses to accept any but 1000. Looking at code and output it indicates that this leads to 10MHz which would be incorrect.